m-labs/artiq #642: HasEnvironment prepare method

dleibrandt, 2016-12-15:
This is a followup to #545.  I'm trying to do similar things to @r-srinivas, but most of my subcomponents inherit from HasEnvironment and not EnvExperiment (since the subcomponents don't have run methods).  I'd like HasEnvironment to have the same prepare method as EnvExperiment, where it calls the prepare method of its children.  Would it be possible to implement this?

sbourdeauducq, 2016-12-16:
``HasEnvironment`` just provides access to the dataset, devices, etc. without the prepare/run/analyze structure that comes with ``Experiment`` and relates to the pipeline stages of the scheduler. So not in ARTIQ. But you can implement it yourself by copy-pasting this small bit of ARTIQ code where you define ``prepare`` (``self.children`` is provided by ``HasEnvironment``):
```python
    def prepare(self):
        for child in self.children:
            if hasattr(child, "prepare"):
                child.prepare()
```

Might be a good idea to call your ``prepare`` something else if it doesn't relate to the preparation stage of the scheduler pipeline.

closed 2016-12-16 by sbourdeauducq

sbourdeauducq, 2019-09-25:
https://github.com/m-labs/artiq/commit/829885d381a1f419876e40ff12fede13bc78dc43#commitcomment-35229255
see ``call_children_method``