m-labs/artiq #643: phaser2 configure device using openocd

jbqubit, 2016-12-16:
This regards phaser2 branch. 

- [ ] runtime test mode via serial command line interface is missing 
- artiq_flash is not supported for phaser2; manual openocd commands are missing from instructions for following operations
    - [ ] artiq_flash -f flash_storage.img proxy storage start
    - [ ] artiq_coreconfig write -f idle_kernel idle.elf

jbqubit, 2016-12-16:
@jordens from your email I resolved the second two

```
(phaser2) britton@britton1:~/artiq-dev-phaser2/artiq/artiq/examples/phaser$ artiq_mkfs phaser_config.bin -s mac 00:0a:35:03:1e:53 -s ip 192.168.1.71 -s startup_clock i -f startup_kernel startup_kernel.elf

(phaser2) britton@britton1:~/artiq-dev-phaser2/artiq/artiq/examples/phaser$ openocd -f board/kc705.cfg -s ~/anaconda3/envs/phaser2/share/openocd/scripts -c "init; jtagspi_program phaser_config.bin 0xb80000; xc7_program xc7.tap; exit"
```

Regarding test mode... Do you intend to dump test mode here forward?

jordens, 2016-12-16:
There should be a `jtagspi_init 0 /path/to/bscan_spi_xc7k325t.bit` after the init.

Test mode was removed in the move to rust. I did like it and miss it as well but it was a bit tedious to maintain and hard to test itself.

closed 2016-12-16 by jordens

whitequark, 2016-12-16:
We can bring back test mode if there's enough interest, there's no fundamental reason for its removal.

jbqubit, 2016-12-16:
Ah, thanks Robert.

```
(phaser2) britton@britton1:~/artiq-dev-phaser2/artiq/artiq/examples/phaser$ 
artiq_mkfs phaser_config.bin \
-s mac 00:0a:35:03:1e:53 \
-s ip 192.168.1.71 \
-s startup_clock i \
-f startup_kernel startup_kernel.elf

(phaser2) britton@britton1:~/artiq-dev-phaser2/artiq/artiq/examples/phaser$ 
openocd -f board/kc705.cfg -s ~/anaconda3/envs/phaser2/share/openocd/scripts -c "init; \
jtagspi_init 0 /home/britton/.migen/bscan_spi_xc7k325t.bit; \
jtagspi_program phaser_config.bin 0xb80000; \
xc7_program xc7.tap; \
exit"
```
I've found the Test Mode useful in a number of circumstances and would miss its absence. This is in part a result of familiarity with that path for configuring MAC/IP and doing simple LED/IO toggle tests. I can see that openocd is a clear alternate path for setting MAC/IP.