m-labs/artiq #645: search feature broken in html manual

jbqubit, 2016-12-16:
At this URL using Chromium browser (53.0.2785.143) on Linux. 
http://m-labs.hk/artiq/manual-release-2

Behavior reproduces for dev branch when I build html on my machine.

jordens, 2016-12-16:
Works for me https://m-labs.hk/artiq/manual-release-2/search.html?q=pulse&check_keywords=yes&area=default

jbqubit, 2016-12-16:
Your search string "pulse" returns
- artiq.coredevice.ttl.TTLInOut.pulse
- artiq.coredevice.ttl.TTLOut.pulse
- etc

However it does not return free-text references to "pulse" in the documentation. This is problematic for other reasonable search terms like: mac, openocd, conda, device_db.pyon.

jordens, 2017-01-17:
works fine in master

jbqubit, 2017-01-25:
I'm on master. I can build html using sphinx 1.5.1. Ubuntu 14.04, Chromium browser. When I load ```artiq/doc/manual/_build/html/index.html``` I see a search box on the left-hand side. 

- A search for "EnvExperiment" returns some results but it not a free-text search of all the documentation. It fails to find for example ```artiq/doc/manual/_build/html/release_notes.html```

- A search for "HDF5" returns no results.

jordens, 2017-01-25:
works fine here: https://m-labs.hk/artiq/manual-master/search.html?q=HDF5&check_keywords=yes&area=default

maybe sphinx-doc/sphinx#3270 or/and sphinx-doc/sphinx#2989
or sphinx-doc/sphinx#2930 sphinx-doc/sphinx#2989 sphinx-doc/sphinx#1918 

could you try `make clean html`?

jbqubit, 2017-01-25:
​$ make clean html 

file:///home/britton/artiq3/artiq/doc/manual/_build/html/search.html?q=HDF5&check_keywords=yes&area=default

I see animated "Searching..." ​

[image]

jordens, 2017-01-25:
Try downgrading to sphinx 1.4.6 (and `make clean html`). That fixes it here.

jbqubit, 2017-01-25:
Yes. That fixes it for me. Confirmed that it's fine on both Chromium and
Firefox. Can you revert sphinx on the build system for the html that is
used by m-labs.hk?

Joe Britton
Sensors and Electron Devices
Army Research Lab
2800 Powder Mill Rd
Adelphi, MD 20783
301-394-3130
joseph.w.britton5.civ@mail.mil

jordens, 2017-01-25:
Your instructions on how we should solve this make me suspect you would like to get involved and do it yourself.

Your instructions are wrong. As you can tell right away from the link I posted above, it works fine on master.
The issue you are seeing locally that I worked around for you is a different one (cross scripting defense) than the one release-2.

jbqubit, 2017-01-25:
I could have been more clear. It's still broken on the URL linked to by "Manual" on the m-labs.hk homepage.

https://m-labs.hk/artiq/
https://m-labs.hk/artiq/manual-release-2/
https://m-labs.hk/artiq/manual-release-2/search.html?q=HDF5&check_keywords=yes&area=default#

jordens, 2017-01-25:
I know. I just explained to you that that one and the one I just resolved for you are different bugs.

commit 151b51f8 (2017-01-25): conda: pin sphinx to 1.4.8 (closes #645)

can be reverted when https://github.com/snide/sphinx_rtd_theme/pull/346 is
released
commit 364bd726 (2017-01-25): conda: pin sphinx to 1.4.8 (closes #645)

can be reverted when https://github.com/snide/sphinx_rtd_theme/pull/346 is
released
closed 2017-01-25 by jordens