m-labs/artiq #646: Passing kernel function pointers between classes

cjbe, 2016-12-17:
I am trying to pass a kernel function pointer to a child class for it to run - this currently does not work. Is there any chance of this being supported?

The experiment:
```python
from artiq.experiment import *

class Child(HasEnvironment):
    def build(self, func_pointer):
        self.setattr_device("core")
        self.func_pointer=func_pointer

    @kernel
    def run(self):
        self.func_pointer()

class FuncPointerTest(EnvExperiment):
    def build(self):
        self.setattr_device("core")

    @kernel
    def kfunc(self):
        pass

    def run(self):
        self.child = Child(self, self.kfunc)
        self.child.run()
```
The traceback:
```
root:Terminating with exception (AttributeError: type object 'ChildTest' has no attribute 'func_pointer')
Traceback (most recent call last):
  File "/home/ion/artiq-dev/artiq/artiq/master/worker_impl.py", line 228, in main
    exp_inst.run()
  File "/home/ion/repo/tests/artiq/func_pointer_test.py", line 23, in run
    self.child.run()
  File "/home/ion/artiq-dev/artiq/artiq/language/core.py", line 54, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "/home/ion/artiq-dev/artiq/artiq/coredevice/core.py", line 111, in run
    self.compile(function, args, kwargs, set_result)
  File "/home/ion/artiq-dev/artiq/artiq/coredevice/core.py", line 95, in compile
    library = target.compile_and_link([module])
  File "/home/ion/artiq-dev/artiq/artiq/compiler/targets.py", line 180, in compile_and_link
    return self.link([self.assemble(self.compile(module)) for module in modules],
  File "/home/ion/artiq-dev/artiq/artiq/compiler/targets.py", line 180, in <listcomp>
    return self.link([self.assemble(self.compile(module)) for module in modules],
  File "/home/ion/artiq-dev/artiq/artiq/compiler/targets.py", line 137, in compile
    llmod = module.build_llvm_ir(self)
  File "/home/ion/artiq-dev/artiq/artiq/compiler/module.py", line 82, in build_llvm_ir
    return llvm_ir_generator.process(self.artiq_ir, attribute_writeback=True)
  File "/home/ion/artiq-dev/artiq/artiq/compiler/transforms/llvm_ir_generator.py", line 408, in process
    self.process_function(func)
  File "/home/ion/artiq-dev/artiq/artiq/compiler/transforms/llvm_ir_generator.py", line 570, in process_function
    llinsn = getattr(self, "process_" + type(insn).__name__)(insn)
  File "/home/ion/artiq-dev/artiq/artiq/compiler/transforms/llvm_ir_generator.py", line 1469, in process_Quote
    return self._quote(insn.value, insn.type, lambda: [repr(insn.value)])
  File "/home/ion/artiq-dev/artiq/artiq/compiler/transforms/llvm_ir_generator.py", line 1416, in _quote
    lambda: path() + ['__class__'])
  File "/home/ion/artiq-dev/artiq/artiq/compiler/transforms/llvm_ir_generator.py", line 1417, in _quote
    return _quote_attributes()
  File "/home/ion/artiq-dev/artiq/artiq/compiler/transforms/llvm_ir_generator.py", line 1391, in _quote_attributes
    attrvalue = getattr(value, attr)
AttributeError: type object 'ChildTest' has no attribute 'func_pointer'
```

whitequark, 2016-12-19:
I see no obvious reason this wouldn't work; I'll look into it.

whitequark, 2017-01-31:
Actually, this is a duplicate of #638, and it works now.

closed 2017-01-31 by whitequark