m-labs/artiq #651: analyzer: support large data

sbourdeauducq, 2017-01-02:
The analyzer record format doesn't have enough data bits for SAWG events.

Fixing this requires defining a new record format that has ideally:
 * is efficient (supports variable length records so that e.g. TTL events do not use the full 512 bits of memory reserved for the data)
 * preserves transaction atomicity (so that data flushed out of the FIFO cannot cause corruption - instead the full record corresponding to the lost data is dropped)

Then it needs to be implemented in gateware and in ``artiq_coreanalyzer``.

whitequark, 2017-01-02:
So a self-synchronizing variable-length encoding? Something like this:
  * `0by0xxxxxx`: 6 data bits
  * `0by10xxxxx`: 5 data bits
  * `0by110xxxx`: 4 data bits
  * `0by1110xxx`: 3 data bits
  * `0by11110xx`: 2 data bits
  * `0by111110x`: 1 data bit
  * `0by1111110`: 0 data bits

where `z` indicates whether this is the initial byte or continuation byte,
and a 0 data bit byte indicating the end of the sequence.

jordens, 2017-01-02:
It would be nice if DMA (input and output), Analyzer, and DRTIO all three could use the same (or at least very similar) serialization formats.

@whitequark `y`?

sbourdeauducq, 2017-01-02:
There cannot be a lot in common, except for trivialities (e.g. the order of those few fields that are common), and more importantly for the way the variable length is communicated.

whitequark, 2017-01-02:
@jordens yes, `y`.

sbourdeauducq, 2017-01-02:
And the analyzer is the only one that requires the variable length encoding to be self-synchronizing, so it's not clear that even that can be shared.

whitequark, 2017-01-02:
Well, it is definitely less error-prone to have one variable-length encoder than two, on both sides.