m-labs/artiq #653: cannot unify numpy.int? with float

jbqubit, 2017-01-11:
```
from artiq.experiment import *

class SAWGTest(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("sawg0")

    @kernel
    def run(self):
        self.core.reset()
        delay(10*ms)
        self.sawg0.phase1.set(0)
```

```
repository/bug.py:14:31-14:32: error: cannot unify numpy.int? with float
        self.sawg0.phase1.set(0)
                              ^ 
repository/bug.py:14:31-14:32: note: expression of type numpy.int?
        self.sawg0.phase1.set(0)
                              ^ 
```
- The compiler be able to handle this type cast. 
- Where is the unit for turns? 1*turn = 2pi
https://github.com/m-labs/artiq/blob/master/artiq/language/units.py

jordens, 2017-01-11:
The argument is a float. Give it a float.
`1 turn = 1 = 1*s*Hz = 1*s*s*Hz*Hz`. It's unitless.

closed 2017-01-11 by jordens

jbqubit, 2017-01-11:
- This is inconsistent. I can set a frequency 50*MHz.
- Since when is phase unitless? Common units are degrees, radians, turns. ARTIQ is explicit about V, s, A, etc.

jordens, 2017-01-11:
No. It's not inconsistent 50*MHz is a float. It is 50*1e6.
You first claim "1*turn=2pi". What unit does that have? Both sides?
Phase has units implied by context or convention: people are allowed to write "2*pi" in papers as a phase. pi has no units.

jbqubit, 2017-01-11:
> No. It's not inconsistent 50MHz is a float. It is 50e6.

It implicitly becomes a float after unit processing. The same happens in the following example too.

``` self.sawg0.frequency0.set(50000000) ```

> You first claim "1turn=2pi". What unit does that have? Both sides?

OK I was imprecise. 0.5 turn is 1*pi rad is 180 deg.

jbqubit, 2017-01-25:
@whitequark Thoughts on this?

whitequark, 2017-01-26:
@jbqubit It is possible to implement automatic coercions for argument passing in ARTIQ Python, but due to the way type inference works, it's a zero-sum game: for every case where I add a possible coercion, there will be another case where a type annotation is now mandatory. Especially since that will break existing code, I think this feature is a net negative, and in any case the added effort would seem to be higher than fixing the type of a numeric constant.

jbqubit, 2017-01-26:
@whitequark Ya, mandatory type annotation isn't desirable. Thanks for the comment.

jordens, 2017-01-26:
@jbqubit 
* `self.sawg0.frequency0.set(50000000)` does not work in exactly the same way as `phase0.set(0)` does not work.
* You have a skewed view of what `units.py` does and what it can do. This is purely syntactic sugar and readability and not -- as you seem to think --  a unit system that does things like conversions for you.
* `0.5*turns` can not be `1*pi` with our unit system. We would have to redefine `pi` or use `rad` as the "default" phase.

whitequark, 2017-01-26:
@jbqubit To add to this, automatic coercions for arithmetic operations already result in necessity for type annotations in some cases. E.g. this function is ambiguous:
```python
def add(x, y):
  return x + y
```
This is not a very large problem right now because passing a variable with a known type, say a field, to such a function monomorphizes it. But if every argument were subject to coercion, that would become impossible.