m-labs/artiq #655: Unbreak the !unconditionally_dereferenceable LLVM patch

whitequark, 2017-01-16:
See the discussion in https://reviews.llvm.org/D18738. Sanjoy Das is convinced that it is unsound. Although I am not quite as convinced his approach is cleaner and could be merged upstream.

whitequark, 2017-03-28:
Triage: the upstream still cannot agree on the desired semantics for a feature like this: https://reviews.llvm.org/D20116. I am now more convinced that the patch we have is, in fact, unsound.

whitequark, 2017-04-28:
D20116 is in.

jordens, 2017-05-11:
@whitequark is this now merely a consolidation/upstreaming thing or should/would the semantics change?

whitequark, 2017-05-11:
@jordens The upstreamed patch is the equivalent of our patch but for function calls rather than loads and it is immediately useless to us in its current form. (It might be useful for other things, not sure yet.) However the argument that allowed it to get in can be used when revisiting !unconditionally_dereferenceable and we should do that.

whitequark, 2017-12-25:
Per correspondence with LLVM developers, the patch is broken by design and cannot be fixed (and likely can lead to miscompilations already). We'll have to find some other way of improving the performance of the loops it was used to speed up.

closed 2017-12-25 by whitequark