m-labs/artiq #658: pipistrello runtime does not compile [PPP]

jordens, 2017-01-23:
Probably some changes in the last lwip update:

http://buildbot.m-labs.hk/builders/artiq-board/builds/319/steps/conda_build/logs/stdio

```
/var/lib/buildbot/slaves/debian-stretch-amd64-2/miniconda/conda-bld/work/artiq/runtime/main.c:156:5: warning: implicit declaration of function 'ppp_set_auth' is invalid in C99 [-Wimplicit-function-declaration]
    ppp_set_auth(ppp, PPPAUTHTYPE_NONE, "", "");
    ^
/var/lib/buildbot/slaves/debian-stretch-amd64-2/miniconda/conda-bld/work/artiq/runtime/main.c:156:23: error: use of undeclared identifier 'PPPAUTHTYPE_NONE'
    ppp_set_auth(ppp, PPPAUTHTYPE_NONE, "", "");
                      ^
/var/lib/buildbot/slaves/debian-stretch-amd64-2/miniconda/conda-bld/work/artiq/runtime/main.c:150:6: warning: no previous prototype for function 'network_init' [-Wmissing-prototypes]
void network_init(void)
     ^```

whitequark, 2017-01-31:
This needs some sort of serial encapsulation, since we don't have a PPP implementation anymore.

@sbourdeauducq suggested SLIP. The argument for SLIP was that it's available on Linux and Windows, but that's not really the case; the last version of Windows that shipped with SLIP support was XP. Also the Linux tooling looks horrible from the quick look I made at it.

What I would prefer to do instead is to use a simple TAP interface on Linux, a framing with fixed maximum overhead, and a bespoke splitter on the host system, using the code already in smoltcp. This gets us several things:
  1. Simpler code on the core device. No need to implement a SLIP interface type alongside the Ethernet interface type in smoltcp. Sharing more code paths between pipistrello and kc705, preventing bit rot.
  2. Being able to see console messages at the same time as forwarding network packets.
  3. This implementation is trivially portable to any version of Windows using the [OpenVPN tap interface drivers](https://community.openvpn.net/openvpn/wiki/ManagingWindowsTAPDrivers) and a few [DeviceIoControl](https://community.openvpn.net/openvpn/wiki/ManagingWindowsTAPDrivers) calls.

sbourdeauducq, 2017-05-14:
I would make the case for dropping Pipistrello from ARTIQ 3.
* Smoltcp is already taking a lot of time, with issues like low throughput still not resolved.
* Only one user survey respondent seems to have a serious interest in it. I have contacted him to find out more.
* If we want to attract beginners, then Pipistrello should have good support and testing. This includes Windows support. Doing all this development and testing takes time. If the goal is to make physics more accessible, then I'd rather spend resources on ionpak or a good laser servo.
* Even with this effort, it would still be substandard due to the low speed of the serial interface, and the confusion that pppd (or equivalent) causes to beginners.
* Spartan-6 FPGAs are getting long in the tooth.
* The Kasli board ought to be reasonably priced and a much better alternative - easier to use, faster, and more powerful.

closed 2017-05-15 by sbourdeauducq

jbqubit, 2017-05-19:
Several prospective ARTIQ users had poor experience with Pipistrello due to it's second-class support. Agreed that it's better to drop support and focus on supporting Kasli in the future.

commit 9ab63920 (2017-05-15): Remove Pipistrello support

Closes #658
Closes #381