m-labs/artiq #659: Lack of information on stacktrace

vontell, 2017-01-24:
For exceptions where an operation is attempted between an `numpy.int32` and `numpy.int64`, there is no information (i.e. line number in my code) where the error occurs. Is there a way to enable more stacktrace information to find this error? Below is an example:

```
$ artiq_run repository/initialize-test.py 
Traceback (most recent call last):
  File "/home/artiq/anaconda3/envs/artiq-main/bin/artiq_run", line 11, in <module>
    load_entry_point('artiq==2.1', 'console_scripts', 'artiq_run')()
  File "/home/artiq/anaconda3/envs/artiq-main/lib/python3.5/site-packages/artiq/frontend/artiq_run.py", line 213, in main
    return run(with_file=True)
  File "/home/artiq/anaconda3/envs/artiq-main/lib/python3.5/site-packages/artiq/frontend/artiq_run.py", line 199, in run
    raise exn
  File "/home/artiq/anaconda3/envs/artiq-main/lib/python3.5/site-packages/artiq/frontend/artiq_run.py", line 192, in run
    exp_inst.run()
  File "/home/artiq/anaconda3/envs/artiq-main/lib/python3.5/site-packages/artiq/language/core.py", line 54, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "/home/artiq/anaconda3/envs/artiq-main/lib/python3.5/site-packages/artiq/coredevice/core.py", line 113, in run
    self.compile(function, args, kwargs, set_result)
  File "/home/artiq/anaconda3/envs/artiq-main/lib/python3.5/site-packages/artiq/coredevice/core.py", line 97, in compile
    library = target.compile_and_link([module])
  File "/home/artiq/anaconda3/envs/artiq-main/lib/python3.5/site-packages/artiq/compiler/targets.py", line 180, in compile_and_link
    return self.link([self.assemble(self.compile(module)) for module in modules],
  File "/home/artiq/anaconda3/envs/artiq-main/lib/python3.5/site-packages/artiq/compiler/targets.py", line 180, in <listcomp>
    return self.link([self.assemble(self.compile(module)) for module in modules],
  File "/home/artiq/anaconda3/envs/artiq-main/lib/python3.5/site-packages/artiq/compiler/targets.py", line 137, in compile
    llmod = module.build_llvm_ir(self)
  File "/home/artiq/anaconda3/envs/artiq-main/lib/python3.5/site-packages/artiq/compiler/module.py", line 84, in build_llvm_ir
    attribute_writeback=self.attribute_writeback)
  File "/home/artiq/anaconda3/envs/artiq-main/lib/python3.5/site-packages/artiq/compiler/transforms/llvm_ir_generator.py", line 408, in process
    self.process_function(func)
  File "/home/artiq/anaconda3/envs/artiq-main/lib/python3.5/site-packages/artiq/compiler/transforms/llvm_ir_generator.py", line 570, in process_function
    llinsn = getattr(self, "process_" + type(insn).__name__)(insn)
  File "/home/artiq/anaconda3/envs/artiq-main/lib/python3.5/site-packages/artiq/compiler/transforms/llvm_ir_generator.py", line 883, in process_Arith
    name=insn.name)
  File "/home/artiq/anaconda3/envs/artiq-main/lib/python3.5/site-packages/llvmlite_artiq/ir/builder.py", line 24, in wrapped
    % (lhs.type, rhs.type))
ValueError: Operands must be the same type, got (i32, i64)
```

whitequark, 2017-01-24:
> For exceptions where an operation is attempted between an numpy.int32 and numpy.int64, there is no information (i.e. line number in my code) where the error occurs.

Most operations between np.int32 and np.int64 will result in the desired effect i.e. promotion. However, you've managed to find a bug in the compiler where that doesn't happen. This does not indicate a bug in your code (at least, this backtrace doesn't).

To fix this, I will need the specific code that causes the crash above.

vontell, 2017-01-24:
My code begins with this [`initialize-test.py`](https://github.com/vontell/artiq-control/blob/master/samples/repository/initialize-test.py) EnvExperiment. Before line 52, there is mostly just variable declaration, which has compiled fine in the past. On line 52 is a method `rabi.get_photon_windows()`, which I have also run and tested with no trouble.

Therefore, line 58 seems to be where the trouble begins. On this line, `rabi.get_time_to_detect()` is a call to the file [`rle/rabi.py`](https://github.com/vontell/artiq-control/blob/master/samples/repository/rle/rabi.py). Lines [79 to 114](https://github.com/vontell/artiq-control/blob/master/samples/repository/rle/rabi.py#L79-L114) hold the relevant code, which none or little of which has code which may cause this issue.

The final destination is the method `register_rising_in_window()` in the file [`rle/pipistrello.py`](https://github.com/vontell/artiq-control/blob/master/samples/repository/rle/pipistrello.py), and can be found on line [216 to 245](https://github.com/vontell/artiq-control/blob/master/samples/repository/rle/pipistrello.py#L216-L245).

Sorry that my methods pass many variables along the way (I have been running into errors where "variables might be uninitialized within closures"). My thoughts are that the error is most likely occurring somewhere in `rabi.py`, possibly in the [`record` function](https://github.com/vontell/artiq-control/blob/master/samples/repository/rle/rabi.py#L66-L76) which I use to store results from my experiment into an array. I can also reproduce the code here in this thread, but I didn't want to cause too much clutter.

vontell, 2017-01-24:
The ['rotate()` method](https://github.com/vontell/artiq-control/blob/master/samples/repository/rle/pipistrello.py#L208) of `rabi.py` may also be the culprit.

vontell, 2017-01-30:
After some debugging, I have narrowed it down to the `rotate()` method:

```Python
@kernel
def rotate(self, array):
	'''Rotates an array, deleting the oldest value'''
	length = len(array)
	for i in range(np.int64(len(array)) - 1):
		array[length - i - 1] = array[length - i - 2]
	array[0] = 0
```

vontell, 2017-01-30:
I have avoided this error by changing the rotate function to the following:

```Python
neg_one = np.int32(-1)
array[neg_one:] + array[:neg_one]
```

If I change the code to use `neg_one = np.int64(-1)`, I get the error above, so I suppose it is related to the type needed for array access.

commit ab71c9a0 (2017-01-31): compiler: the len builtin is not polymorphic, coerce the result.

Fixes #659.
commit 31dd99e8 (2017-02-01): compiler: the len builtin is not polymorphic, coerce the result.

Fixes #659.
closed 2017-01-31 by whitequark