m-labs/artiq #661: unittest soft-FP/DDS

jordens, 2017-01-25:
We want to include a realistic unittest that exercises soft-FP and DDS event emission to track execution speed.

@r-srinivas I remember you ran into speed problems with sideband cooling. Could you give us the relevant part of your code?

sbourdeauducq, 2017-01-25:
The current DDS test was written to measure the speed of the C routines. This is why there is no sweep or other computations.

r-srinivas, 2017-01-26:
bermudez_cooling() in cooling.py requires about 600 - 1000 us of slack to
run. I think these files would be all you need. I think it's not the most
demanding sideband cooling sequence because some of our pi times for those
modes are pretty long.

jordens, 2021-11-22:
addressed by #1777

closed 2021-11-22 by jordens