m-labs/artiq #669: Dashboard prevents connection to coredevice

r-srinivas, 2017-02-07:
I wanted to test 3.0.dev py_820+gitf4ae166 on Windows. I installed artiq through conda, flashed the coredevice and modified its IP an MAC address following the instructions. At this point, I can ping the device and run experiments using artiq_run. However, if I start both artiq_master and artiq_dashboard, and try to run experiments from the dashboard, I get a timeout error,

```
root:Terminating with exception (timeout: timed out)
Traceback (most recent call last):
  File "C:\Anaconda3\envs\artiq-2017-02-06\lib\site-packages\artiq\master\worker_impl.py", line 251, in main
    exp_inst.run()
  File "C:\Anaconda3\envs\artiq-2017-02-06\lib\site-packages\artiq\language\core.py", line 53, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "C:\Anaconda3\envs\artiq-2017-02-06\lib\site-packages\artiq\coredevice\core.py", line 118, in run
    self.comm.check_system_info()
  File "C:\Anaconda3\envs\artiq-2017-02-06\lib\site-packages\artiq\coredevice\comm_generic.py", line 204, in check_system_info
    self._write_empty(_H2DMsgType.SYSTEM_INFO_REQUEST)
  File "C:\Anaconda3\envs\artiq-2017-02-06\lib\site-packages\artiq\coredevice\comm_generic.py", line 169, in _write_empty
    self._write_header(ty)
  File "C:\Anaconda3\envs\artiq-2017-02-06\lib\site-packages\artiq\coredevice\comm_generic.py", line 161, in _write_header
    self.open()
  File "C:\Anaconda3\envs\artiq-2017-02-06\lib\site-packages\artiq\coredevice\comm_tcp.py", line 45, in open
    self.socket = initialize_connection(self.host, self.port)
  File "C:\Anaconda3\envs\artiq-2017-02-06\lib\site-packages\artiq\coredevice\comm_tcp.py", line 28, in initialize_connection
    sock = socket.create_connection((host, port), 5.0)
  File "C:\Anaconda3\envs\artiq-2017-02-06\lib\socket.py", line 711, in create_connection
    raise err
  File "C:\Anaconda3\envs\artiq-2017-02-06\lib\socket.py", line 702, in create_connection
    sock.connect(sa)
socket.timeout: timed out
```

After this, even if I close the master and dashboard, the device still can't be pinged. If I power cycle the device then I can ping it again.  If I only start the master, I can ping the device. It's once I start the dashboard as well that socket connections fail. This seems to happen on the virtual machine as well. Any idea what might have caused this? The experiment I tried using artiq_run just to see if things worked was

```python
from artiq.experiment import *

class LED(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("led")

    @kernel
    def run(self):
        self.core.reset()
#        self.led.output()
        for i in range(10):        
            self.led.on()
            delay(500*ms)
            self.led.off()
            delay(500*ms)
```

And I could see the LED flash.

whitequark, 2017-02-07:
What's on the core device console?

r-srinivas, 2017-02-07:
You mean the corelog? 

```
(artiq-2017-02-06) C:\artiq-3-test>artiq_corelog
Traceback (most recent call last):
  File "C:\Anaconda3\envs\artiq-2017-02-06\Scripts\artiq_corelog-script.py", lin
e 10, in <module>
    load_entry_point('artiq==3.0.dev0+820.gf4ae166', 'console_scripts', 'artiq_c
orelog')()
  File "C:\Anaconda3\envs\artiq-2017-02-06\lib\site-packages\artiq\frontend\arti
q_corelog.py", line 25, in main
    comm.check_system_info()
  File "C:\Anaconda3\envs\artiq-2017-02-06\lib\site-packages\artiq\coredevice\co
mm_generic.py", line 204, in check_system_info
    self._write_empty(_H2DMsgType.SYSTEM_INFO_REQUEST)
  File "C:\Anaconda3\envs\artiq-2017-02-06\lib\site-packages\artiq\coredevice\co
mm_generic.py", line 169, in _write_empty
    self._write_header(ty)
  File "C:\Anaconda3\envs\artiq-2017-02-06\lib\site-packages\artiq\coredevice\co
mm_generic.py", line 161, in _write_header
    self.open()
  File "C:\Anaconda3\envs\artiq-2017-02-06\lib\site-packages\artiq\coredevice\co
mm_tcp.py", line 45, in open
    self.socket = initialize_connection(self.host, self.port)
  File "C:\Anaconda3\envs\artiq-2017-02-06\lib\site-packages\artiq\coredevice\co
mm_tcp.py", line 28, in initialize_connection
    sock = socket.create_connection((host, port), 5.0)
  File "C:\Anaconda3\envs\artiq-2017-02-06\lib\socket.py", line 711, in create_c
onnection
    raise err
  File "C:\Anaconda3\envs\artiq-2017-02-06\lib\socket.py", line 702, in create_c
onnection
    sock.connect(sa)
socket.timeout: timed out
```

using flterm I had to reset the device

```
(artiq-2017-02-06) C:\artiq-3-test>python -m misoc.tools.flterm COM2

MiSoC BIOS
(c) Copyright 2007-2016 M-Labs Limited
Built Feb  7 2017 06:41:20

BIOS CRC passed (78401bc9)
Initializing SDRAM...
Write leveling: 14* 15* 14  13   9   8   5   6  completed
Read bitslip: 7 6 5 4 3 2
Read delays: 7:01-11  6:01-12  5:03-14  4:03-15  3:11-21  2:10-20  1:00-09  0:00
-09  completed
Memtest OK
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Timeout
Booting from flash...
Loading 344096 bytes from flash...
Executing booted program.
[         110us]  INFO(runtime): ARTIQ runtime starting...
[        3937us]  INFO(runtime): software version 3.0.dev+820.gf4ae166
[       10115us]  INFO(runtime): gateware version 3.0.dev+820.gf4ae166
[       16277us]  INFO(runtime): press 'e' to erase startup and idle kernels...
[     1016075us]  INFO(runtime): continuing boot
[     1019401us]  INFO(runtime): using MAC address 00-0a-35-03-39-02
[     1025081us]  INFO(runtime): using IP address 132.163.182.89
[     1030748us] ERROR(runtime::rtio_mgt): unrecognized startup_clock configurat
ion entry
[     1038490us]  INFO(runtime::rtio_mgt): startup RTIO clock: Internal
[     1062304us]  INFO(runtime::session): accepting network sessions
[     1067151us]  INFO(runtime::session): running startup kernel
[     1072732us]  INFO(runtime::session): no startup kernel found
[     1078436us]  INFO(runtime::session): no connection, starting idle kernel
[     1085285us]  INFO(runtime::session): no idle kernel found
panic at src/libcore/result.rs:868: undersized payload: Truncated
```

r-srinivas, 2017-02-07:
Resetting flterm while the dashboard is running prevents you from being able to ping the device. This is the log after I reset with the dashboard terminated but master still running. At 32 I use `artiq_run` to run the experiment from above which works. I then started the dashboard which resulted in that panic line which then prevents me from communicating with the device.

```
MiSoC BIOS
(c) Copyright 2007-2016 M-Labs Limited
Built Feb  7 2017 06:41:20

BIOS CRC passed (78401bc9)
Initializing SDRAM...
Write leveling: 14* 15* 14  13   9   8   5   6  completed
Read bitslip: 7 6 5 4 3 2
Read delays: 7:00-11  6:01-12  5:03-14  4:03-15  3:11-21  2:10-21  1:00-09  0:00
-09  completed
Memtest OK
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Timeout
Booting from flash...
Loading 344096 bytes from flash...
Executing booted program.
[         110us]  INFO(runtime): ARTIQ runtime starting...
[        3937us]  INFO(runtime): software version 3.0.dev+820.gf4ae166
[       10115us]  INFO(runtime): gateware version 3.0.dev+820.gf4ae166
[       16277us]  INFO(runtime): press 'e' to erase startup and idle kernels...
[     1016075us]  INFO(runtime): continuing boot
[     1019401us]  INFO(runtime): using MAC address 00-0a-35-03-39-02
[     1025081us]  INFO(runtime): using IP address 132.163.182.89
[     1030748us] ERROR(runtime::rtio_mgt): unrecognized startup_clock configurat
ion entry
[     1038490us]  INFO(runtime::rtio_mgt): startup RTIO clock: Internal
[     1062410us]  INFO(runtime::session): accepting network sessions
[     1067262us]  INFO(runtime::session): running startup kernel
[     1072837us]  INFO(runtime::session): no startup kernel found
[     1078545us]  INFO(runtime::session): no connection, starting idle kernel
[     1085391us]  INFO(runtime::session): no idle kernel found
[    49340639us]  INFO(runtime::session): new connection from 132.163.182.41:614
32
[    50872110us]  INFO(runtime::session): resetting RTIO
[    51383662us]  INFO(runtime::session): no connection, starting idle kernel
[    51389405us]  INFO(runtime::session): no idle kernel found
panic at src/libcore/result.rs:868: undersized payload: Truncated
```

r-srinivas, 2017-02-09:
Is this something you guy see on your setup as well? Any other information I can try to get to help debug this?

cballance, 2017-02-09:
I see this as well, running artiq_master (commit f4ae166) on a Linux machine, and with the dashboard running on Windows. (My Linux machine for this is headless, so I cannot test the GUI under Linux).

cjbe, 2017-02-09:
I see this as well, running artiq_master (commit f4ae166) on a Linux machine, and with the dashboard running on Windows. (My Linux machine for this is headless, so I cannot test the GUI under Linux).

whitequark, 2017-02-10:
@r-srinivas @cjbe You could enable full logging with the patch below, and post the output from UART.
```diff
diff --git a/artiq/firmware/runtime/Cargo.toml b/artiq/firmware/runtime/Cargo.toml
index 01aca828..cf24c243 100644
--- a/artiq/firmware/runtime/Cargo.toml
+++ b/artiq/firmware/runtime/Cargo.toml
@@ -17,7 +17,7 @@ alloc_artiq = { path = "../liballoc_artiq" }
 std_artiq = { path = "../libstd_artiq", features = ["alloc"] }
 logger_artiq = { path = "../liblogger_artiq" }
 cslice = { version = "0.3" }
-log = { version = "0.3", default-features = false, features = ["max_level_debug"] }
+log = { version = "0.3", default-features = false, features = [] }
 board = { path = "../libboard", features = ["uart_console"] }
 fringe = { version = "= 1.1.0", default-features = false, features = ["alloc"] }
 byteorder = { version = "1.0", default-features = false }
@@ -31,4 +31,4 @@ features = ["mem"]
 git = "https://github.com/m-labs/smoltcp"
 rev = "e8ece3e"
 default-features = false
-features = ["use_alloc", "use_collections", "use_log"]#, "verbose"]
+features = ["use_alloc", "use_collections", "use_log", "verbose"]
diff --git a/artiq/firmware/runtime/session.rs b/artiq/firmware/runtime/session.rs
index 85afb310..b211a33e 100644
--- a/artiq/firmware/runtime/session.rs
+++ b/artiq/firmware/runtime/session.rs
@@ -642,7 +642,7 @@ pub fn thread(io: Io) {
     listener.listen(1381).expect("session: cannot listen");
     info!("accepting network sessions");
 
-    BufferLogger::with_instance(|logger| logger.disable_trace_to_uart());
+    // BufferLogger::with_instance(|logger| logger.disable_trace_to_uart());
 
     let congress = Urc::new(RefCell::new(Congress::new()));
 
```

cjbe, 2017-02-10:
@whitequark here is the UART output after applying your patch and trying to run an experiment from the dashboard (using artiq_run still works fine):
[uart-log.txt](https://github.com/m-labs/artiq/files/767860/uart-log.txt)

whitequark, 2017-02-10:
Oh wonderful, that's actually two bugs masking each other...

whitequark, 2017-03-29:
I believe that both smoltcp bugs were fixed, and we *also* stopped using UDP for moninj, so this is not reproducible anymore.

closed 2017-03-29 by whitequark