m-labs/artiq #671: Broadcasting large datasets fails

cjbe, 2017-02-16:
The following fails:
```python
data = np.random.rand(4e5)
self.set_dataset("big_data", data, broadcast=True)
```
It is useful to be able to broadcast datasets this large e.g. for displaying images. 
At minimum, it would be useful to have an error message that cleanly identifies the error.

The full error message is:
```
2017-02-16 22:29:52,306 ERROR:master:artiq.master.scheduler:got worker exception in run stage, deleting RID 63868
2017-02-16 22:29:52,306 ERROR:master:artiq.master.worker:worker exception details
Traceback (most recent call last):
  File "/home/ion/anaconda3/lib/python3.5/asyncio/streams.py", line 481, in readline
    line = yield from self.readuntil(sep)
  File "/home/ion/anaconda3/lib/python3.5/asyncio/streams.py", line 578, in readuntil
    'Separator is found, but chunk is longer than limit', isep)
asyncio.streams.LimitOverrunError: Separator is found, but chunk is longer than limit

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/ion/artiq-dev/artiq/artiq/master/scheduler.py", line 270, in _do
    completed = await run.run()
  File "/home/ion/artiq-dev/artiq/artiq/master/scheduler.py", line 32, in worker_method
    return await m(*args, **kwargs)
  File "/home/ion/artiq-dev/artiq/artiq/master/worker.py", line 268, in run
    completed = await self._worker_action({"action": "run"})
  File "/home/ion/artiq-dev/artiq/artiq/master/worker.py", line 242, in _worker_action
    completed = await self._handle_worker_requests()
  File "/home/ion/artiq-dev/artiq/artiq/master/worker.py", line 198, in _handle_worker_requests
    obj = await self._recv(self.watchdog_time())
  File "/home/ion/artiq-dev/artiq/artiq/master/worker.py", line 184, in _recv
    line = fs[0].result()
  File "/home/ion/anaconda3/lib/python3.5/asyncio/futures.py", line 274, in result
    raise self._exception
  File "/home/ion/anaconda3/lib/python3.5/asyncio/tasks.py", line 239, in _step
    result = coro.send(None)
  File "/home/ion/artiq-dev/artiq/artiq/protocols/pipe_ipc.py", line 17, in readline
    return await self.reader.readline()
  File "/home/ion/anaconda3/lib/python3.5/asyncio/streams.py", line 490, in readline
    raise ValueError(e.args[0])
ValueError: Separator is found, but chunk is longer than limit
```

sbourdeauducq, 2017-02-17:
This is coming from python asyncio. How large is too large?

whitequark, 2017-02-17:
I investigated this a bit, the default limit in asyncio is 64k (https://github.com/python/cpython/blob/master/Lib/asyncio/streams.py#L22); calling `open_connection(limit=...)` will fix the issue.

sbourdeauducq, 2017-02-17:
Yes, we're already doing that, but the current value is apparently too small - hence my question.

cjbe, 2017-02-17:
I don't know how large is too large, but several colleagues of mine have hit the ~400k double limit doing things that do not seem like an abuse of the broadcasting system. For example broadcasting a long scope trace or a vector of photon time-stamps.

If there has to be an arbitrary limit, it would be good for it to be at least an order of magnitude or two higher that at the moment - it seems like the uses cases for broadcasting ~100 MB datasets are far fewer than ~1 MB.

commit bd554366 (2017-02-17): protocols: increase asyncio line limit. Closes #671
commit c777ae8a (2017-02-17): protocols: increase asyncio line limit. Closes #671
closed 2017-02-17 by sbourdeauducq