m-labs/artiq #672: SPI intermittent output failure

dleibrandt, 2017-02-17:
I'm running artiq 2.2 on the kc705 nist clock hardware.

I'm testing SPI communication with a slave device by writing random values to configuration registers and reading them back, and I see intermittent failures of the SPI RTIO outputs for SPI clock frequencies faster than about 2 MHz.  The rate of these failures gets higher as the SPI clock frequency is increased.  Below 2 MHz, I never see failures (in > 1e5 trials).  Below are a few example scope traces.  Yellow is data in, blue is data out, purple is chip select out, and green is clock out.

Successful trial:
[image]

Failed trial 1 (the final few clock cycles were not output):
[image]

Failed trial 2 (the final few clock cycles were not output and chip select was late to be pulled high):
[image]

Failed trial 3 (similar mode):
[image]

sbourdeauducq, 2017-02-17:
Can you post kernel code that produces this issue?

dleibrandt, 2017-02-17:
Something like this:

```
    @kernel
    def run(self):
        self.core.reset()
        delay(10*ms)

        self.setup_bus()
        delay(10*ms)

        self.servo._get_config_uint(addr)

class DigitalServo:
    """
    SPI communication with NIST digital servos.

    :param spi_device: Name of the SPI bus this device is on.
    :param sample_device: Name of the TTL device that the is connected to
      the sample input (DIN2) of the digital servo (optional).
    """

    def __init__(self, dmgr, spi_device, sample_device=None):
        self.core = dmgr.get("core")
        self.bus = dmgr.get(spi_device)
        if sample_device is not None:
            self.sample = dmgr.get(sample_device)
        
        self.address = 0

    @kernel
    def setup_bus(self, write_div=50, read_div=50):
        self.bus.set_config_mu(_DIGITAL_SERVO_SPI_CONFIG, write_div, read_div)
        self.bus.set_xfer(1, 16, 16)
    
    @kernel
    def _get_config_uint(self, address):
        self.bus.write(((self.address & 0x3) << 30) | (0x1 << 28) | ((address & 0xFFF) << 16))
        self.bus.read_async()
        
        return int(self.bus.input_async() & 0xFFFF)

```

jordens, 2017-02-17:
I am worried about that `core.reset()`.
Could you provide an analyzer dump? I think the SPI channels should be covered by coreanalyzer but haven't checked lately.

sbourdeauducq, 2017-02-17:
There will be in the raw dump but decoding into VCD is not supported. @dleibrandt please provide the raw dump.

dleibrandt, 2017-02-17:
I've attached both below.

[rtio.zip](https://github.com/m-labs/artiq/files/783927/rtio.zip)

jordens, 2017-02-18:
@dleibrandt thanks.
* Did that dump include such an intermittent failure?
* What do you need the `core.reset()` for? Could you try without it (e.g. `break_realtime()` where needed instead)?
* The "intermittent" baffles me. I can't come up with a reason based on your description and your code that could make something like this intermittent.
* Your dump shows that there are `read_async()` immediately followed by `set_config_mu()`/`set_xfer()`. That's not how you do it in your example code. If you reset/change `config` or `xfer` or start a new `write` before the `read` has finished, or if you `core.reset()` the entire thing, that would obviously cause problems. Please provide the code that shows the problem and the corresponding dump.

I have added SPI support to coreanalyzer (in master).

dleibrandt, 2017-02-22:
> Did that dump include such an intermittent failure?

Yes.

> What do you need the core.reset() for? Could you try without it (e.g. break_realtime() where needed instead)?

You're right.  I don't actually need the `core.reset()`.  However, I still see the problem when I replace it with `core.break_realtime()`.

> The "intermittent" baffles me. I can't come up with a reason based on your description and your code that could make something like this intermittent.

Agreed.  I also don't have an explanation.

> Your dump shows that there are read_async() immediately followed by set_config_mu()/set_xfer(). That's not how you do it in your example code. If you reset/change config or xfer or start a new write [...]

Ah, yes.  Sorry about that.  I was trying to simplify things a bit, and I've been changing the code a lot lately.  This code:
```
from artiq.experiment import *
import time

class TestDigitalServo(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("scheduler")
        
        self.setattr_device("servo0")
        self.setattr_device("servo1")
        
        self.setattr_device("ttl12")
    
    def run(self):
        self.servo0.test_spi_com()

from artiq.language.core import (kernel, portable, delay_mu, delay)
from artiq.language.units import ns, us, ms
from artiq.language.types import TInt64, TInt32, TNone, TList
from artiq.coredevice import spi

_DIGITAL_SERVO_SPI_CONFIG = (0*spi.SPI_OFFLINE | 0*spi.SPI_CS_POLARITY |
                             0*spi.SPI_CLK_POLARITY | 0*spi.SPI_CLK_PHASE |
                             0*spi.SPI_LSB_FIRST | 0*spi.SPI_HALF_DUPLEX)

class DigitalServo:
    """
    SPI communication with NIST digital servos.

    :param spi_device: Name of the SPI bus this device is on.
    :param address: SPI address (used for daisy chaining multiple boxes).
    :param sample_device: Name of the TTL device that the is connected to
      the sample input (DIN2) of the digital servo (optional).
    """

    def __init__(self, dmgr, spi_device, address, sample_device=None):
        self.core = dmgr.get("core")
        self.bus = dmgr.get(spi_device)
        self.address = address
        if sample_device is not None:
            self.sample = dmgr.get(sample_device)

    @kernel
    def setup_bus(self, write_div=25, read_div=50): # write_div=25, read_div=125):
        """Configure the SPI bus and the SPI transaction parameters
        for this device. This method has to be called before any other method
        if the bus has been used to access a different device in the meantime.

        This method advances the timeline by the duration of two
        RTIO-to-Wishbone bus transactions.

        :param write_div: Write clock divider.
        :param read_div: Read clock divider.
        """
[... 224 lines cut ...]
INFO:worker(6108,TestDigitalServo.py):print:    failed: trial = 802 , set = 30373 , get = 30375
INFO:worker(6108,TestDigitalServo.py):print:    failed: trial = 804 , set = 9550 , get = 9551
INFO:worker(6108,TestDigitalServo.py):print:    failed: trial = 815 , set = 8873 , get = 8872
INFO:worker(6108,TestDigitalServo.py):print:    failed: trial = 819 , set = 43650 , get = 27264
INFO:worker(6108,TestDigitalServo.py):print:    failed: trial = 841 , set = 5546 , get = 5544
INFO:worker(6108,TestDigitalServo.py):print:    failed: trial = 858 , set = 52907 , get = 52904
INFO:worker(6108,TestDigitalServo.py):print:    failed: trial = 863 , set = 41813 , get = 41812
INFO:worker(6108,TestDigitalServo.py):print:    failed: trial = 864 , set = 36166 , get = 36161
INFO:worker(6108,TestDigitalServo.py):print:    failed: trial = 881 , set = 33343 , get = 22941
INFO:worker(6108,TestDigitalServo.py):print:    failed: trial = 882 , set = 42402 , get = 42400
INFO:worker(6108,TestDigitalServo.py):print:    failed: trial = 895 , set = 60742 , get = 60736
INFO:worker(6108,TestDigitalServo.py):print:    failed: trial = 900 , set = 46531 , get = 27643
INFO:worker(6108,TestDigitalServo.py):print:    failed: trial = 924 , set = 38238 , get = 38271
INFO:worker(6108,TestDigitalServo.py):print:    failed: trial = 933 , set = 63636 , get = 51547
INFO:worker(6108,TestDigitalServo.py):print:    failed: trial = 961 , set = 31427 , get = 18251
INFO:worker(6108,TestDigitalServo.py):print:    failed: trial = 976 , set = 21929 , get = 21928
INFO:worker(6108,TestDigitalServo.py):print:    failed: trial = 981 , set = 41081 , get = 42131
INFO:worker(6108,TestDigitalServo.py):print:    failed: trial = 989 , set = 58793 , get = 58792
INFO:worker(6108,TestDigitalServo.py):print:    failed: trial = 999 , set = 9045 , get = 9044
INFO:worker(6108,TestDigitalServo.py):print:    failed 103 out of 1000 trials
```

And here is the physical scope output:

[image]

jordens, 2017-03-09:
@dleibrandt just a quick note to confirm that I am still looking at this. Right now, the only things I can come up with is some weirdness or deep bug in the gateware or in Vivado that we haven't seen before or a spurious RTIO reset interfering.

sbourdeauducq, 2017-06-05:
Did anyone make any progress?

jordens, 2017-06-06:
I ran SPI reads over night on PDQ and couldn't reproduce this.

dleibrandt, 2017-06-06:
Using the 2.2 nist_clock gateware, or a different gateware?

sbourdeauducq, 2017-06-06:
I suppose 3.0/master, but the SPI code should be the same.

jordens, 2017-06-06:
Yes.

sbourdeauducq, 2017-06-06:
@dleibrandt Do you still have the problem with 2.3 (instead of 2.2) and that commit Robert mentions?

sbourdeauducq, 2017-06-06:
@dleibrandt Also it would be helpful if you found the simplest possible experiment code that produces the problem.

jordens, 2017-06-07:
With current master, I ran a couple million of these 16 MHz reads and they all passed fine.

```python
from artiq.experiment import *
from artiq.coredevice import spi

_DIGITAL_SERVO_SPI_CONFIG = (0*spi.SPI_OFFLINE | 0*spi.SPI_CS_POLARITY |
                             0*spi.SPI_CLK_POLARITY | 0*spi.SPI_CLK_PHASE |
                             0*spi.SPI_LSB_FIRST | 0*spi.SPI_HALF_DUPLEX)

class Test(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("spi_sma")
        self.setattr_device("led")

    def run(self):
        self.q(10000000)

    @kernel
    def q(self, n):
        self.core.reset()
        self.core.break_realtime()

        self.spi_sma.set_config_mu(flags=_DIGITAL_SERVO_SPI_CONFIG,
                write_div=8, read_div=8)
        self.spi_sma.set_xfer(chip_select=0b1, write_length=16, read_length=16)

        for i in range(n):
            self.led.on()
            self.spi_sma.write(0x8001)
            delay(20*ns)
            self.led.off()
            self.spi_sma.read_sync()
            delay(10*us)
```

dleibrandt, 2017-06-12:
Just checked 2.3, and I still see the problem.

sbourdeauducq, 2017-06-12:
Can you test with master, and post a minimal way to reproduce the problem?

dleibrandt, 2017-06-13:
Spent some time trying to come up with a simple way to reproduce the problem, and learned that it is only present when the SPI lines are connected to my slave device.  Further testing revealed that the SPI lines coming out of the kc705 were actually OK, but previously I had only measured the SPI lines after an isolator chip on the TTL breakout board.  The isolator chip outputs were dropping out because the MISO line was ringing (overshooting), causing the voltage to be outside the specs of the chip.  I fixed this by reducing the MISO drive current compliance on the slave device.

Anyway, problem solved.  Sorry about the wild goose chase.

closed 2017-06-13 by dleibrandt

jordens, 2017-06-13:
No problem. Thanks for tracking it down.