m-labs/artiq #674: Problem with Pipistrello board

trxw, 2017-02-25:
Hi everyone, 
After installation via Conda I was able to flash Pipistrello board but I keep getting the following error when running led.py (copied from artiq manual) in artiq-main directory. How could I solve this problem?

[image]

thanks

sbourdeauducq, 2017-02-26:
Did you establish the PPP link?

trxw, 2017-02-28:
Hi Sébastien, 
I have added two files to the ~/miniconda3/envs/artiq-main directory, one is device_db.pyon and another is led.py from the manual. Here's the command that I run on one tab while I run   source activate artiq-main  while in the artiq-main directory:

sudo pppd /dev/ttyUSB1 115200 noauth nodetach local nocrtscts novj 10.0.0.1:10.0.0.2

which I am guessing is PPP that you are referring to. After which I run the command artiq_run on file led.py  
However I still get the same error.
thanks again for your help.

sbourdeauducq, 2017-02-28:
The device_db.pyon and led.py files can be put in any folder, they do not need to be in the conda environment folder.

Is the PPP connection succeeding? Have you edited device_db.pyon to set the IP address of the board (10.0.0.2)?

trxw, 2017-02-28:
Thanks for your rapid answer. Yes, I set one to 10.0.0.2, and it didn't work. I also set all of them to the same address and got exact same error as before.

[image]

sbourdeauducq, 2017-02-28:
You are editing the wrong entries, you should only edit the ``comm`` entry. Can you ping the board?

trxw, 2017-03-05:
Hi Sébastien,
Thanks for your help. We got it work but I also had to replace the device_db.pyon file from miniconda3 with the respective file from conda installation, or else it wouldn't work. 
[image]

sbourdeauducq, 2017-03-05:
> I also had to replace the device_db.pyon file from miniconda3 with the respective file from conda installation

What do you mean?

sbourdeauducq, 2017-03-05:
Miniconda3 itself doesn't come with a device_db.pyon.

trxw, 2017-03-05:
I see what you're saying. I just assumed that I need to put that file in the directory where I run the script.

closed 2017-03-05 by sbourdeauducq