m-labs/artiq #675: SAWG moninj support

sbourdeauducq, 2017-02-27:


hartytp, 2018-08-11:
Is this funded at the moment? If so, I'd be interested to hear from those who funded it as to what level of moninj support they anticipate, and how much of a priority this is for them.

hartytp, 2018-08-11:
> In any reasonable implementation it is likely going to be
> resource-hungry and routing-intensive without scalable moninj (#676).

@jordens AFAICT the current scalable moninj proposal as outlined in #676 is concerned with the GUI only, not gateware optimization. How would the changes suggested in #676 help with routing and resource consumption? Or is there more to this proposal that I've missed?

hartytp, 2018-08-12:
Fine, thanks for the info.

Well, let's hear back about whether it's still a priority before we worry about the details.

jbqubit, 2018-08-15:
@hartytp Yes, this is funded. The contract calls for "SAWG monitoring and injection (sum only)." The detail is provided in the artiq-hardware writeup written with @jordens. [L195-6](https://github.com/m-labs/artiq-hardware/blob/1cec6224bfaf148cba14a5a1f69f537966d6887b/artiq_hardware.tex#L195)
```
- All spline interpolators and the total channel output shall be monitored 
by the ARTIQ channel monitoring infrastructure. 
- All spline interpolators shall support ARTIQ injection/override.
```

hartytp, 2018-08-15:
@jbqubit in the original SU-servo contract I allocated funds for moninj support. However, on more detailed discussion with @jordens, it turned out that moninj support was going to be a lot more work than I'd anticipated. Discussing this with the other ARTIQ-users at Oxford, we agreed that, given the workflow we have in our experiments, SU-servo moninj wasn't going to be particularly useful. So, we agreed to reallocate those funds to other higher priority jobs.

I wonder if the situation here is similar: moninj support for the SAWG is probably not going to be that useful in practice once we start using Sayma in real lab conditions; moninj for Sayma is going to be a lot of work (see Robert's comments above and on the mailing list); and, moninj is likely to complicate other higher-priority tasks, like getting the SAWG running at 1GSPS data rate.

Give that, would you consider dropping moninj support from Sayma (assuming @jordens and @sbourdeauducq are okay with that) and reallocating these funds to help with other, more pressing tasks? We can always come back and add moninj later on, if we really feel it would help out in the lab.

jbqubit, 2018-08-15:
The only problem I'm aware of with moninj has to do with the GUI layout. An analysis of the resource consumption of moninj in a new Issue would help clarify the situation.

hartytp, 2018-08-16:
>The only problem I'm aware of with moninj has to do with the GUI layout. 

There is likely to be a resource issue as well. cf https://ssl.serverraum.org/lists-archive/artiq/2018-August/001223.html and https://ssl.serverraum.org/lists-archive/artiq/2018-August/001227.html and other posts in that thread.

> An analysis of the resource consumption of moninj in a new Issue would help clarify the situation.

Yes. @jordens already gave his initial impressions about this. I suspect (although he should correct me if I'm wrong) that a design study to get an acuccurate estimate of resources would be quite time consuming, and probably not that much quicker than an implementation.

Before we go down that route, I want to check that we still think that moninj is a high-priority feature that needs to be implemented now. As I said, my impression from surveying the people at Oxford who've used in the lab is that it wouldn't actually be that useful. To the extent that it's likely to hamper progress on other higher-priority issues, such as the 1GSPS data rate, I'd argue that we should drop support for it, at least for the time being...

jordens, 2019-07-17:
Canceled from the ARL contract.