m-labs/artiq #676: scalable moninj

sbourdeauducq, 2017-02-27:
Currently, the dashboard displays every TTL or DDS channel available in the device database, with no way to select or reorder them. This is not nice on a large system (e.g. with DRTIO) that can have hundreds of channels.

I would propose the following:
- merge the TTL and DDS panels.
- the user selects which channels to display (they can add and remove them) from any TTL, DDS, or SAWG device.
- channels can be reordered via drag-and-drop.
- the current selection of channels, and their position, can be saved and restored in "configurations".
- there can be an arbitrary number of configurations, which are created, deleted and named by the user.

See also #603 and #675.

hartytp, 2019-01-09:
Additionally, while not explicit mentioned in @sbourdeauducq's above post, scalable moninj should also include a rethink of the low-level implementation e.g. to make it play nicer with things like the SU-servo.

sbourdeauducq, 2019-01-10:
> scalable moninj should also include a rethink of the low-level implementation e.g. to make it play nicer with things like the SU-servo.

That's just feeding the currently monitored address into the PHY so that its internal state can be mapped to block-RAM, right?

jordens, 2019-01-10:
There are several aspects for scaling up moninj. There is the **data exfiltration** aspect from the RTIO Phys:

There is no intrinsic problem with adding moninj taps to SU-servo.
It's just that doing so for all channels and all profiles is getting
towards monitoring and injecting more than a thousand values.
Monitoring/injecting the current values (current ADC value, active
profile number, rf switch/hold bits, clipping, current ASF, current
FTW) might already be very useful and is much more doable (there are
still some restrictions and interference by e.g. the global servo
enable bit) and would be significantly less complicated or expensive
resource-wise.

In general, if we can come up with a really scalable moninj (scaling
to ~10k values, update based, with the proper state management, and
with proper GUI design etc etc) that would be great for this and also
for other things like SAWG.

Then on the **protocol/networking** side:

There are probably protocols and architectures around that are
well suited and we don't need to invent something. Maybe MQTT, or one of the many other *MQ, or something like what @klickverbot did with the servos in Zurich. Such an approach
would definitely also be nice for Thermostat, Humpback and Booster. It tackles a lot of the issues that crop up with simple SCPI-like interfaces (streaming, broadcasting, multiplexing, notification/callbacks, wire format, relaying, interfaces to databases).