m-labs/artiq #679: Link Scannable LinearScan and RandomScan ranges

cjbe, 2017-03-03:
It would be useful for the LinearScan and RandomScan min/max/nsteps to be linked, so that when one changes from LinearScan to RandomScan the scan parameters are copied across.

This seems to match typical use cases better than the current behavior (RandomScan and LinearScan parameters independent, so that if you set up your scan nicely, then want to change mode, you have to set the parameters again).

sbourdeauducq, 2017-03-18:
We can rename the "Linear" scan to a "Range" scan, add a "randomize" checkbox, and remove the "Random" scan. The user experience is the same IMO and it is easier to program.

cjbe, 2017-03-18:
Agreed - this is clearly the right solution

commit 07c71bf0 (2017-03-31): language,gui: combine LinearScan and RandomScan into RangeScan. Closes #679
closed 2017-03-31 by sbourdeauducq