m-labs/artiq #683: Problems connecting logic analyzer to Pipistrello

StanleyEDavidsdavid30, 2017-03-14:
Hello!

I am a new ARTIQ user who is learning ARTIQ with Pipistrello. I am trying to run rtio dot py progam and observe the waveforms that it produces. I am using an Intronix LOGICPORT 34 channel logic analyzer. However, I am not sure how to interpret the results. One problem might be that I am running ARTIQ through a Ubuntu Linux virtual machine and having the logic analyzer software downloaded on  Windows machine. I do not believe this should be a problem though.

Can you please provide a video or instructions you used to connect logic analyzer to the Pipistrello and how to interpret the results.

Thanks!
Stanley Ebenezer David

sbourdeauducq, 2017-03-14:
You are not providing the relevant details. Is the PPP connection successful and are you running the kernel correctly?

StanleyEDavidsdavid30, 2017-03-14:
Yes, I was able to run the led dot py successfully and can control the state of the LED by entering inputs to the artiq command line

StanleyEDavidsdavid30, 2017-03-14:
So I guess this means the PPP connection is successful and I am running the kernel correctly

sbourdeauducq, 2017-03-14:
Yes. Which TTL pin are you using? If it's a bidirectional pin, you need to put it into output mode by calling ``output()`` followed by ``delay()``. Other than that, it's certainly an electrical or logic analyzer problem, check that you are connected to the right pin and try with a different measurement device (e.g. oscilloscope or even digital multimeter if you make the waveform slow enough).

StanleyEDavidsdavid30, 2017-03-14:
I am connecting to TTL0 
My issue is that I am a bit new to logic analyzers and oscilloscopes so I was wondering if you could might post a video showing how to do this connection. I think this would be very helpful for new users because generating waveforms is a common task in quantum control I guess

sbourdeauducq, 2017-03-14:
There are multiple rows of pins on the Pipistrello (A/B/C) and you also need to connect the ground.
Surely there must be someone in your group who can show you how to measure voltages?

StanleyEDavidsdavid30, 2017-03-15:
Yes I am fairly sure I have everything connected properly. So the problem is most likely due to the fact that I am using an Ubuntu Linux Virtual Machine to run ARTIQ and a Windows laptop to run the Intronix Logicport logic analyzer that I am using. The Intronix Logicport software does not seem to be supported by Linux. 

What type of logic analyzer did you use to test Pipistrello? Perhaps I will try to buy that one. I guess the best way to test is to use an oscilloscope because that is software independent, but correct me if I am wrong on this.

sbourdeauducq, 2017-03-15:
There is certainly no particular logic analyzer that I can recommend for ARTIQ, but if you are looking for a cheap one that works with Linux, you can check out Saleae. Yes, as I said, you can test ARTIQ itself with an oscilloscope, or even DMM. Or test your Intronix device with some other signal source (e.g. function generator) and fix any problems.

closed 2017-03-15 by sbourdeauducq