m-labs/artiq #684: Documentation of analyze function

r-srinivas, 2017-03-14:
If something, for example a fit function, during the `analyze` stage breaks for some reason, the data isn't saved. This problem can be solved by a try, except for example, but could this be added to the documentation? The user should be aware that if for some reason the fit function throws an exception in `analyze` after the data has been generated, the data isn't saved.

sbourdeauducq, 2017-03-27:
Changing behavior so that results are still saved when `` analyze`` fails (this seems to be a better default behavior, otherwise you could easily lose results that took a long time to collect).

closed 2017-03-27 by sbourdeauducq

jordens, 2017-03-27:
c.f. #505

r-srinivas, 2017-03-27:
Thanks! Could this be added to 2.3 as well?

sbourdeauducq, 2017-03-28:
Well, it's not a bug, and we'd like to minimize the amount of code changes on release branches.

commit 432c6b99 (2017-03-27): master: still save results when analyze fails. Closes #684