m-labs/artiq #685: TCP throughput

jordens, 2017-03-17:
... seems to be around 25 kB/s currently (rust/smoltcp). Should and could be much faster.

klickverbot, 2017-05-12:
(Commenting here, since #732 was closed.)

I've had a quick look at our specific setup with varying levels of background (broadcast) traffic and different window sizes. I don't really have the time to fix this myself right now, but here are some bullet points:
 - [ ] ACKs should be sent immediately on zero-window (or after having received two MSS-size packets). Currently, they always seem to be delayed. This is the cause for the bad performance when the window is full – e.g. when using a window only a single segment big and streaming data –, where the effective RTT becomes something like 40ms.
 - [ ] Packets with non-consecutive sequence numbers shouldn't be dropped silently, but cause a duplicate ACK so that fast retransmission can be triggered for the dropped segments. This is the cause for ~1 s pauses in some of the traces floating around here.
 - [ ] TCP reassembly would help a lot with recovering from dropped packets. Since those are going to be relatively frequent with the slow NIC, I think implementing this should be a priority after fixing the ACK issues.
 - [ ] Related to the previous points, advertising the full receive window size and leaving the limited rx ring buffer size for the remote slow start/congestion control implementation to deal with seems like the much saner choice than trying to limit the window size, as the rx buffers are shared between all traffic and should be considered part of the link.
 - [ ] ACKs with the wrong sequence number should trigger a challenge ACK instead of being dropped. This is of much lower priority than the other issues, because while it is easily possible to get reordering on saturated commodity switches, I can't think of a realistic scenario where this would seriously hurt RPC performance.

And on the MAC side of things:

 - [ ] The current rx ring buffer size of 4 is rather small for a Gb link, making it more sensitive to bursty but small broadcast traffic than necessary.

whitequark, 2017-05-12:
> ACKs should be sent immediately on zero-window (or after having received two MSS-size packets). Currently, they always seem to be delayed.

smoltcp does not implement delayed ACKs, this is an artifact of how we perform scheduling. I think this can be addressed, I'm currently trying a few approaches.

> Packets with non-consecutive sequence numbers shouldn't be dropped silently, but cause a duplicate ACK so that fast retransmission can be triggered for the dropped segments.

Correct. This is the main reason for slowness.

> ACKs with the wrong sequence number should trigger a challenge ACK instead of being dropped.

This would just be handled by the same logic as above.

> Related to the previous points, advertising the full receive window size and leaving the limited rx ring buffer size for the remote slow start/congestion control implementation to deal with seems like the much saner choice than trying to limit the window size

It *seems* like a much saner choice but after experimentation I have determined that the Linux and Widnows TCP stacks in the default configuration are unable to deal with it correctly. The current situation is far better, if flawed conceptually.
> TCP reassembly would help a lot with recovering from dropped packets. Since those are going to be relatively frequent with the slow NIC, I think implementing this should be a priority after fixing the ACK issues.

Implementing standards-compliant TCP reassembly is very tricky to do without heap allocation, especially in a space-efficient way. I've been thinking about the best way to do it, but I do not have a particularly good solution so far.

whitequark, 2017-05-12:
> The current rx ring buffer size of 4 is rather small for a Gb link, making it more sensitive to bursty but small broadcast traffic than necessary.

Indeed, I'm rather unhappy with the current MAC. It doesn't support jumbo packets (which would remove a lot of overhead in the runtime), it doesn't support MAC address filtering (which would eliminate packet loss due to broadcast traffic), and it doesn't do DMA, which limits buffer sizes by the amount of free BRAMs in gateware. MAC address filtering is perhaps the easiest to address here...

sbourdeauducq, 2017-05-13:
Theoretically, jumbo frames are easy, and would boil down to changing a few numbers in the MAC followed by the usual yak shaving.
Anyway, lwip gave out 1MB/s with the current MAC and only 2 buffers.

whitequark, 2017-05-13:
@sbourdeauducq This is just a wish, or maybe even a premature optimization. It should not be necessary to get lwip-like speeds.

sbourdeauducq, 2017-05-14:
@cjbe How was the KC705 connected to the PC when you made the dump in #731? Did it go through some network router that can fragment a packet that is too large for MTU of the outbound port?

cjbe, 2017-05-14:
@sbourdeauducq it was connected through a D-Link switch (DGS108B). I can try it with a direct connection tomorrow if that would be useful?

klickverbot, 2017-05-14:
> smoltcp does not implement delayed ACKs, this is an artifact of how we perform scheduling.

Makes sense, thanks – I _was_ wondering where the delay came from, since I couldn't see anything obvious in the stack itself.

> It seems like a much saner choice but after experimentation I have determined that the Linux and Widnows TCP stacks in the default configuration are unable to deal with it correctly. The current situation is far better, if flawed conceptually.

Fair enough. I wonder whether the outcome would change with "proper" ACK handling and out-of-order support, though.

> Implementing standards-compliant TCP reassembly is very tricky to do without heap allocation, especially in a space-efficient way. I've been thinking about the best way to do it, but I do not have a particularly good solution so far.

It should be perfectly workable to keep a free list backed by a static pool around to store per-segment metadata (sequence number ranges, …), while building up the payload directly in the circular buffer. The tricky bit seems to be to make sure to get the various state transitions/edge cases right, not the method used for memory allocation (given that you can always just bail and revert to dropping the packet). What am I missing?

You don't even need to support IP reassembly; a simple queue of incoming segments would be enough for this to be a huge improvement. This might not qualify as a "particularly good solution" according to your criteria, but a few-entry queue should get the job done admirably for our use case.

If you have any patches to test on our network environment(s), just ping me and I'll be happy to oblige.

cjbe, 2017-05-15:
@sbourdeauducq [here](https://github.com/m-labs/artiq/files/1000670/dump_directly_connected.pcap.zip) is a packet dump with the KC705 directly connected to the PC - it looks qualitatively similar to the dump through the router.

sbourdeauducq, 2017-05-22:
Tentative patch by @whitequark:
```
diff --git a/src/socket/tcp.rs b/src/socket/tcp.rs
index f0ab25f..164faae 100644
--- a/src/socket/tcp.rs
+++ b/src/socket/tcp.rs
@@ -717,6 +717,7 @@ impl<'a> TcpSocket<'a> {
             (_, TcpRepr { seq_number, .. }) => {
                 let next_remote_seq = self.remote_seq_no + self.rx_buffer.len();
                 if seq_number > next_remote_seq {
+                    self.retransmit.reset();
                     net_trace!("[{}]{}:{}: unacceptable SEQ ({} not in {}..)",
                                self.debug_id, self.local_endpoint, self.remote_endpoint,
                                seq_number, next_remote_seq);
```

sbourdeauducq, 2017-06-25:
@klickverbot @cjbe @dhslichter should be better now.

whitequark, 2017-06-26:
> It should be perfectly workable to keep a free list backed by a static pool around to store per-segment metadata (sequence number ranges, …), while building up the payload directly in the circular buffer.

Yeah, this is how I would also go at implementing this. However this is a rather nontrivial chunk of work.

> Packets with non-consecutive sequence numbers shouldn't be dropped silently, but cause a duplicate ACK so that fast retransmission can be triggered for the dropped segments. This avoids the ~1 s pauses in some of the traces floating around here.

I've implemented duplicate ACK responses in smoltcp. This improves things considerably. The following test:

```python
from artiq.experiment import *

class Transfer(EnvExperiment):
    def build(self):
        self.setattr_device("core")

    def run(self):
        print(self.transfer(10**6), "bps")

    def get(self, n) -> TBytes:
        return b"\x00"*n

    @kernel
    def transfer(self, n):
        t0 = self.core.get_rtio_counter_mu()
        self.get(n)
        t1 = self.core.get_rtio_counter_mu()
        return n/self.core.mu_to_seconds(t1-t0)
```
running over the LAN reports speeds of at least 1 Mbps, and up to 1.8 Mbps. I believe this can be further improved, but this should already be acceptable.

whitequark, 2017-06-26:
Here are the TCP throughput graphs for the test above.

First for a capture with 1.1 MB/s rate (which is consistently reproducible):
[image]

Second for a capture with 1.8 MB/s rate (ideal conditions):
[image]

Neither has any retransmissions so I'm not sure exactly where the difference comes from. 
[Pcap files](https://github.com/m-labs/artiq/files/1101702/throughput_out.zip)

whitequark, 2017-06-26:
@klickverbot 

> advertising the full receive window size and leaving the limited rx ring buffer size for the remote slow start/congestion control implementation to deal with seems like the much saner choice than trying to limit the window size

I re-checked this again after implementing duplicate ACK responses. This doesn't work. The core device ends up sending exactly thirty eight (amusingly it's very consistent) duplicate ACKs in response to the host pushing the window, and this drops throughput to less than 100 kB/s. 
[See for yourself](https://github.com/m-labs/artiq/files/1101733/thirty_eight_acks.zip)

jordens, 2017-06-26:
The two pcaps (1.1M and 1.8M) are exactly the same.

whitequark, 2017-06-26:
> The two pcaps (1.1M and 1.8M) are exactly the same.

Hm, wireshark has messed up filename management.

whitequark, 2017-06-26:
@jordens Packet captures reacquired and comment above updated.

klickverbot, 2017-06-26:
@sbourdeauducq: It wasn't better at the time of your message, but the duplicate ACK handling from earlier today should indeed make a difference. I'll have a go at reproducing the results soon.

@whitequark: Thanks for the details on the window size test, the results are quite interesting to see. I didn't have a look at the capture yet – so this is just empty guesswork – but note that one of the core assumptions made in designing and implementing TCP congestion control is that the receiving side can actually handle data that fits inside the advertised receive window. Without reordering implemented, this is obviously not the case (hence also why the point about window sizes was below the one on reordering on my original list).

cjbe, 2017-06-27:
@whitequark when I run your transfer experiment I see different results - the transfer hangs (does not complete within 200s). I am using the current software and gateware from conda (3.0.dev+1186.gf36f00a8).

Here is a [packet dump](https://github.com/m-labs/artiq/files/1106660/dump.zip)

I see the following in the core log:
```
[    41253255us]  WARN(runtime): network error: dropped by socket
[    67429376us]  WARN(runtime): network error: dropped by socket
[   119845511us]  WARN(runtime): network error: dropped by socket 
[   224549696us]  WARN(runtime): network error: dropped by socke
[   344614028us]  WARN(runtime): network error: dropped by socket
```

whitequark, 2017-06-27:
@cjbe Ok, that's a different issue. I believe I have already found it while auditing the TCP packet handler for correctness and am already working on it.

To make sure I can reproduce the issue you observe, can you reproduce it after invoking `artiq_corelog set_level TRACE` and then post the more verbose log?

cjbe, 2017-06-27:
@whitequark here is a verbose core log and packet dump:
[corelog.txt](https://github.com/m-labs/artiq/files/1106753/corelog.txt)
[dump.zip](https://github.com/m-labs/artiq/files/1106754/dump.zip)

whitequark, 2017-06-27:
@cjbe Oh. I'm really sorry. I did not update ARTIQ so that it uses the version of smoltcp with the bug fixed. It is building now.

whitequark, 2017-06-28:
@cjbe You can update ARTIQ and try again now.

cjbe, 2017-06-28:
@whitequark using 3.0.dev+1186.gf36f00a8 I see a great improvement in speed for 'Transfer' (0.8 MB/s - 1.8 MB/s) but I still see 'Transfer' getting stuck sporadically (every 1-5 runs).

I scheduled 5 'Transfer' experiments in a row. The first two ran to completion, with transfer rates 1.26 MB/s and 0.76 MB/s. The third one got stuck with what looks like a similar pattern to before.

[corelog.txt](https://github.com/m-labs/artiq/files/1108293/corelog.txt) (TRACE log level)
[dump.zip](https://github.com/m-labs/artiq/files/1108292/dump.zip)

whitequark, 2017-06-29:
@cjbe This is a new problem, not seen before. I believe I have identified the cause and am working on a fix.

cjbe, 2017-07-09:
@whitequark any update on this?

sbourdeauducq, 2017-08-07:
@whitequark ping

whitequark, 2017-08-25:
@cjbe Can you recheck with ARTIQ git hash 7c340f6b?

whitequark, 2017-08-25:
Ah wait, found a minor issue with it.

whitequark, 2017-08-30:
This is finally fixed! :confetti_ball: 

I very consistently get ~300 kB/s with the latest master, measured at the same LAN as the core device:
```
$ artiq_run test_thru1.py
291882.2459315135 B/s
$ artiq_run test_thru1.py
291765.49065326096 B/s
$ artiq_run test_thru1.py
291893.81590445706 B/s
$ artiq_run test_thru2.py
283804.8550823502 B/s
$ artiq_run test_thru2.py
283803.68106043333 B/s
$ artiq_run test_thru2.py
283496.9025411925 B/s
```

dhslichter, 2017-08-30:
This is nice to have the improvement, thanks @whitequark.  However, this is still very slow in the grand scheme of things, especially if we are going to larger Sinara setups.  It seems that we should be aiming to go at least an order of magnitude faster (if not more) once we are working with Metlino cards driving multiple uTCA crates etc -- complex experiments, with all the attendant parameterized waveforms, plus the possibility of a lot of data needing to be RPC'ed back to the PC.  It would be a big problem if all the fancy hardware were laid low by a slow TCP connection to the core device.

whitequark, 2017-08-30:
@dhslichter I agree. This is actually slower than the results I've presented earlier in this thread.

What I find encouraging is that unlike the previous problem ("the TCP stack completely breaks the expectations of the peer and destroys throughput"), the new problem ("the TCP stack is not tuned quite optimally") is much more tractable, and shouldn't require anywhere as much work.

I'm not yet sure why it's three times slower yet, I've just started investigating it.

dhslichter, 2017-08-30:
Sounds good @whitequark, and I am glad that you have managed to improve the reliability here (I do recall your reporting better numbers before, but with consistency/reliability issues).  Definitely seems like tuning the TCP stack to optimize throughput will be a more straightforward task.  Anyway, my main message is "good work!" :)

whitequark, 2017-08-31:
@dhslichter You wanted an order of magnitude faster? Well, how about ~2.2 MB/s? The throughput graph looks like this now. I believe it's limited by TCP Slow Start at the very beginning:
[image]

Note that this is only host to core device though, core device to host is still 300 kB/s. I don't know why, everything seems perfectly normal both in packet dumps and TCP debug log, except that occasionally the core device stops sending packets for 150ms. (No retransmits.) I think it might be an artifact in the test.

cjbe, 2017-08-31:
@whitequark I just tried 3.0.dev+1303.gc5fe2799 and I still see very low throughput with continuously increasing retransmit delays.

Here is the [packet dump](https://github.com/m-labs/artiq/files/1267380/dump.zip) from running transfer.py (which did not complete before I got bored).

cjbe, 2017-08-31:
Ooops - that build (which conda install gave me) was from the sinara branch, hence does not have the latest fixes. Despite that, using 3.0.dev+1280.g20f43d57 I see the same problem. 
[dump.zip](https://github.com/m-labs/artiq/files/1267455/dump.zip)

whitequark, 2017-08-31:
@cjbe What is the OS used on the host machine?

cjbe, 2017-08-31:
Linux (3.19.0-84-generic)

jordens, 2017-08-31:
@whitequark Good work on smoltcp!

But (assuming RTT is in the ms range and you are testing this locally) I would not expect any of those 200 ms quiet periods in there. They look like some retransmission timeout. And from what I have seen in on other TCP those are usually not encountered at this stage.
Those long timeouts might not affect sustained transfer speed. But for the small transfer (kernels << 1 MB) that we are doing they are critical.

whitequark, 2017-08-31:
@jordens No timeouts (those go into the log at WARN level, easily noticeable). The quiet periods are all on the host side. One of them is the compiler latency, the other I'm not sure.

whitequark, 2017-08-31:
@jordens Actually, I'm not sure why the graph above is so pessimistic. Here I ran three kernels in rapid succession. It takes about 9ms from the end of one kernel to the start of another. No 200 ms quiet periods:
[image]

jordens, 2017-08-31:
@whitequark ack. i thought they were raw transfers with fast source and and discard sink.

whitequark, 2017-08-31:
> Note that this is only host to core device though, core device to host is still 300 kB/s. [...] I think it might be an artifact in the test.

I have determined that this is an artifact in the test. If, instead of an RPC, I directly transmit a large chunk of data (e.g. core log or analyzer buffer) then the throughput is also ~2.2 MB/s. Of course, this is still a problem, but it's not a problem with the TCP/IP stack.

jordens, 2017-09-01:
[image]

dhslichter, 2017-09-01:
@whitequark interesting.  Definitely will need to see why the RPC does so much worse.  And good to see that rates >1 MB/s are appearing now.  The throughput and latency issues with the host-core device communication is probably the most major roadblock to switching to ARTIQ 3 for us in the lab right now, so keep us posted on how the debugging progresses.

whitequark, 2017-09-22:
I've implemented reassembly of out-of-order segments in smoltcp.

klickverbot, 2017-09-22:
@whitequark: For a simple test kernel that just sends 1 MB of zeros back and forth, I am now seeing about 18 MB / s (!) from master to kernel, and about 19 MB / s from kernel to master (same Linux 3.19.0-84-generic system as earlier). This almost seems too good to be true – if those numbers hold up in real-world usage, this is definitely good enough for us in the short term. Long-term, we would like to get close to GbE line speeds, but this will of course involve more than just a few tweaks to the TCP stack (it might not even be doable with the current SoC configuration).

Having a quick look at the package captures, there still appear to be more duplicate ACKs than expected in the steady state (with https://github.com/m-labs/smoltcp/commit/64369d9801ef10594476503de02caa10d68cdc24), but since the core device receive performance is almost identical to the other direction where Linux just gobbles up packets at full speed, I didn't have a closer look at it yet. 

Anyway, great work!

sbourdeauducq, 2017-09-23:
Great work @whitequark ! I get 2.3MB/s.
@klickverbot can you share your source code that displays 18MB/s?

closed 2017-09-23 by sbourdeauducq

whitequark, 2017-09-23:
@klickverbot Or packet captures, in both directions, I'd be quite curious to look at the duplicate ACKs.

klickverbot, 2017-09-23:
That machine seems to have crashed, it will take me a while to get back to the lab to revive it. But by now, I suspect there might be some funny business going on with the clocks in the firmware build I was using, as I was using the core device RTIO counter for timing.

cjbe, 2017-09-23:
@whitequark there was indeed a factor 8 error in the numbers from @klickverbot : we were seeing about 2.4 MB/s in the above tests, not 18 MB/s. Here is a [dump](https://github.com/m-labs/artiq/files/1326555/dump.zip) of a 1 MB master -> kernel transfer showing duplicate acks, then a 1 MB kernel -> master transfer.

This is all with the core device connected through a 100M switch - I will try it with a 1G switch on Monday.

klickverbot, 2017-09-23:
Indeed – while I didn't mix up bits and bytes, the code mistook eight nanoseconds for just one due to a mismatching device db, as Chris pointed out. At least you now know what I would consider acceptable transfer speeds. ;)

@cjbe: Your dump presumably didn't include https://github.com/m-labs/smoltcp/commit/64369d9801ef10594476503de02caa10d68cdc24? Or was this with my last firmware build?

I've also resurrected the test box and tried swapping out the 100M switch to our Netgear ProSafe GbE switch again. The extreme packet loss issue still persists (less than 1 % ICMP pings make it through, with no networking-related messages at TRACE log level). I (or Chris) will give it another go on Monday with a stock gateware build and open a separate issue to keep track.

cjbe, 2017-09-23:
@klickverbot my dump was from the firmware build you put on the board