m-labs/artiq #686: track performance

jordens, 2017-03-17:
Track the long term performance of ARTIQ components and alogrithms.

* TTL, DDS, SPI, SAWG event throughput, input, outputs.
* FP performance.
* connection latency, RPC latency, RPC throughput.
* worker startup, kernel compilation, linking, kernel upload.

https://irclog.whitequark.org/m-labs/2017-03-17#1489752484-1489752048;

jbqubit, 2017-03-17:
Thank you for advocating for this @jordens. Much needed. Add TCP throughput to the list #685.

whitequark, 2017-03-17:
That's RPC throughput in the list.

jordens, 2017-05-11:
@whitequark Now that we have PCUs, would it be quick to have another stab at trying #667?

whitequark, 2017-05-11:
It certainly won't be quick as I don't have an easy way to reproduce the issue. But it will be possible.

lriesebos, 2021-05-05:
I randomly stumbled on this issue. We currently do not track performance with steady time intervals, though we might start doing that later this summer. We do sometimes take performance samples with benchmarks we wrote ourselves. See https://gitlab.com/duke-artiq/dax/-/wikis/ARTIQ/Core-device-hardware