m-labs/artiq #687: support for Zotino DAC and Novogorny ADC

sbourdeauducq, 2017-03-23:
* kernel functions to read/write samples using RTIO SPI
* DMA supported for DAC
* monitoring supported for DAC

https://github.com/m-labs/sinara/wiki/Zotino
https://github.com/m-labs/sinara/wiki/Novogorny

Funded by Oxford University and Freiburg University.

sbourdeauducq, 2017-11-02:
Zotino done including monitoring. For Novo there now is a generic driver for shift registers that can be used for the PGIA. Talking to the main ADC chip still needs to be done.

jordens, 2018-02-22:
Novogorny support is done. RTIO, SPI, Gain shift register, conversion functions, documentation.

hartytp, 2018-02-22:
@jordens Novogorny/Novo has been obsoleted by Sampler, so can we rename this in the code as well, please? I would like the naming in ARTIQ to match the front panels/schematics/Wiki!

hartytp, 2018-02-22:
Other than that, looks good! Can't wait to try this out once the hw arrives!

jordens, 2018-02-22:
No. Novogorny stays. We are using that hardware.

hartytp, 2018-02-22:
How much of it do you have? Are you planning to buy more? Are you planning to use Novogorny and Sampler? We can chat about this offline, but since we funded it, I would like to have the driver/docs match the PCB name we chose.

jordens, 2018-02-22:
We are using two Novogorny v1.0 because we needed them. That means we'll keep it for now. Sampler is different enough to get its own codebase. And since almost everything on that wiki page refers to Novogorny, you may want to rename it back to Novogorny and create a page for Sampler ;)
Alternatively if we can swap those Novogornys for Samplers I'd be happy to dump the Novogorny code.

jordens, 2018-02-22:
I don't see the problem with consistency. Novogorny refers to Novogorny v1.0 and Sampler refers to Sampler v1.1. Treat them as different boards (they are). There is no confusion.

hartytp, 2018-02-22:
Well, if you want to add another driver for Sampler then I'm not fussed if you keep Novogorny as well.

But, I would certainly not advise anyone to buy more Novogorny now -- the cost should be roughly the same as Sampler and there shouldn't be any situation where Novogorny gives an advantage over Sampler; it's the same just slower.

Generally, I'm strongly of the opinion that v1.0 was just prototype hardware that should be obsoleted once the production hardware is available. Maintaining multiple code + support for prototype hardware is a waste of time.

I certainly also wouldn't support having a Wiki page for Novogorny -- almost no one has one, and no one should be buying more.

jordens, 2018-02-22:
It's an investment. There is no point in trashing it. And there is very little wrong with Novogorny. And it's just slightl slower (if at all) for non-NU-Servo usage. If there are issues with Sampler then Novogorny will be even more relevant.

You should really rename that wiki page from Sampler back to Novogorny. It is confusing and misleading people who are buying this.

hartytp, 2018-02-22:
> We are using two Novogorny v1.0 because we needed them.

At least one of those is ours IIRC, I'm happy to recall that one if it simplifies this discussion.

>  And there is very little wrong with Novogorny.

@jordens It also has a few smaller bugs which were never fixed. But, yes, it basically works fine.

> If there are issues with Sampler then Novogorny will be even more relevant.

If there are issues with Sampler, we'll fix them.

> You should really rename that wiki page from Sampler back to Novogorny. It is confusing and misleading people who are buying this.

No. Firstly, I'm asserting our right as the people who funded this and lead the work to choose the name. That right is a basic given in any of these things (you'll note that I've never tried to rename Kasli). There is only one project here, just two versions -- if we were to fix the bugs on Novogorny and produce more then I'd still want it renamed to Sampler v1.0. 

Secondly, I do not expect anyone to buy Novogorny. Why waste time supporting two boards which are nearly identical when we only need to support one? Not to mention that producing two boards makes both of them more expensive due to the decreased bulk-buy saving. Assuming there are no unexpected issues with Sampler, under what circumstances would it be better for someone to buy Novogorny than Sampler? Assuming the answer is "none" why would we have a Wiki page to it when only 5 boards were ever made?

jordens, 2018-02-22:
I don't think a discussion about "rights" is helpful here. It would involve talking about the right to the photos, to the analysis etc. I also see no reason why you would (or could) demand that we change the name for the code we write for the hardware we use. Especially since it seems like you don't want anything to do with it.
Recalling the one Novogorny loaner would be destructive. I don't see why you would do that other than to cause problem.

jordens, 2018-02-22:
The reason for having a wiki page and supporting is simply that those boards are being used and that there is funding for the code.

dtcallcock, 2018-02-22:
> I certainly also wouldn't support having a Wiki page for Novogorny -- almost no one has one, and no one should be buying more.

I think there is a more general problem here -- for someone new to Sinara, the Github wiki is pretty obscurely located and impenetrable.  

Perhaps once there is a reasonable amount of hardware available to purchase, there could be a high-level Sinara webpage (like artiq has on the M-Labs website)?  This could have an introduction, the formal documentation for released boards and purchasing information.

jordens, 2018-02-22:
IMHO the challenge is content quality and responsibility/accountability. Whether it's a wiki on Github or something else is very much secondary. And is the content is good, the Wiki is a fine platform.

dtcallcock, 2018-02-22:
> Whether it's a wiki on Github or something else is very much secondary

I was trying to give an obvious example of where someone with 'responsibility/accountability' over the Sinara project (or at least the documentation of it) could do a better job of presenting and promoting it to the wider community.

Has any thought been given to implementing some sort of accountability structure outside of people just calling the shots on their own bit of turf that they funded/contributed?  Perhaps it's too much overhead at this early stage or people's strongly-held philosophies on such matters will be too divergent.

jordens, 2018-02-22:
The idea of a ARTIQ/Sinara "board"/"council"/"committee" was aired a couple of times. IMHO it's a good idea; it wouldn't be too complicated. And we could take lessons from the governance structure of other projects. Debian and others come to mind.
But yeah. Overhead and the conflict between the contractual relation and project governance.

hartytp, 2018-02-22:
@jordens to be clear, my expectations are:
1. That there will be a Sampler ARTIQ driver as well/instead of the Novogorny driver
2. That Novogorny will not be advertised to new users as a product that they should consider buying.

Beyond that, I'm not fussed.

In detail:
1. Correct me if I'm wrong, but my memory is that we agreed to change the Novogorny contract to target Sampler. If you've got additional funding for an ARTIQ Novogorny driver that's fine by me, but I do expect the Sampler driver.
2. Novogorny v1.0 still has bugs that require manual modifications to each PCB. There is no point in anyone putting the effort in to fix that when it's superseded by Sampler.
3. Novogorny isn't expected to provide any performance advantage over Sampler; it's just slower. So, there is no reason anyone should choose to buy it over Sampler once that's finished. The cost difference between the two boards is small compared with the impact of bulk-buying so it's cheaper to only have one design.
4. If you have concerns about Sampelr then I'm happy to more forwards with testing of that faster (it will only take me a day to reproduce all the test that Greg did on Novogorny). If we find issues with that then we will fix them to ensure that Sampler's performance is not worse than Novogorny's.
5. Supporting both Novogorny and Sampler is likely to mean that both are more expensive and less well supported. I want to avoid that. That's my reason for wanting to kill Novogorny.
6. Since only a handful of Novogornys have been produced it's not worth having a Wiki page for it. Having a Wiki page suggests to new Sinara users that Novogorny is a product that is actively supported that they should consider buying, IMO that's not the case.
7. If the only argument for supporting Novogorny is that there are currently a handful that have been produced then I'll happily recall the ones we paid for to nix that argument. In any case, it would be useful to have some more ADCs around the lab for non-ARTIQ use.

hartytp, 2018-02-22:
>I think there is a more general problem here -- for someone new to Sinara, the Github wiki is pretty obscurely located and impenetrable.

> Perhaps once there is a reasonable amount of hardware available to purchase, there could be a high-level Sinara webpage (like artiq has on the M-Labs website)? This could have an introduction, the formal documentation for released boards and purchasing information.

It's a nice idea, the problem is that writing good documentation is extremely dull and time consuming, so it's hard to get anyone to actually do it. FWIW, my feeling is that we need someone who has a financial incentive to write this all up properly. eg if Creotech or someone else want to become a supplier to the wider AMO community then it would be in their interest to write some glossy documentation to advertise Sinara. Short of that, I think it will be hard to convince anyone else to do this.

There is a Sinara issue about updating the Wiki that aims to address this to a certain extent, so maybe better to discuss this there?

jordens, 2018-02-22:
@hartytp Exactly. Nobody said that the Novogorny driver would replace the Sampler driver. There is thus no reason to bad-mouth Novogorny or the driver. Claiming that the presence of a Novogorny driver makes Sampler more expensive or the other way around is absurd. And I don't need to remind you that also for Sampler there are a similar number of manual fixes done to the boards. The reason for a wiki page is not purely advertisement and lobbying as you describe it, it is also about supporting and documenting hardware that is in use. And I bet Creotech won't do the documentation you are looking for.

hartytp, 2018-02-23:
> Claiming that the presence of a Novogorny driver makes Sampler more expensive or the other way around is absurd. 

I never said that. I'm not fussed if you want an ARTIQ driver for it. I just don't want Novogorny advertised as a product that people should consider buying. 

>  And I don't need to remind you that also for Sampler there are a similar number of manual fixes done to the boards. 

Yes, neither are production ready boards. The difference is that we plan to fix the issues with Sampler and go into production mode for it. That's not the case with Novogorny.

>  reason for a wiki page is not purely advertisement and lobbying as you describe it, it is also about supporting and documenting hardware that is in use. 

Having Wiki pages devoted to legacy products with only a couple in existence does more harm than good. It's confusing to new users who are trying to figure out what hardware is actively supported and produced.

If you want to keep the couple of Novogorny you have and support a drive for them then that's fine. But, I don't think that warrants a page on the project Wiki. I'd rather recall Novogorny v1.0 than clutter the Wiki with pages describing it.

hartytp, 2018-02-23:
> And I bet Creotech won't do the documentation you are looking for.

Well, maybe not, but if they don't then who will? The current lack of documentation is definitely off putting to new users and I don't see anyone stepping up to fix that.

It's somewhat similar with ARTIQ. There is a general lack of documentation for the overall stack (misoc/migen/etc) and the internal workings of ARTIQ, which makes it more difficult to contribute. But, beyond that, last time I checked there was no documentation for things like the applet interface, which are pretty significant user-facing parts of ARTIQ.

Anyway, this has veered wildly off topic, and I've said all I have to say on the matter.

sbourdeauducq, 2018-02-23:
> I never said that. I'm not fussed if you want an ARTIQ driver for it. I just don't want Novogorny advertised as a product that people should consider buying.

Speaking of rights, it was discussed that those projects should be under CERN OHL, so if someone wanted to fork Sampler, put a different DAC and call it Novogorny, that would be allowed. Unless you have trademark rights for "Novogorny", but considering your distaste for this naming scheme, I doubt you would be interested in that :)

hartytp, 2018-02-23:
@sbourdeauducq "rights" not in a legal sense, but more in an etiquette/not being a jerk sense. Like, you don't rename a boat once you buy it. Nothing is stopping you, it's just not productive to go around making trivial changes to other people's work and then changing the name.

Sometimes I do wonder whether it would be less painful to bang my head against a wall until HDL comes out than to deal with you guys.

hartytp, 2018-02-23:
I actually sometimes look back nostalgically at the days when my experimental code was written in turbo pascal and labview, but I didn't have to deal with all this BS.

whitequark, 2018-02-23:
> Like, you don't rename a boat once you buy it.

I thought it was you who renamed Novogorny to Sampler in the first place.

hartytp, 2018-02-23:
> I thought it was you who renamed Novogorny to Sampler in the first place.

sigh...(a) we provided the funding and did the bulk of the design work, so it was our decision (b) Novogorny was a working title that we adopted for the prototypes as a placeholder until we thought of a better name -- the boat hadn't been christened at that point. 

This is silly. I give in.

jordens, 2018-02-23:
@hartytp 

> > > Supporting both Novogorny and Sampler is likely to mean that both are more expensive

> > Claiming that the presence of a Novogorny driver makes Sampler more expensive or the other way around is absurd.

> I never said that.

Your complaining about the documentation is as annoying as other people complaining about hardware or silicon quality. It's not helping. If nobody has the resources (funding, commercial incentive, or free time) then who do you think will write the documentation? If Creotech or we are supposed to write the kind of documentation you are looking for, you already know exactly what needs to be done.

I'll let the other provocations pass silently.

jordens, 2018-03-21:
Sampler is done. 97918447a3426c39f23d3766049c7fb644c0a33e

jordens, 2018-03-24:
Zotino done as well a992a672 up to a20dfd9c

closed 2018-03-24 by jordens