m-labs/artiq #692: Method to set default experiment priority / log level / pipeline

cjbe, 2017-03-27:
It would be really useful to have a method that can be called from build that sets the default priority for the experiment. It would potentially also be useful to be able to set a default log level and pipeline. This default would be used when the experiment is first opened in a GUI, or used if the user clicks "recompute all arguments".

An example where this is useful is for servo experiments. One often has 3 or so servo experiments that interrupt the running experiment to tweak parameters, with the servos each running at different priorities so they interrupt each other appropriately. At the moment the user needs to set these experiments to have the correct priority in each instance of the GUI they use - this is fragile.

sbourdeauducq, 2017-07-20:
@cjbe Would Oxford fund this?

cjbe, 2017-07-20:
@sbourdeauducq it might be possible to tack this on to a future contract, else one of us will implement it.

PR #1262 (2019-03-12): Add methods to set experiment pipeline/priority/flush defaults

# ARTIQ Pull Request

## Description of Changes
Add methods callable from experiment build() to set pipeline/priority/flush defaults.
Suggestions on a cleaner way of indicating the functionality of the "recompute scheduler options" button without bloating the GUI appreciated.
Closes #692

## Type of Changes

|   | Type |
| ------------- | ------------- |
| ✓  | :sparkles: New feature |

## Steps (Choose relevant, delete irrelevant before submitting)

### All Pull Requests

- [x] Use correct spelling and grammar.
- [x] Update [RELE [... cut]
closed 2019-03-12 by jordens