m-labs/artiq #700: DMA sequence error that should not happen

sbourdeauducq, 2017-03-31:
```
from artiq.experiment import *

class DMABlink(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("core_dma")
        self.setattr_device("led")

    @kernel
    def record(self):
        with self.core_dma.record("blink"):
            for i in range(5):
                self.led.pulse(100*ms)
                delay(250*ms)
            for i in range(5):
                self.led.pulse(50*ms)
                delay(150*ms)

    @kernel
    def run(self):
        self.core.reset()
        self.record()
        self.core.break_realtime()
        for i in range(5):
            self.core_dma.replay("blink")
```

```
$ artiq_run.py ./repository/coredevice_examples/simple/dma_blink.py
WARNING:artiq.coredevice.comm_kernel:Mismatch between gateware (3.0.dev+960.g28211e0) and software (3.0.dev+948.g6caab4d.dirty) versions
Core Device Traceback (most recent call last):
  File "./repository/coredevice_examples/simple/dma_blink.py", line 26, in artiq_run_dma_blink.DMABlink.run(..., ...) (RA=+0x8dc)
    self.core_dma.replay("blink")
  File "<artiq>/coredevice/dma.py", line 87, in __modinit__ (inlined)
    dma_playback(now_mu(), name)
  File "/var/lib/buildbot/slaves/debian-stretch-amd64-2/miniconda/conda-bld/artiq-kc705-nist_clock_1490946366711/work/artiq/firmware/ksupport/lib.rs", line 358, column 13, in (Rust function)
    <unknown>
                ^
artiq.coredevice.exceptions.RTIOSequenceError(0): RTIO sequence error at 1080980977408 mu, channel 19
Traceback (most recent call last):
  File "/home/sb/artiq_drtio/artiq/frontend/artiq_run.py", line 223, in <module>
    main()
  File "/home/sb/artiq_drtio/artiq/frontend/artiq_run.py", line 219, in main
    return run(with_file=True)
  File "/home/sb/artiq_drtio/artiq/frontend/artiq_run.py", line 205, in run
    raise exn
  File "/home/sb/artiq_drtio/artiq/frontend/artiq_run.py", line 198, in run
    exp_inst.run()
  File "/home/sb/artiq_drtio/artiq/language/core.py", line 53, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "/home/sb/artiq_drtio/artiq/coredevice/core.py", line 124, in run
    self.comm.serve(embedding_map, symbolizer, demangler)
  File "/home/sb/artiq_drtio/artiq/coredevice/comm_kernel.py", line 589, in serve
    self._serve_exception(embedding_map, symbolizer, demangler)
  File "/home/sb/artiq_drtio/artiq/coredevice/comm_kernel.py", line 581, in _serve_exception
    raise python_exn
artiq.coredevice.exceptions.RTIOSequenceError: RTIO sequence error at 1080980977408 mu, channel 19
```

This started occuring after 28211e0b32f92e73a650364f129ce4cc9ba7aac2.

sbourdeauducq, 2017-03-31:
```
            t1 = now_mu()
            self.core_dma.replay("blink")
            t2 = now_mu()
            print(t2-t1)
```
incorrectly prints 0.

whitequark, 2017-04-01:
@sbourdeauducq This is not a bug per se, rather this is how I understood the specification.

1. Do I understand it right that `replay` should advance the timeline?
2. What is the length of a replay, then? Is it measured through the `now` difference in `record`, or from the actual recorded messages?

sbourdeauducq, 2017-04-01:
> Do I understand it right that replay should advance the timeline?

Yes.

> What is the length of a replay, then?

It should be as if the content of the ``with record`` had been inserted instead of the ``replay``, in this case 5*(100+250)+5*(50+150) ms. In other words, value of ``now`` at the end of the ``with`` block, minus value at the beginning. How did you understand the specification otherwise?

commit 14ae1cc1 (2017-04-05): runtime: advance now on DMA replay.

Fixes #700.
closed 2017-04-05 by whitequark