m-labs/artiq #701: conda leaves build files around

sbourdeauducq, 2017-04-01:
After a build, files are left over in ``/var/lib/buildbot/slaves/debian-stretch-amd64-2/miniconda/conda-bld``. This wastes disk space, and more importantly randomly breaks builds when the file names conflict, resulting in errors like:
``"fatal: destination path '/var/lib/buildbot/slaves/debian-stretch-amd64-2/miniconda/conda-bld/artiq-kc705-nist_clock_1491023827445/work' already exists and is not an empty directory.``

sbourdeauducq, 2017-04-01:
Deleting all this trash before a build isn't always enough to prevent the error, so there is another obscure conda problem here.

sbourdeauducq, 2017-04-12:
"Source and build intermediates have been left in /var/lib/buildbot/slaves/debian-stretch-amd64-1/miniconda/conda-bld.
There are currently 172 accumulated.
To remove them, you can run the ```conda build purge``` command"

whitequark, 2017-04-19:
I believe I've fixed this, by replacing our homegrown cache buster with `conda build clean`.

closed 2017-04-19 by whitequark