m-labs/artiq #703: unification issue

jordens, 2017-04-07:
Shouldn't the inferencer be able to cope with this?

```python
from artiq.experiment import *

class Test(EnvExperiment):
    def build(self):
        self.setattr_device("core")

    @kernel
    def run(self):
        delay_mu(1 + 0)
```

```python
Traceback (most recent call last):
  File "/home/rj/src/conda/envs/py35/bin/artiq_run", line 11, in <module>
    load_entry_point('artiq', 'console_scripts', 'artiq_run')()
  File "/home/rj/work/nist/artiq/artiq/frontend/artiq_run.py", line 219, in main
    return run(with_file=True)
  File "/home/rj/work/nist/artiq/artiq/frontend/artiq_run.py", line 205, in run
    raise exn
  File "/home/rj/work/nist/artiq/artiq/frontend/artiq_run.py", line 198, in run
    exp_inst.run()
  File "/home/rj/work/nist/artiq/artiq/language/core.py", line 53, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "/home/rj/work/nist/artiq/artiq/coredevice/core.py", line 115, in run
    self.compile(function, args, kwargs, set_result)
  File "/home/rj/work/nist/artiq/artiq/coredevice/core.py", line 95, in compile
    attribute_writeback=attribute_writeback)
  File "/home/rj/work/nist/artiq/artiq/compiler/module.py", line 69, in __init__
    inferencer.visit(src.typedtree)
  File "/home/rj/src/conda/envs/py35/lib/python3.5/site-packages/pythonparser/algorithm.py", line 41, in visit
    return self._visit_one(obj)
  File "/home/rj/src/conda/envs/py35/lib/python3.5/site-packages/pythonparser/algorithm.py", line 34, in _visit_one
    return self.generic_visit(node)
  File "/home/rj/src/conda/envs/py35/lib/python3.5/site-packages/pythonparser/algorithm.py", line 27, in generic_visit
    self.visit(getattr(node, field_name))
  File "/home/rj/src/conda/envs/py35/lib/python3.5/site-packages/pythonparser/algorithm.py", line 39, in visit
    return [self.visit(elt) for elt in obj]
  File "/home/rj/src/conda/envs/py35/lib/python3.5/site-packages/pythonparser/algorithm.py", line 39, in <listcomp>
    return [self.visit(elt) for elt in obj]
  File "/home/rj/src/conda/envs/py35/lib/python3.5/site-packages/pythonparser/algorithm.py", line 41, in visit
    return self._visit_one(obj)
  File "/home/rj/src/conda/envs/py35/lib/python3.5/site-packages/pythonparser/algorithm.py", line 32, in _visit_one
    return getattr(self, visit_attr)(node)
  File "/home/rj/work/nist/artiq/artiq/compiler/transforms/inferencer.py", line 1280, in visit_FunctionDefT
    self.generic_visit(node)
  File "/home/rj/src/conda/envs/py35/lib/python3.5/site-packages/pythonparser/algorithm.py", line 27, in generic_visit
    self.visit(getattr(node, field_name))
  File "/home/rj/src/conda/envs/py35/lib/python3.5/site-packages/pythonparser/algorithm.py", line 39, in visit
    return [self.visit(elt) for elt in obj]
  File "/home/rj/src/conda/envs/py35/lib/python3.5/site-packages/pythonparser/algorithm.py", line 39, in <listcomp>
    return [self.visit(elt) for elt in obj]
  File "/home/rj/src/conda/envs/py35/lib/python3.5/site-packages/pythonparser/algorithm.py", line 41, in visit
    return self._visit_one(obj)
  File "/home/rj/src/conda/envs/py35/lib/python3.5/site-packages/pythonparser/algorithm.py", line 34, in _visit_one
    return self.generic_visit(node)
  File "/home/rj/src/conda/envs/py35/lib/python3.5/site-packages/pythonparser/algorithm.py", line 27, in generic_visit
    self.visit(getattr(node, field_name))
  File "/home/rj/src/conda/envs/py35/lib/python3.5/site-packages/pythonparser/algorithm.py", line 41, in visit
    return self._visit_one(obj)
  File "/home/rj/src/conda/envs/py35/lib/python3.5/site-packages/pythonparser/algorithm.py", line 32, in _visit_one
    return getattr(self, visit_attr)(node)
  File "/home/rj/work/nist/artiq/artiq/compiler/transforms/inferencer.py", line 905, in visit_CallT
    self.generic_visit(node)
  File "/home/rj/src/conda/envs/py35/lib/python3.5/site-packages/pythonparser/algorithm.py", line 27, in generic_visit
    self.visit(getattr(node, field_name))
  File "/home/rj/src/conda/envs/py35/lib/python3.5/site-packages/pythonparser/algorithm.py", line 39, in visit
    return [self.visit(elt) for elt in obj]
  File "/home/rj/src/conda/envs/py35/lib/python3.5/site-packages/pythonparser/algorithm.py", line 39, in <listcomp>
    return [self.visit(elt) for elt in obj]
  File "/home/rj/src/conda/envs/py35/lib/python3.5/site-packages/pythonparser/algorithm.py", line 41, in visit
    return self._visit_one(obj)
  File "/home/rj/src/conda/envs/py35/lib/python3.5/site-packages/pythonparser/algorithm.py", line 32, in _visit_one
    return getattr(self, visit_attr)(node)
  File "/home/rj/work/nist/artiq/artiq/compiler/transforms/inferencer.py", line 454, in visit_BinOpT
    node.left  = self._coerce_one(left_type, node.left, other_node=node.right)
  File "/home/rj/work/nist/artiq/artiq/compiler/transforms/inferencer.py", line 296, in _coerce_one
    node.type.unify(typ)
  File "/home/rj/work/nist/artiq/artiq/compiler/types.py", line 130, in unify
    self.params[param].unify(other.params[param])
  File "/home/rj/work/nist/artiq/artiq/compiler/types.py", line 89, in unify
    self.find().unify(other)
  File "/home/rj/work/nist/artiq/artiq/compiler/types.py", line 477, in unify
    raise UnificationError(self, other)
artiq.compiler.types.UnificationError: (artiq.compiler.types.TValue(64), artiq.compiler.types.TValue(32))
```

whitequark, 2017-04-07:
Well, the compiler should never raise an exception instead of a diagnostic, so there's at least one clear bug here.

whitequark, 2017-04-07:
That said, Python arithmetics, which automatically expands widths, and is also polymorphic in terms of operands, is problematic to implement in an unification-based system, so the answer to your question is "not necessarily".

jordens, 2017-04-07:
but isn't at least this specific case of `i32 -> i64` well defined?

whitequark, 2017-04-07:
I think it's just a bug, yes.

commit 31048f4b (2017-04-12): compiler: fix monomorphization of coerced integer literals.

Fixes #703.
commit be2683e0 (2017-04-12): compiler: fix monomorphization of coerced integer literals.

Fixes #703.
closed 2017-04-12 by whitequark