m-labs/artiq #704: plot_xy applet fails with pyqtgraph < 0.10.0

npisenti, 2017-04-07:
The plot_xy applet fails after a clean conda install (v2.2, on ubuntu, which packaged pyqtgraph v0.9.10). This is fixed with `pip install --upgrade pyqtgraph`, which installs `v0.10.0`. This seems to be a known bug elsewhere online with the switch to Qt5.

Applet signature:  `$python -m artiq.applets.plot_xy --embed $ipc_address --title "test" temp` (where `temp` is a 1D ndarray).

Traceback:
```
INFO:applet():print:error processing parent message
INFO:applet():print:Traceback (most recent call last):
INFO:applet():print:  File "/home/sr/miniconda3/envs/artiq-main/lib/python3.5/site-packages/artiq/applets/simple.py", line 59, in listen
INFO:applet():print:    self.mod_cb(mod)
INFO:applet():print:  File "/home/sr/miniconda3/envs/artiq-main/lib/python3.5/site-packages/artiq/applets/simple.py", line 201, in sub_mod
INFO:applet():print:    self.emit_data_changed(self.data, [mod])
INFO:applet():print:  File "/home/sr/miniconda3/envs/artiq-main/lib/python3.5/site-packages/artiq/applets/simple.py", line 266, in emit_data_changed
INFO:applet():print:    self.main_widget.data_changed(data, mod_buffer, title)
INFO:applet():print:  File "/home/sr/miniconda3/envs/artiq-main/lib/python3.5/site-packages/artiq/applets/plot_xy.py", line 40, in data_changed
INFO:applet():print:    self.plot(x, y, pen=None, symbol="+")
INFO:applet():print:  File "/home/sr/miniconda3/envs/artiq-main/lib/python3.5/site-packages/pyqtgraph/graphicsItems/PlotItem/PlotItem.py", line 636, in plot
INFO:applet():print:    item = PlotDataItem(*args, **kargs)
INFO:applet():print:  File "/home/sr/miniconda3/envs/artiq-main/lib/python3.5/site-packages/pyqtgraph/graphicsItems/PlotDataItem.py", line 177, in __init__
INFO:applet():print:    self.setData(*args, **kargs)
INFO:applet():print:  File "/home/sr/miniconda3/envs/artiq-main/lib/python3.5/site-packages/pyqtgraph/graphicsItems/PlotDataItem.py", line 461, in setData
INFO:applet():print:    self.updateItems()
INFO:applet():print:  File "/home/sr/miniconda3/envs/artiq-main/lib/python3.5/site-packages/pyqtgraph/graphicsItems/PlotDataItem.py", line 493, in updateItems
INFO:applet():print:    self.scatter.setData(x=x, y=y, **scatterArgs)
INFO:applet():print:  File "/home/sr/miniconda3/envs/artiq-main/lib/python3.5/site-packages/pyqtgraph/graphicsItems/ScatterPlotItem.py", line 297, in setData
INFO:applet():print:    self.addPoints(*args, **kargs)
INFO:applet():print:  File "/home/sr/miniconda3/envs/artiq-main/lib/python3.5/site-packages/pyqtgraph/graphicsItems/ScatterPlotItem.py", line 398, in addPoints
INFO:applet():print:    self.updateSpots(newData)
INFO:applet():print:  File "/home/sr/miniconda3/envs/artiq-main/lib/python3.5/site-packages/pyqtgraph/graphicsItems/ScatterPlotItem.py", line 551, in updateSpots
INFO:applet():print:    self.fragmentAtlas.getAtlas() # generate atlas so source widths are available.
INFO:applet():print:  File "/home/sr/miniconda3/envs/artiq-main/lib/python3.5/site-packages/pyqtgraph/graphicsItems/ScatterPlotItem.py", line 193, in getAtlas
INFO:applet():print:    self.buildAtlas()
INFO:applet():print:  File "/home/sr/miniconda3/envs/artiq-main/lib/python3.5/site-packages/pyqtgraph/graphicsItems/ScatterPlotItem.py", line 186, in buildAtlas
INFO:applet():print:    self.atlasData[x:x+w, y:y+h] = rendered[key]
INFO:applet():print:TypeError: slice indices must be integers or None or have an __index__ method
INFO:applet():print:Task was destroyed but it is pending!
INFO:applet():print:task: <Task pending coro=<AppletIPCClient.listen() running at /home/sr/miniconda3/envs/artiq-main/lib/python3.5/site-packages/artiq/applets/simple.py:47> wait_for=<Future pending cb=[Task._wakeup()]>>
ERROR:dashboard:artiq.gui.applets:error processing data from applet, server stopped
Traceback (most recent call last):
  File "/home/sr/miniconda3/envs/artiq-main/lib/python3.5/site-packages/artiq/gui/applets.py", line 56, in serve
    obj = await self.read_pyon()
  File "/home/sr/miniconda3/envs/artiq-main/lib/python3.5/site-packages/artiq/gui/applets.py", line 33, in read_pyon
    return pyon.decode(line.decode())
  File "/home/sr/miniconda3/envs/artiq-main/lib/python3.5/site-packages/artiq/protocols/pyon.py", line 202, in decode
    return eval(s, _eval_dict, {})
  File "<string>", line 0
    
    ^
SyntaxError: unexpected EOF while parsing

```

jordens, 2017-04-07:
We have been using Qt5 for much longer. That itself can't be it.
But maybe related to the switch to anaconda's qt5 packages. @cjbe ?
Also the backtrace looks like it might be something else entirely.

npisenti, 2017-04-07:
My thought about Qt5 and pyqtgraph 0.10.0 came from google pulling up this issue: https://github.com/Ulm-IQO/qudi/issues/47. Based on here (https://github.com/Ulm-IQO/qudi/issues/47), it might also originate in numpy 1.11 -> 1.12?

sbourdeauducq, 2017-04-08:
@npisenti With pyqtgraph from pip, does the image applet exit cleanly? This patch was required: https://github.com/m-labs/pyqtgraph/commit/7d34e22da900e07d9d9d6009fc4bc1e91c4ac9a5

sbourdeauducq, 2017-04-08:
@npisenti Well, this particular bug does seem fixed in 0.10 (https://github.com/pyqtgraph/pyqtgraph/blob/1426e334e1d20542400d77c72c132b04c6d17ddb/pyqtgraph/imageview/ImageView.py#L396). Can you try other applets, try to break them, and make sure that they behave?

sbourdeauducq, 2017-04-10:
Adding the conda-forge channel before installing ARTIQ fixes the problem, as it installs pyqtgraph 0.10 then. AFAICT there are no adverse effects.

npisenti, 2017-04-10:
I don't see any issues either. All seems to work as expected.

closed 2017-04-13 by sbourdeauducq