m-labs/artiq #706: constant folding is inaccurate

jordens, 2017-04-08:
```python
        while True:
              delay(8*us)
              self.led.on()
              delay(8*us)
              self.led.off()

```

With a `ref_period` of `1e-9` those two delays should be 8000 mu, but they are 7999 mu.

```llvm
while.head.i:                                     ; preds = %while.head.i, %_Z41artiq.coredevice.core.Core.break_realtimeI26artiq.coredevice.core.CoreEzz.exit.i
  %now.old.i = phi i64 [ %now.new.1.i, %while.head.i ], [ %now.new.i5.i.i, %_Z41artiq.coredevice.core.Core.break_realtimeI26artiq.coredevice.core.CoreEzz.exit.i ], !      + dbg !57
  %now.new.i = add i64 %now.old.i, 7999, !dbg !57
  store i64 %now.new.i, i64* @now, align 4, !dbg !57, !tbaa !42
  call void @rtio_output(i64 %now.new.i, i32 %val.LOC.self.FLD.channel.i.i4.i.pre, i32 0, i32 1), !dbg !58, !tbaa !17
  store i64 %now.new.i, i64* getelementptr inbounds (%I.artiq.coredevice.ttl.TTLOut.2474, %I.artiq.coredevice.ttl.TTLOut.2474* @O.11, i32 0, i32 2), align 4, !dbg !59
  %now.new.1.i = add i64 %now.old.i, 15998, !dbg !60
  store i64 %now.new.1.i, i64* @now, align 4, !dbg !60, !tbaa !42
  call void @rtio_output(i64 %now.new.1.i, i32 %val.LOC.self.FLD.channel.i.i4.i.pre, i32 0, i32 0), !dbg !61, !tbaa !17
  store i64 %now.new.1.i, i64* getelementptr inbounds (%I.artiq.coredevice.ttl.TTLOut.2474, %I.artiq.coredevice.ttl.TTLOut.2474* @O.11, i32 0, i32 2), align 4, !dbg !65
  br label %while.head.i, !dbg !66
}

```

whitequark, 2017-04-12:
I strongly suspect this is how LLVM folds floating point operations; it seems hard to imagine what ARTIQ could get wrong to get such a result. I will investigate why, though.

whitequark, 2017-04-21:
Let me illustrate why this happens by adding some ARTIQ IR:
```
; test.py:11               delay(8*us)
  %8.float = float coerce numpy.int32 8
; test.py:11               delay(8*us)
  %UNN.5 = float arith(Mult) float %8.float, float 1e-06
; test.py:11               delay(8*us)
  %UNN.6 = float arith(Div) float %UNN.5, float 1e-09
  %UNN.7 = numpy.int64 coerce float %UNN.6
  %"BLT.delay_mu" = NoneType builtin(delay_mu) numpy.int64 %UNN.7
```
```
>>> 8*1e-06/1e-09
7999.999999999999
```

commit ed2b10c5 (2017-04-21): compiler: in codegen for delay(), round fp instead of truncating.

Consider delay(8*us). It results in the following computation...
  >>> 8*1e-06/1e-09
  7999.999999999999
with the result promptly getting truncated to 7999.

Fixes #706.
commit c33f9637 (2017-04-22): compiler: in codegen for delay(), round fp instead of truncating.

Consider delay(8*us). It results in the following computation...
  >>> 8*1e-06/1e-09
  7999.999999999999
with the result promptly getting truncated to 7999.

Fixes #706.
closed 2017-04-21 by whitequark