m-labs/artiq #707: prefix git tags with "release-"

jordens, 2017-04-08:
... so that versioneer does not get confused by project tags (for phaser, smoltcp etc).
Needs duplication of the last tags in release-2 and master, needs changes in versioneer config.

whitequark, 2017-04-08:
OK but then we should rename `release-2` to at least `release-2.x` so that people don't get confused. Because if I can check out `release-2` and `release-2.0` that *is* confusing.

That said, will versioneer even not get confused by the branch with the same prefix by itself?

jordens, 2017-04-08:
Ack. I don't think it gets confused by that branch vs tag. But will check.

sbourdeauducq, 2017-04-08:
just use version-x.y tags.

klickverbot, 2017-04-08:
`vX.Y` seems to be most common format here on GitHub.

jordens, 2017-04-08:
I would also voted for vX.Y but that seems to pollute the namespace a bit too much.

whitequark, 2017-04-08:
`vX.Y` is great but I believe that this issue exists precisely because versioneer cannot cope with the `v` prefix. I.e. it treats anything starting with `v` (say, a branch called "virtual") as a release, and then breaks in an obscure way.

whitequark, 2017-04-08:
Arguably versioneer should be fixed.

sbourdeauducq, 2017-04-09:
If we're going to mess with versioneer, we can also keep the current tags and make it recognize only those that consist in a version number.

sbourdeauducq, 2017-04-09:
@jordens Also, a pragmatic solution is to use tags for versions only, and put other commit IDs elsewhere, e.g. in the wiki.

sbourdeauducq, 2017-04-09:
Versioneer seems a bit tedious to fix anyway, they've had a corresponding issue open for more than a year. https://github.com/warner/python-versioneer/issues/110

sbourdeauducq, 2017-05-15:
Can we go with using tags only for versions? This is the only thing versioneer actually supports.

jordens, 2017-05-23:
Ok. Let's use well-organized branches to store/tag things then, i.e. `old/phaser`.

closed 2017-05-23 by jordens

sbourdeauducq, 2017-05-23:
OK.