m-labs/artiq #708: ksupport crash

jordens, 2017-04-09:
```
[ 15568059547us]  INFO(kernel): panic at /home/rj/src/artiq/artiq/firmware/ksupport/lib.rs:460: exception 6 at PC 0x40840490, EA 0x4084406c
[ 15568070876us] ERROR(runtime::session): unexpected request RunAborted from kernelCPU
[ 15568078178us] ERROR(runtime::session): session aborted: protocol error
```

This is

```python
from artiq.experiment import *

class PDQ2SPI(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("pdq0")

    @kernel
    def run(self):
        self.core.reset()
        self.core.break_realtime()
        self.pdq0.setup_bus()

        while True:
            delay(1*ms)
            self.pdq0.write_crc(0)
            self.pdq0.write_config()
            self.pdq0.write_frame(0)
            self.pdq0.write_mem(0, 0xfedc, [
                    1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17,
                        18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31,
                        32,
                    1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17,
                        18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31,
                        32,
                    1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17,
                        18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31,
                        32,
                    1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17,
                        18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31,
                        32,
                    1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17,
                        18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31,
                        32,
                    1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17,
                        18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31,
                        32,
                    1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17,
                        18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31,
                        32,
                    1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17,
                        18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31,
                        32,
                    1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17,
                        18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31,
                        32,
                    1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17,
                        18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31,
                        32,
                    1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17,
                        18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31,
                        32,
                    1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17,
                        18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31,
                        32,
                    1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17,
                        18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31,
                        32,
                    1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17,
                        18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31,
                        32,
                    1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17,
                        18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31,
                        32,
                    1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17,
                        18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31,
                        32,
                        ])
```

It crashes after about 2 minutes of running that loop.
I am working on ed8edf31. Runtime/ksupport is unaltered from 1ee3f964.

How do I debug?

whitequark, 2017-04-09:
> exception 6 at PC 0x40840490, EA 0x4084406c

That's quite baffling. It's an unaligned access exception, but it's reported at an aligned address.

Anyway what I would do is to apply objdump to ksupport.elf and then see the function in which the problem is. In ksupport specifically we can also use the builtin libunwind to extract a backtrace so let me know if that's not enough and I'll implement that.

jordens, 2017-04-09:
i remember you (or someone) reporting that same thing (unaligned access with aligned address reported) a few months ago. what was the issue back then?

whitequark, 2017-04-09:
I have no recollection of that.

whitequark, 2017-04-12:
Should I wait for you to reduce the crash (or dissect it at all), or do it myself?

jordens, 2017-04-12:
I'll look into it.

jordens, 2017-04-13:
Can't reproduce.
May have been old/bad runtime.

closed 2017-04-13 by jordens