m-labs/artiq #711: pre-link scripts conda warning

sbourdeauducq, 2017-04-10:
Conda now spits this out when installing ARTIQ:
```
/home/sb/miniconda3/lib/python3.6/site-packages/conda/core/link.py:470: UserWarning: Package m-labs/label/main::lit-0.4.1-py_2 uses a pre-link script. Pre-link scripts are potentially dangerous.
This is because pre-link scripts have the ability to change the package contents in the
package cache, and therefore modify the underlying files for already-created conda
environments.  Future versions of conda may deprecate and ignore pre-link scripts.

  """ % dist))
/home/sb/miniconda3/lib/python3.6/site-packages/conda/core/link.py:470: UserWarning: Package m-labs/label/main::outputcheck-0.4.2-py_1 uses a pre-link script. Pre-link scripts are potentially dangerous.
This is because pre-link scripts have the ability to change the package contents in the
package cache, and therefore modify the underlying files for already-created conda
environments.  Future versions of conda may deprecate and ignore pre-link scripts.

  """ % dist))
/home/sb/miniconda3/lib/python3.6/site-packages/conda/core/link.py:470: UserWarning: Package m-labs/label/main::aiohttp-0.17.2-py_3 uses a pre-link script. Pre-link scripts are potentially dangerous.
This is because pre-link scripts have the ability to change the package contents in the
package cache, and therefore modify the underlying files for already-created conda
environments.  Future versions of conda may deprecate and ignore pre-link scripts.

  """ % dist))
/home/sb/miniconda3/lib/python3.6/site-packages/conda/core/link.py:470: UserWarning: Package m-labs/label/main::asyncserial-0.1-py_11+git758c1e5 uses a pre-link script. Pre-link scripts are potentially dangerous.
This is because pre-link scripts have the ability to change the package contents in the
package cache, and therefore modify the underlying files for already-created conda
environments.  Future versions of conda may deprecate and ignore pre-link scripts.

  """ % dist))
/home/sb/miniconda3/lib/python3.6/site-packages/conda/core/link.py:470: UserWarning: Package m-labs/label/main::dateutil-2.4.2-py_2 uses a pre-link script. Pre-link scripts are potentially dangerous.
This is because pre-link scripts have the ability to change the package contents in the
package cache, and therefore modify the underlying files for already-created conda
environments.  Future versions of conda may deprecate and ignore pre-link scripts.

  """ % dist))
/home/sb/miniconda3/lib/python3.6/site-packages/conda/core/link.py:470: UserWarning: Package m-labs/label/main::pythonparser-1.0-py_7 uses a pre-link script. Pre-link scripts are potentially dangerous.
This is because pre-link scripts have the ability to change the package contents in the
package cache, and therefore modify the underlying files for already-created conda
environments.  Future versions of conda may deprecate and ignore pre-link scripts.

  """ % dist))
/home/sb/miniconda3/lib/python3.6/site-packages/conda/core/link.py:470: UserWarning: Package m-labs/label/dev::quamash-0.5.5-py_2 uses a pre-link script. Pre-link scripts are potentially dangerous.
This is because pre-link scripts have the ability to change the package contents in the
package cache, and therefore modify the underlying files for already-created conda
environments.  Future versions of conda may deprecate and ignore pre-link scripts.

  """ % dist))
/home/sb/miniconda3/lib/python3.6/site-packages/conda/core/link.py:470: UserWarning: Package m-labs/label/dev::artiq-3.0.dev-py_984+gitf314f8b uses a pre-link script. Pre-link scripts are potentially dangerous.
This is because pre-link scripts have the ability to change the package contents in the
package cache, and therefore modify the underlying files for already-created conda
environments.  Future versions of conda may deprecate and ignore pre-link scripts.

  """ % dist))
```

sbourdeauducq, 2017-06-05:
The solution is to use the new noarch system (https://www.continuum.io/blog/developer-blog/condas-new-noarch-packages).
But of course, it doesn't always work. Entry points are broken (http://buildbot.m-labs.hk/builders/conda-lin64/builds/310/steps/conda_build/logs/stdio), and usage of setuptools is problematic.

sbourdeauducq, 2017-06-05:
Current status on 3.0:
```
/home/sb/miniconda3/lib/python3.6/site-packages/conda/core/link.py:470: UserWarning: Package m-labs/label/main::lit-0.4.1-py_2 uses a pre-link script. Pre-link scripts are potentially dangerous.
This is because pre-link scripts have the ability to change the package contents in the
package cache, and therefore modify the underlying files for already-created conda
environments.  Future versions of conda may deprecate and ignore pre-link scripts.

  """ % dist))
/home/sb/miniconda3/lib/python3.6/site-packages/conda/core/link.py:470: UserWarning: Package m-labs/label/main::outputcheck-0.4.2-py_1 uses a pre-link script. Pre-link scripts are potentially dangerous.
This is because pre-link scripts have the ability to change the package contents in the
package cache, and therefore modify the underlying files for already-created conda
environments.  Future versions of conda may deprecate and ignore pre-link scripts.

  """ % dist))
/home/sb/miniconda3/lib/python3.6/site-packages/conda/core/link.py:470: UserWarning: Package m-labs/label/dev::quamash-0.5.5-py_2 uses a pre-link script. Pre-link scripts are potentially dangerous.
This is because pre-link scripts have the ability to change the package contents in the
package cache, and therefore modify the underlying files for already-created conda
environments.  Future versions of conda may deprecate and ignore pre-link scripts.

  """ % dist))
/home/sb/miniconda3/lib/python3.6/site-packages/conda/core/link.py:470: UserWarning: Package m-labs/label/dev::artiq-3.0.dev-py_1083+git7f2f0031 uses a pre-link script. Pre-link scripts are potentially dangerous.
This is because pre-link scripts have the ability to change the package contents in the
package cache, and therefore modify the underlying files for already-created conda
environments.  Future versions of conda may deprecate and ignore pre-link scripts.
```

lit, outputcheck and artiq are difficult to fix due to the above-mentioned entry point breakage.

sbourdeauducq, 2017-06-05:
And rebuilding quamash is also difficult due to the Qt5 packaging mess.

sbourdeauducq, 2017-06-05:
Quamash fixed, it wasn't a mess for once.

sbourdeauducq, 2017-06-06:
Conda entry point breakage is fixed here, but not in releases yet:
https://github.com/conda/conda-build/commit/6aa32e103226fa61a03bfc8a0b935d73c4572d5e

sbourdeauducq, 2017-06-06:
All fixed.

dhslichter, 2017-10-23:
Getting pre-link warnings again today with a fresh install on a new Windows machine.  I made sure to update conda, running 4.3.27 py27hcd9d231_0.  The only wrinkle I can think of here is that my root environment from which I am creating the new environment to install ARTIQ is a Python 2 environment, but my impression was that conda ought to be able to handle this.  Full output is below.

```
>conda create -n artiq3_102317 python=3.5 artiq=3.0 artiq-kc705-nist_qc2
Fetching package metadata ...................
Solving package specifications: .

Package plan for installation in environment C:\Anaconda2\envs\artiq3_102317:

The following NEW packages will be INSTALLED:

    aiohttp:              0.17.2-py_3           m-labs/label/main
    artiq:                3.0-py_0+git04a9a0ce  m-labs/label/main
    artiq-kc705-nist_qc2: 3.0-py_0+git04a9a0ce  m-labs/label/main
    asyncserial:          0.1-py_13+git340e430  m-labs/label/main
    binutils-or1k-linux:  2.25.1-4              m-labs/label/main
    ca-certificates:      2017.7.27.1-0         conda-forge/label/main
    certifi:              2017.7.27.1-py35_0    conda-forge/label/main
    cffi:                 1.10.0-py35_0         conda-forge/label/main
    chardet:              2.2.1-py_3            m-labs/label/main
    curl:                 7.49.1-vc14_2         conda-forge/label/main [vc14]
    h5py:                 2.7.1-py35_1          conda-forge/label/main
    hdf5:                 1.8.18-vc14_0         conda-forge/label/main [vc14]
    icc_rt:               2017.0.4-h97af966_0
    icu:                  58.1-vc14_1           conda-forge/label/main [vc14]
    intel-openmp:         2018.0.0-hcd89f80_7
    jpeg:                 9b-vc14_1             conda-forge/label/main [vc14]
    levenshtein:          0.12.0-py35_1         m-labs/label/main
    libgit2:              0.25.0-vc14_0         conda-forge/label/main [vc14]
    libpng:               1.6.28-vc14_0         conda-forge/label/main [vc14]
    libusb:               1.0.20-0              m-labs/label/main
    lit:                  0.4.1-py_9            m-labs/label/main
    llvmlite-artiq:       0.12.0.dev-py35_30    m-labs/label/main
    mkl:                  2018.0.0-h36b65af_4
    numpy:                1.13.3-py35hb7e80fb_0
    openocd:              0.10.0-1              m-labs/label/main
    openssl:              1.0.2l-vc14_0         conda-forge/label/main [vc14]
    outputcheck:          0.4.2-py_7            m-labs/label/main
    pip:                  9.0.1-py35_0          conda-forge/label/main
    prettytable:          0.7.2-py_3            m-labs/label/main
    pycparser:            2.18-py35_0           conda-forge/label/main
    pygit2:               0.25.0-py35_vc14_0    conda-forge/label/main [vc14]
    pyqt:                 5.6.0-py35_4          conda-forge/label/main
    pyqtgraph:            0.10.0-py35_2         conda-forge/label/main
    pyserial:             3.4-py35_0            conda-forge/label/main
    python:               3.5.4-0               conda-forge/label/main
    python-dateutil:      2.6.1-py35_0          conda-forge/label/main
    pythonparser:         1.1-py_8              m-labs/label/main
    qt:                   5.6.2-vc14_1          conda-forge/label/main [vc14]
    quamash:              0.5.5-py_4            m-labs/label/main
    regex:                2015.11.22-py35_1     m-labs/label/main
    scipy:                0.19.1-py35h1e2829e_3
    setuptools:           36.6.0-py35_1         conda-forge/label/main
    sip:                  4.18-py35_1           conda-forge/label/main
    six:                  1.11.0-py35_1         conda-forge/label/main
    vc:                   14-0                  conda-forge/label/main
    vs2015_runtime:       14.0.25420-0          conda-forge/label/main
    wheel:                0.30.0-py_1           conda-forge/label/main
    wincertstore:         0.2-py35_0            conda-forge/label/main
    zlib:                 1.2.11-vc14_0         conda-forge/label/main [vc14]

Proceed ([y]/n)? y

binutils-or1k- 100% |###############################| Time: 0:00:03   8.28 MB/s
ca-certificate 100% |###############################| Time: 0:00:00   1.11 MB/s
icc_rt-2017.0. 100% |###############################| Time: 0:00:01   7.65 MB/s
intel-openmp-2 100% |###############################| Time: 0:00:00   3.34 MB/s
libusb-1.0.20- 100% |###############################| Time: 0:00:00 390.19 kB/s
vs2015_runtime 100% |###############################| Time: 0:00:00   2.18 MB/s
jpeg-9b-vc14_1 100% |###############################| Time: 0:00:00   1.01 MB/s
[... 51 lines cut ...]

  """) % dist)
C:\Anaconda2\lib\site-packages\conda\core\link.py:475: UserWarning: Package m-la
bs/label/main::prettytable-0.7.2-py_3 uses a pre-link script. Pre-link scripts a
re potentially dangerous.
This is because pre-link scripts have the ability to change the package contents
 in the
package cache, and therefore modify the underlying files for already-created con
da
environments.  Future versions of conda may deprecate and ignore pre-link script
s.

  """) % dist)
C:\Anaconda2\lib\site-packages\conda\core\link.py:475: UserWarning: Package m-la
bs/label/main::aiohttp-0.17.2-py_3 uses a pre-link script. Pre-link scripts are
potentially dangerous.
This is because pre-link scripts have the ability to change the package contents
 in the
package cache, and therefore modify the underlying files for already-created con
da
environments.  Future versions of conda may deprecate and ignore pre-link script
s.

  """) % dist)
```

sbourdeauducq, 2017-10-24:
The problematic packages (chardet, prettytable, aiohttp) are now in conda-forge, and the conda-forge versions (which do not have the problem) should be used instead of ours. I will move ours into an ``obsolete`` label so that conda won't easily find them.

closed 2017-10-24 by sbourdeauducq