m-labs/artiq #712: dma sequence recording slow

cjbe, 2017-04-10:
DMA sequence recording seems to take an extraordinary amount of time.

The below simple experiment outputs
```
total call time : 22.754499673843384
8.688764104
0.020485168
```
so compiling and running the sequence takes ~20s, and the DMA recording runtime is ~9s

The experiment:
```python
class DmaTimingTest(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("core_dma")
        self.setattr_device("ttl0")

    def run(self):
        t0 = time.time()
        t1,t2,t3 = self.run_dma()
        dt = time.time()-t0
        print("total call time : {}".format(dt))
        print(self.core.mu_to_seconds(t2-t1))
        print(self.core.mu_to_seconds(t3-t2))

    @kernel
    def run_dma(self):
        self.core.reset()
        self.ttl0.output()

        t1 = self.core.get_rtio_counter_mu()

        with self.core_dma.record("test"):
            for i in range(10000):
                delay(1*us)
                self.ttl0.pulse(1*us)

        t2 = self.core.get_rtio_counter_mu()

        self.core.break_realtime()
        self.core_dma.replay("test")
        self.core_dma.erase("test")

        t3 = self.core.get_rtio_counter_mu()

        return t1,t2,t3
```

As a comparison, calling run_loop() takes ~0.26s
```python
    @kernel
    def run_loop(self):
        self.core.reset()
        self.ttl0.output()
        for i in range(10000):
            delay(1*us)
            self.ttl0.pulse(1*us)
```

whitequark, 2017-04-10:
That's what I expect from the current implementation, yes. I recall we've discussed this with @sbourdeauducq and decided that doing a syscall for every `rtio_output` is acceptable.

Trying to manage memory shared between comms and kernel CPU is nontrivial (that was my initial implementation) and it's far easier to guarantee the correctness of the current implementation.

jordens, 2017-04-10:
Why does it have to go through the comms CPU at all? Why can't the kernel CPU own the DMA engine? Why does there need to be any shared memory between comms and kernel?
We need to find a way that does not lead to a 100-fold slow down. The suggestion was to have the gateware do the recording transparently, in a similar way as it would for input DMA or distributed DMA.

whitequark, 2017-04-10:
> Why can't the kernel CPU own the DMA engine?

It does. This is purely a memory management issue.

> Why does it have to go through the comms CPU at all?

Because the DMA traces are stored between kernel runs, whereas kernel CPU state is not.

> Why does there need to be any shared memory between comms and kernel?

See above, and because ksupport does not currently have an allocator (which creates all sorts of problems like balancing the sizes of the comms CPU and kernel CPU arenas...).

sbourdeauducq, 2017-04-10:
@jordens 

> The suggestion was to have the gateware do the recording transparently

Have you seen how much time was spent fixing gateware bugs and Vivado breakdowns just for DMA playback? And gateware does not solve the memory management issue, but makes it worse.

@whitequark I had not understood that it would make a round-trip to the comms CPU at every RTIO event. How about making those round-trips more rare, e.g. by speculatively allocating extra space in buffers obtained from the comms CPU?

whitequark, 2017-04-10:
> How about making those round-trips more rare, e.g. by speculatively allocating extra space in buffers obtained from the comms CPU?

That's not how it works. Comms CPU allocates buffers and writes into them; kernel CPU sends it a high-level representation of an RTIO event. But what I could rather easily do is accumulate RTIO events in some sort of static buffer, and only send a slice from the buffer over to comms CPU once in a while.

jordens, 2017-04-10:
@sbourdeauducq You seem to claim that because gateware is difficult, we can't use it and have to live with and extend/patch an unacceptably slow implementation that also intentionally makes implementing future features harder.
Time spent on fixing gateware bugs or Vivado breakdowns should arguably not be the guiding principle when judging design decisions.
And how exactly does gateware make the memory management issues worse? Worse than what?

sbourdeauducq, 2017-04-10:
I don't see how a software implementation is fundamentally slow (writing some data to a DMA buffer vs. interfacing with the RTIO core), nor how it makes implementing future features harder. And we don't have infinite resources, so development time is definitely a factor to consider.
With a gateware implementation, growing buffers is difficult. In software, there are already standard techniques and existing code for doing that.

whitequark, 2017-04-10:
> also intentionally makes implementing future features harder.

Which features are that? Also, *intentionally*?

sbourdeauducq, 2017-04-10:
> But what I could rather easily do is accumulate RTIO events in some sort of static buffer, and only send a slice from the buffer over to comms CPU once in a while.

Yes, that's equivalent. Sounds good to me.

jordens, 2017-04-10:
It is slow irrespective of that it fundamentally could be.
As for future features: DMA input and distributed DMA. Both probably need DMA gateware that takes RTIO events (from the PHYs or from DRTIO upstream), (re)serializes them and writes them into memory. Using a software implementation that achieves the bare minimum for non-distributed output-only DMA appears to shy away from a significant but inevitable amount of work, paving the way for ample opportunity to later again be surprised about unexpected complexity. DRTIO-transceiver-test vs DRTIO-reality would be precedence.
I'd love to see that this approach to DMA is not just some bare minimum shortcut.
We probably won't need infinite resources for that.
And it appears impossible to gauge and compare the actual net development time from the amount of IRC discussions, and commit logs.

sbourdeauducq, 2017-04-10:
Why should writing a record to a memory buffer be slower than writing it to the RTIO(/DMA) core?

I disagree about distributed DMA. The current implementation actually makes it pretty simple: reuse the front-end of the current DMA core, drive it from the satellite manager firmware and control it via aux packets. DMA buffers can also be transferred via software on the aux channel. The only difficulty is the DMA back-end - muxing the output of the DMA front-end with the events received from DRTIO (unlike RTIO, DRTIO uses synchronous FIFOs and a different interface for receiving events).

Input events are not the same as output events, and taking an event from the CPU is not the same as taking it from a RTIO PHY (flow control, clock domains, ...), so I also mostly disagree on the topic of DMA input.

jordens, 2017-04-10:
You will have to battle with the AUX bandwidth limitation, muxing in/out the already existing AUX traffic, probably yet more special casing for all those different RTIO event paths in the already overloaded RTIO API...
Won't we need a couple CDCs on the DRTIO satellites anyway to get the data from and to the DDR?
Anyway. Instead of stressing the differences between these features and arguing for separate implementation I would have thought there are sufficient commonalities that would make abstracting over the differences and reusing code worthwhile.

sbourdeauducq, 2017-04-10:
Worry not about the overloaded RTIO functions - the syscalls are replaced via the linker when a DMA context manager is active, to avoid slowing down the normal RTIO path. And I'd rather deal with special casing in software than in gateware.
Auxiliary bandwidth isn't that scarce, Metlino/Sayma will have 300Mbps in the absence of RT traffic.

cjbe, 2017-04-11:
In addition, the replay function is very slow - the run-time of `self.core_dma.replay("test")` is ~1.3 ms! This is 2 or 3 orders of magnitude longer than certain DMA sequence durations.

whitequark, 2017-04-11:
That stems out of a similar design decision--every DMA replay first requests the address from the comms CPU, which performs a cache flush (even two, really). I do not see any especially good way to work around that. It might be possible to write a hack.

cjbe, 2017-04-11:
Sharing recorded DMA sequences between kernel runs is a much lower priority to me than having fast DMA recording and low-latency DMA playback.

Would removing the persistance of DMA sequences between kernels allow us to solve these speed issues?

whitequark, 2017-04-11:
> Would removing the persistance of DMA sequences between kernels allow us to solve these speed issues?

Certainly. Note that this would result in a significantly different API. In particular, no string names for recorded traces, you get back a unique "handle" object, with O(1) lookup at replay time. And you cannot remove recorded traces, except wholesale.

cjbe, 2017-04-11:
Assuming that the time taken to record a DMA sequence is comparable to the time taken to directly output these events, these do not seem like problematic issues - this means that for any sequence fragment that one wants to replay more than a handful of times per kernel, it is cheaper to record it and play it back than it is to directly output the events each time.

(That is, assuming that
```python
with dma.record():
    kfun()
```
and `kfun()` take a comparable time to run)

whitequark, 2017-04-11:
> Assuming that the time taken to record a DMA sequence is comparable to the time taken to directly output these events

They take comparaible time to run iff the sequences are not persisted (because that means there need be no reallocations and no name->sequence maps). Anyway, I'd like to hear from @jordens, who wanted persistence in the first place.

hartytp, 2017-04-11:
> @cjbe Sharing recorded DMA sequences between kernel runs is a much lower priority to me than having fast DMA recording and low-latency DMA playback.

To clarify, do you mean:

a) Sharing recorded DMA sequences between kernel runs is not a something you value in general, and you'd be happy to live without it permanently.

or,

b) You need at least a basic form of DMA working asap, but the current slowness essentially cripples the present implementation. If removing DMA sequence persistence is the best way of getting DMA working properly then it's an acceptable short-term solution while a better longer-term plan is worked out.

cjbe, 2017-04-11:
I think sharing state between different experiments and different kernels is a pain to manage for the user. I think it is much better for each kernel to explictly record the sequences it needs. The only reason not to do this is if there is too much overhead, either in the compile / upload step, or the run-time of the kernel DMA recording is too large.

jordens, 2017-04-11:
Veto.
There are plenty of use cases where DMA sequences need to persist: everything where e.g. generating or uploading waveforms is a significant amount of CPU time or storage. Don't take a shortcut here.

cjbe, 2017-04-12:
OK - I see the use cases for persistance, but I want to make sure we don't hobble the record speed and especially the playback speed over this.

For playback we are currently messaging the comms CPU with a sequence name (~1ms), which then sets the DMA base address and time offset (a couple of instruction cycles).

Can we not split this playback operation up into two: the first operation messages the comms CPU with the sequence name, and returns a sequence handle object (containing the base address). The actual playback operation then just writes out the DMA base address and time offset (which should be very fast).

sbourdeauducq, 2017-04-12:
> the first operation messages the comms CPU with the sequence name, and returns a sequence handle object (containing the base address)

Not really, because if you create such an object, then delete the sequence, then play it back, you are programming the DMA engine with an invalid address which will result in very obscure bugs.

We can have the kernel CPU access the name->address map directly during playback. And if the name lookup is slow, add another "fast" path where the sequence name is a 8-bit integer, which then is simply used to index a table of up to 256 DMA sequences during playback.

whitequark, 2017-04-13:
> Not really, because if you create such an object, then delete the sequence, then play it back, you are programming the DMA engine with an invalid address which will result in very obscure bugs.
>
> We [...]

A better idea is something like this: you can ask the comms CPU for a handle, and you can ask for an unlimited amount of handles, with no "fast" paths or other strange limitations, but every handle gets invalidated if you ever erase a trace.

dhslichter, 2017-04-13:
Improving playback speed is one of the major goals of DMA, and persistence between kernels is definitely a feature one would like to have (imagine a DMA sequence for sideband cooling -- you'll probably call this in most experiments, and don't want to have to re-transfer and re-load into memory for each and every kernel).  So whatever solution comes needs to work with both of these goals in mind.  

A third consideration is that eventually it is desired to have low latencies (~10 us) between kernels as well (this is an as-yet-unfunded ARTIQ improvement), so it is important to choose a design that will not immediately thwart this goal either.  

Could one have a table of DMA handles at a specific location in memory (which persists between kernels), where the kernel CPU knows to look to get the start addresses for DMA sequences?  The comms CPU could add a handle to this table when it loads in new DMA sequences.  When a kernel calls for a DMA sequence to be deleted, the request is passed to the comms CPU, which then does the deletion and updates the table of DMA handles accordingly.  The comms CPU can do this asynchronously while the kernel CPU continues its own execution, modulo considerations of having to share the memory controller.

Also, why is the kernel CPU the thing that creates the DMA sequence in the format to be stored in memory?  Can this not be done at compile time on the PC and just sent directly into the appropriate location in memory by the comms CPU when the kernel is uploaded to the core device?

whitequark, 2017-04-13:
> Could one have a table of DMA handles at a specific location in memory (which persists between kernels), where the kernel CPU knows to look to get the start addresses for DMA sequences?

Any tables of handles have the usual problems associated with caching, and sharing it between kernel and comms CPU adds issues associated with shared mutable state, which is especially bad in our non-cache-coherent system. My proposal of using a global generation counter has none of these issues, and possibly even a lower time overhead, at the cost of having to re-request the sequences every time one of them is updated or erased. Since recording sequences is inherently costly (it is a non-realtime operation involving messaging and allocation) it does not seem that this tradeoff is undesirable.

> Also, why is the kernel CPU the thing that creates the DMA sequence in the format to be stored in memory?

It doesn't. The comms CPU does that; the kernel CPU merely serializes the arguments of rtio_output.

> Can this not be done at compile time on the PC and just sent directly into the appropriate location in memory by the comms CPU when the kernel is uploaded to the core device?

No, because it is legal to use arbitrary code when creating the DMA sequence (I don't remember whether @rjo or @sbourdeauducq requested that). If we moved to only allow creating the DMA sequences before compiling the kernel, that would certainly solve most of the other problems--the sequence is now simply an opaque pointer.

However, that would create one potentially major one. Right now, a very large sequence can be stored once and then re-used. But with the system you propose, it would have to be reuploaded every time, or we'll just go back to current situation except with less flexibility.

dhslichter, 2017-04-13:
> No, because it is legal to use arbitrary code when creating the DMA sequence (I don't remember whether @rjo or @sbourdeauducq requested that). If we moved to only allow creating the DMA sequences [...]

We definitely want to be able to re-use sequences and have them persist between kernels, but I am not so sure about the arbitrary code in DMA.  Are there really things we will be putting into a write DMA which are not known at compile time?  If so, it seems this would represent a small subset of cases rather than most cases, at least of the ones I envision using in our lab.  The primary purpose would be for outputting lengthy sequences of short pulses, where the process of putting pulses in the RTIO queue from the kernel CPU eats the slack over time.  These pulse sequences are for tasks like sideband cooling or randomized benchmarking.  One might wish to update such a DMA pulse sequence (e.g. for sideband cooling, changing the pulse time after a recalibration is done, or for randomized benchmarking choosing a new set of randomized gates), but these tasks would involve changing the timing for many, many pulses in the sequence.  It seems it would be faster (at least for long sequences) to have this done by the PC at compile time and re-uploaded, rather than having the core device doing the recomputation.  Is this correct?  

The case that this does not cover is if there is a parameter stored on the core device, and updated by some kernel on the core device, that one then wants to use to adjust the timing properties of an existing DMA in memory without having to involve the PC.  I would imagine that the process of doing this modification on the core device would probably be more costly in terms of time than splitting the entire DMA up into a loop of short, unchanging DMAs interleaved with pulses calculated on the fly by the kernel CPU in the normal manner, where the pulse duration is set by the parameter on the core device.  This will cost you maybe ~1 us of slack per pulse adjusted in this manner versus having it all in DMA, so even if there are 10,000 such pulses (which would be a lot!), any alternative solution would have to take less than 10 ms to be competitive in terms of experiment dead time.  

For read DMA, one is streaming timestamps (or ADC voltages) into memory for subsequent processing, and I haven't thought about whether arbitrary code is important.

whitequark, 2017-04-13:
> It seems it would be faster (at least for long sequences) to have this done by the PC at compile time and re-uploaded, rather than having the core device doing the recomputation. Is this correct?

This does not seem correct to me at all. There are all sorts of overhead associated with transporting large kernel images over the network, whereas generating simple sequences on the core device is extremely cheap. To give some numbers, right now the TCP throughput is 25 kilobytes per second (which is unnecessarily low; #685); based on tests I ran earlier I don't expect it to rise above low hundreds of kilobytes per second with current hardware and software architecture.

dhslichter, 2017-04-13:
One more comment, a bit of a step back: to my mind, the rationale behind DMA is that enables us to emit or receive pulses at a much higher rate than would be possible by simply using the kernel CPU to calculate them (or read them in) on the fly.  To that end, it seems wise to involve the kernel CPU and comms CPU as little as possible in the generation of DMA output sequences.

dhslichter, 2017-04-13:
>>It seems it would be faster (at least for long sequences) to have this done by the PC at compile time and re-uploaded, rather than having the core device doing the recomputation. Is this correct?

> This does not seem correct to me at all. There are all sorts of overhead associated with transporting large kernel images over the network, whereas generating simple sequences on the core device is [...]

As pointed out, this is a very low TCP throughput, but to rise above 100 kbps seems like it should be an achievable task, no?  For the general future performance of ARTIQ, it's going to be important to have a much fatter pipe than that between the PC and the FPGA hardware.  The FPGA is capable of running gigabit ethernet (nominally) -- could one transmit DMA sequences in such a way (requiring some additional gateware) that bypasses the comms CPU and just streams them as binary data from ethernet into RAM, and then (small) pointer information is sent to the comms CPU?

whitequark, 2017-04-13:
> As pointed out, this is a very low TCP throughput, but to rise above 100 kbps seems like it should be an achievable task, no? For the general future performance of ARTIQ, it's going to be important to have a much fatter pipe than that between the PC and the FPGA hardware.

I believe that right now the limiting step is the TCP stack, which has to run concurrently with the rest of the code (making the packet processing latency in principle unbounded, and in practice often fairly high), and only has very small (four 1500-byte, in every direction) buffers in the network core. If I recall correctly, we got only slightly above 100 kbps with lwIP, and lwIP was working in easier conditions than smoltcp. In any case, I expect that once the smoltcp bug that causes #685 is fixed, we will see somewhere between 100 kbps and 200 kbps of throughput.

> The FPGA is capable of running gigabit ethernet (nominally) -- could one transmit DMA sequences in such a way (requiring some additional gateware) that bypasses the comms CPU and just streams them as binary data from ethernet into RAM, and then (small) pointer information is sent to the comms CPU?

That would require implementing a reliable data transfer protocol in gateware. While a TCP offload engine in gateware is not unknown of, it is a nontrivial effort.

> To that end, it seems wise to involve the kernel CPU and comms CPU as little as possible in the generation of DMA output sequences.

I don't understand why you emphasize generation and not playback here.

dhslichter, 2017-04-13:
>I don't understand why you emphasize generation and not playback here.

Nomenclature failure on my part -- when I said "generation", I mean both generation and playback.  Basically, we want to involve the CPUs as little as possible, since they will likely be an early limiting step.

>That would require implementing a reliable data transfer protocol in gateware. While a TCP offload engine in gateware is not unknown of, it is a nontrivial effort.

It may be that as we work towards Metlino and larger distributed systems, this really becomes necessary to keep transfers from PC to hardware of all the experiment details from becoming too slow.  Understanding that this is a nontrivial effort, are there other ways in which one might increase the speed of the communications past the 100-200 kbps mark?  For a long DMA sequence with 10,000 pulses, this would still be ~seconds to upload a compiled sequence from the PC.

whitequark, 2017-04-13:
> Understanding that this is a nontrivial effort, are there other ways in which one might increase the speed of the communications past the 100-200 kbps mark?

An obvious way would be to dedicate a CPU core to running the TCP/IP stack. Another obvious way would be to redo the Ethernet core such that it can process larger packets (i.e. jumbo frames) and write directly to DRAM, so that scarce BRAM resources are not needed. Yet another way, less obvious, would be to offload the hottest parts of the TCP/IP stack to hardware--checksum calculation, copying data to its final destination in RAM, feeding an RTIO FIFO.

But before all that, we should implement a sampling profiler for our firmware and see where the time is actually spent.

sbourdeauducq, 2017-04-14:
The throughput should be around 1MB/s (like the previous runtime), not 100k/s.

whitequark, 2017-04-14:
Yes. Please multiply all numbers I mentioned previously (except 25 kbps) by a factor of 10, I misremembered, and could not find the logs as it was a while ago.

dhslichter, 2017-04-14:
So is this due in part to the reimplementation of the runtime in Rust?

whitequark, 2017-04-14:
> So is this due in part to the reimplementation of the runtime in Rust? Does that slow the soft CPUs?

The slowdown is due to a regression in smoltcp that has not been fixed yet. There is no inherent reason the Rust runtime could not have a throughput as high (and in fact it did before the regression).

dhslichter, 2017-04-14:
Ack.  But to put some numbers on it, a single RTIO event is ~10 bytes, so a longer DMA sequence with 10,000 pulses (20,000 events) would take ~200 ms to send even at 1 MB/s, so increasing the speed beyond that level (as discussed above, by making various possible design modifications) is definitely something to put in the "future improvements" column.

sbourdeauducq, 2017-04-15:
@whitequark About output_wide and kcpu-side buffering: the max event data size in the gateware is 512 bits, assume the same there and make it work.

sbourdeauducq, 2017-04-15:
And remember we have a lot of SDRAM so efficient buffer space usage is not required. Speed is more important.

whitequark, 2017-04-15:
That's 80 bytes per event, are you sure? But ok I can do that.

whitequark, 2017-04-15:
I guess we could actually move the serialization code to the kernel CPU side, since we have to do it somewhere anyway.

sbourdeauducq, 2017-04-15:
Playback trigger is still slow, isn't it?

whitequark, 2017-04-18:
Faster playback trigger in 41c4de4; now 6.5us per playback. I have reviewed the assembly for the loop and I do not see any way it could be further optimized unless we link kernels directly with ksupport (and I doubt that will result in much benefit either).

However it is not really clear to me where the code spends ~812 instruction times per iteration. The loop itself contains 70 instructions.

closed 2017-04-18 by whitequark

sbourdeauducq, 2017-04-19:
I got 1.53µs in my test, and I could bring it down to 1.37µs.

commit 9dfe9c12 (2017-04-15): DMA: improve recording performance.

This commit moves DMA serialization code to the kernel CPU
(to cope with the existence of rtio_output_wide) and batches
the resulting sequences. This results in less data being transferred
between kernel and comms CPUs (24 octets with one pointer before,
18 octets with no pointers now, for the common case of rtio_output),
but most importantly reduces cache flushes, which now happen
once per 64k octets.

On average, it now takes about 15us to record a single RTIO event
in a DMA trace.

Fixes #712.