m-labs/artiq #713: inferring return types with annotations

jordens, 2017-04-10:
```python
from artiq.experiment import *
from artiq.coredevice.rtio import rtio_input_data
from artiq.language.types import TInt32

class Test(EnvExperiment):
    def build(self):
        self.setattr_device("core")

    @kernel
    def x(self) -> TInt32:
        return rtio_input_data(0) & 0xff

    @kernel
    def run(self):
        self.x() == 0
```

```
$ artiq_run test_data.py
test_data.py:16:9-16:17: error: the type of this expression cannot be fully inferred
        self.x() == 0
        ^^^^^^^^
test_data.py:16:9-16:17: note: the expression has type numpy.int?
        self.x() == 0
        ^^^^^^^^
```

* Do I really need to wrap `rtio_input_data()` in an `int()`?
* The `TInt32` annotation does not change the outcome.
* `rtio_input_data()` is already annotated. 
* Also fails with `now_mu()`.
* The error message is only readable for experts and is confusing. Superficially it says that the type is unknown. But then it tells you that it still has a certain type. Could it say something like "the expression has incomplete/partially inferred type ..."?

whitequark, 2017-04-10:
> ```
> test_data.py:16:9-16:17: error: the type of this expression cannot be fully inferred
>         self.x() == 0
>         ^^^^^^^^
> ```

This part is definitely a bug, at least.

whitequark, 2017-04-12:
Root causes:
  * The logical AND of (say) `int32` with an `int?` literal gives an `int?`.
  * We ignore any type annotations on kernel functions.

whitequark, 2017-04-12:
> The logical AND of (say) int32 with an int? literal gives an int?.

On reflection, this isn't really a bug, since any logical operation other than AND may need to have the result as wide as the widest operand.

whitequark, 2017-04-13:
> We ignore any type annotations on kernel functions.

We actually didn't ignore them, but the old implementation was very weird and needlessly fragile.

commit 3a1fc729 (2017-04-13): compiler: refactor type annotations recognizing in kernels.

The new implementation is much more generic, more robust,
and shares code with the same for syscalls as well as RPCs.

Fixes #713.
closed 2017-04-13 by whitequark