m-labs/artiq #719: Set/mutate dataset from applet

npisenti, 2017-04-13:
We will want to interact with datasets (beyond `Subscribe`-based plotting) in GUI applets -- for example, setting and changing image ROIs, fussing with fits, etc. From what I gather, the `Publisher`/`Subscriber` pattern is meant as a one-way sync of some data structure from (eg) the master to other objects, but not the reverse. If a "subscriber" wants to mutate that dataset, they must interact with it through the `set-dataset` rpc.

There may be a more elegant way to do this, but I'd like to propose adding an `update_dataset` method to the `SimpleApplet` class, modeled off of the similar function in the `artiq_browser`. For example:

```python
    async def _upload_dataset(self, name, value, persist):
         logger.info("Uploading dataset '%s' to master...", name)
         try:
             remote = RPCClient()
             await remote.connect_rpc(self.args.server, 3251,
                                      "master_dataset_db")
 
             try:
                 await remote.set(name, value, persist)
             finally:
                 remote.close_rpc()
         except:
             logger.error("Failed uplading dataset '%s'",
                          name, exec_info=True)
         else:
             logger.info("Finished uploading dataset '%s'", name)
 
     def upload_dataset(self, name, value, persist=None):
         asyncio.ensure_future(self._upload_dataset(name, value, persist))
````
Then in the `SimpleApplet.__init__` method, we could add something like:
```python
self.main_widget_class = main_widget_class
setattr(self.main_widget_class, "upload_dataset", self.upload_dataset)
```

This would make it easy from derived applets to mutate datasets as needed.

Thoughts/suggestions?

jordens, 2017-04-13:
The use case is definitely valid. But the implementation is way too narrow.

* Why just the dataset db RPC client and not any of the other APIs? Why just that method?
* Don't hardcode ports.
* In this case opening and closing the connection every time is probably not the right approach.
* Don't catch errors that broadly unless you are directly called from a user interaction and unless the user can't be bothered to handle errors.

It's really not much code. I'd consider doing what you want to do explicitly in your applet.

npisenti, 2017-04-14:
>
>
>    - Why just the dataset db RPC client and not any of the other APIs?
>    Why just that method?
>
> Good point.  That could definitely be useful, especially for customizing
interactions with the scheduler. Another thing that we use often currently
is a user-generated "pause" event (in case of laser unlock, etc.). If that
is useful for others as well, it could be a nice addition to the main
scheduler widget, which currently only seems to support "terminate"
actions. I'm thinking something which submits a high priority `while x...`
experiment, with a "resume" button setting `x=false` to return to the
original experiment.

>
>    - Don't hardcode ports.
>
> Yes definitely -- I had added an argparse `--port-rpc` option, but forgot
to change the code I pasted.

>
>    - In this case opening and closing the connection every time is
>    probably not the right approach.
>
> I spent some time looking into using the IPC client, which seems to be a
more natural option (with fallback to use RPC when the applet is run
standalone), but from what I understood that was going to require deeper
modification of the server too in order to support the right actions.
Either way though, I agree leaving the client connection open makes sense.

>
>
> It's really not much code. I'd consider doing what you want to do
> explicitly in your applet.
>
You're right, but this seems to be something many applets could benefit
from, in which case locating the code at a higher level of abstraction
would be nice.

sbourdeauducq, 2017-04-14:
In addition to what Robert said: the applet doesn't know the master address when embedded, all the data goes through the dashboard using the IPC pipe. And when the applet is used in the browser, there is no master.
A good approach would be to proxy the RPC through the IPC pipe, or just send the applet the master connection information through it. Unlike for dataset subscriptions, having each applet connect to the master would not waste much network and processing resources. The standalone applet mode should still be supported, and there should be a well-defined standard about what to do when the master is not available (browser usage).

dhslichter, 2019-09-25:
@sbourdeauducq @jordens have your design ideas from above evolved in the past 2 years?  Seems like a relatively low-complexity item, is this accurate?

sbourdeauducq, 2019-09-30:
Still the same. Yes, it's not very complex, but what to do when the applet is run standalone or in the browser? Should those calls be allowed to fail in regular situations and should the user code be expected to deal with the failures?

dhslichter, 2019-09-30:
I think we just need to agree on some graceful method for the applet to handle the the lack of a master.  Personally I am content to have user code be in charge of dealing with this failure in the appropriate manner, because I think the desired behavior will probably be quite different depending on the situation.  What would be nice would be to have something in ARTIQ that cleanly connects an applets to the master, and raises a reasonable exception when this fails, which the user can catch in the manner they desire.  I think doing this at a higher abstraction level, and perhaps providing an example applet or two (one for interactive plots, one with control buttons like "pause" as @npisenti mentioned, for example), would be nice.  This could be hacked together by people currently, but having it done properly, document, and put in tree seems to me to have real advantages.