m-labs/artiq #721: github buildbot triggers broken

sbourdeauducq, 2017-04-16:
Returns HTTP 500 and here's what's in the log:
```
2017-04-13 00:58:13+0800 [_GenericHTTPChannelProtocol,25,] Attempting to load module buildbot.status.web.hooks.github
2017-04-13 00:58:13+0800 [_GenericHTTPChannelProtocol,25,] processing changes from web hook
        Traceback (most recent call last):
        Failure: exceptions.TypeError: character mapping must return integer, None or unicode
```

whitequark, 2017-04-19:
Fixed.

whitequark, 2017-04-19:
They aren't. The hook invocation succeeds but the changesets are not accepted...

whitequark, 2017-04-20:
The changesets were accepted but the "Project" field changed from e.g. "artiq" to "m-labs/artiq". Fixed now and working.

closed 2017-04-20 by whitequark