m-labs/artiq #722: intermittent core device crash when running test_dma_record_time

sbourdeauducq, 2017-04-19:
Sometimes this happens when running the whole ``test_rtio`` suite:
```
======================================================================
ERROR: test_dma_record_time (test_rtio.DMATest)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/sb/artiq_drtio/artiq/test/coredevice/test_rtio.py", line 597, in test_dma_record_time
    exp.record_many(count)
  File "/home/sb/artiq_drtio/artiq/language/core.py", line 53, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "/home/sb/artiq_drtio/artiq/coredevice/core.py", line 126, in run
    self.comm.serve(embedding_map, symbolizer, demangler)
  File "/home/sb/artiq_drtio/artiq/coredevice/comm_kernel.py", line 560, in serve
    self._read_header()
  File "/home/sb/artiq_drtio/artiq/coredevice/comm_kernel.py", line 153, in _read_header
    (sync_byte, ) = struct.unpack("B", self.read(1))
  File "/home/sb/artiq_drtio/artiq/coredevice/comm_kernel.py", line 134, in read
    rn = self.socket.recv(min(8192, length - len(r)))
ConnectionResetError: [Errno 104] Connection reset by peer

----------------------------------------------------------------------
```

core device log:
```
[   302883957us]  INFO(runtime::session): new connection from 192.168.1.100:36664
[   302890633us]  INFO(runtime::session): idle kernel interrupted
panic at /home/sb/artiq_drtio/artiq/firmware/runtime/lib.rs:184: aborted
rebooting...
```

whitequark, 2017-04-19:
This is just an OOM, I think. Let me add some code to elaborate that out.

whitequark, 2017-04-21:
Root cause identified as a leak of TCP buffers in the analyzer thread.

whitequark, 2017-04-21:
It's a classic: we close the analyzer connection locally, send a FIN, move the socket to TIME-WAIT and it gets stuck there.

commit 726ee737 (2017-04-21): runtime: update smoltcp.

This brings in correct TIME-WAIT handling. Fixes #722.
closed 2017-04-21 by whitequark