m-labs/artiq #723: reduce compile time by replacing llvmlite with bindings to LLVM-C

whitequark, 2017-04-21:
As discussed elsewhere, there are (I think still unpublished) bindings by Alessandro Di Federico. Since we currently spend up to a half (!) of the time concatenating strings in llvmlite, migrating to the LLVM-C API could potentially result in significant speedup.

jordens, 2017-05-11:
Note to self. New bindings are MIT. Involves moving to llvmlite 4.0.

whitequark, 2017-05-11:
Bindings: https://github.com/revng/llvmcpy

whitequark, 2017-05-11:
Or more precisely it is an LLVM binding generator, not LLVM bindings. This also greatly reduces (ideally to zero) the amount of work binding-side that needs to be done when upgrading LLVM in the future. No more hacking around llvmlite limitations \o/

whitequark, 2017-11-01:
LLVM-C binding just [gained DIBuilder support](https://reviews.llvm.org/rL317135), which means we will retain the debug information when migrating to LLVM-C.

sbourdeauducq, 2018-01-27:
What kind of speed improvement can we expect from this, exactly?

whitequark, 2018-01-27:
You can run the profiler, then cut the time spent doing anything LLVM-related in half.

sbourdeauducq, 2021-12-13:
llvmlite isn't the biggest problem (https://github.com/m-labs/artiq/issues/1380) and is not used anymore in NAC3 anyway.

closed 2021-12-13 by sbourdeauducq