m-labs/artiq #724: conda numba dependency conflicts with llvmlite-artiq

jbqubit, 2017-04-25:
Using dev branch. Fresh conda environment. 

```
(artiq3) britton@boshi:~/github/m-labs/artiq/artiq/examples/master$ artiq_master
ERROR:worker(scan,remote_exec_demo.py):root:Terminating with exception (ImportError: No module named 'numba')
...
```

```
$ conda install numba
Fetching package metadata ...............
Solving package specifications: ..........

Package plan for installation in environment /home/britton/miniconda3/envs/artiq3:

The following packages will be downloaded:

    package                    |            build
    ---------------------------|-----------------
    numba-0.32.0               |      np111py35_0         2.3 MB

The following NEW packages will be INSTALLED:

    llvmlite: 0.17.0-py35_0     
    numba:    0.32.0-np111py35_0

...ABORT...
```

Can't install stock conda numba due to dependency on llvmlite. ARTIQ depends on llvmlite-artiq.
```
             0.12.0.dev              py35_30    m-labs/label/main
```

Can't install both llvmlite and llvmlite-artiq at the same time without other bugs appearing (eg dependence on llvmlite.so).

whitequark, 2017-04-25:
> Can't install both llvmlite and llvmlite-artiq at the same time without other bugs appearing (eg dependence on llvmlite.so).

Which bugs exactly? We have separate llvmlite and llvmlite-artiq specifically so that numba and ARTIQ can coexist.

jordens, 2017-05-11:
@jbqubit ?

closed 2017-06-05 by sbourdeauducq