m-labs/artiq #727: remove multiple jesd initialization attempts if not necessary

enjoy-digital, 2017-04-27:
https://github.com/m-labs/artiq/commit/79c339d4ac1440b24b84a2f4210e773a7ee029c5#commitcomment-21926746

hartytp, 2018-01-03:
Did you try asking about this on the ADI forums? They might be able to tell you if it's expected behaviour for these chips (or a known bug). They're generally pretty helpful, so definitely worth a post.

Also, IIRC (but do correct me if I'm wrong about that) they have now released their own open source JSED core, which they use for testing. @enjoy-digital did you look at how that core handles initialisation? Does it include automatic retries or anything else that would handle this? A quick look at that core might give some useful info on how these chips are intended to be initialised...

jbqubit, 2018-01-08:
> [ADI has] now released their own open source JSED core, which they use for testing.

Is this what you mean @hartytp? 

https://ez.analog.com/thread/96757-xilinx-jesd-core-vs-adi-jesd-core

hartytp, 2018-01-09:
Not really my field of expertise, but I believe it was this: https://github.com/analogdevicesinc/hdl/tree/dev/library/jesd204. @enjoy-digital is probably the person to ask about these things...

enjoy-digital, 2018-01-09:
@hartytp: yes we can look at the core from ADI and see if we are doing initialization correctly.

sbourdeauducq, 2018-05-01:
> can we be 100% that the Xilinx transceivers will initialize correctly??

Xilinx transceivers have many faults, but they are IME reliable when all their annoying needs are catered for.
Let's keep this open, we can close it after we get more experience with the boards and the code with all known bugs fixed.

sbourdeauducq, 2018-05-07:
RTM-1 DAC1 (but not DAC0) sometimes fails with this transient fault and still requires multiple attempts:
```
[    10.837273s]  WARN(board_artiq::ad9154): AD9154-1 config attempt #0 failed (bad SYNC), retrying
```
Problem is not observed on RTM-2.
RTM-F not tested since DAC0 is borked.

enjoy-digital, 2018-05-07:
ok thanks.  That would be interesting to have two full setups on the server (AMC + RTM).

sbourdeauducq, 2018-05-10:
I have now set up RTM-1 and AMC-3 and the problem is easily reproduced. Turns out DAC0 is also affected.
```
[     4.210974s]  INFO(board_artiq::hmc830_7043::hmc830): HMC830 found
[     4.217122s]  INFO(board_artiq::hmc830_7043::hmc830): HMC830 configuration...
[     4.224350s]  INFO(board_artiq::hmc830_7043::hmc830): waiting for lock...
[     4.231069s]  INFO(board_artiq::hmc830_7043::hmc7043): HMC7043 found
[     4.237355s]  INFO(board_artiq::hmc830_7043::hmc7043): HMC7043 configuration...
[     4.255482s]  INFO(board_artiq::ad9154): AD9154-0 found
[     4.259418s]  INFO(board_artiq::ad9154): AD9154-0 configuration...
[     4.347794s]  WARN(board_artiq::ad9154): AD9154-0 config attempt #0 failed (bad SYNC), retrying
[     4.365464s]  INFO(board_artiq::ad9154): AD9154-0 found
[     4.369392s]  INFO(board_artiq::ad9154): AD9154-0 configuration...
[     4.451918s]  INFO(board_artiq::ad9154): AD9154-0 PRBS test
[     5.456504s]  INFO(board_artiq::ad9154): AD9154-0 SYSREF scan/conf...
[     6.053593s]  INFO(board_artiq::ad9154):   phase: 58, sync error: 481
[     6.700853s]  INFO(board_artiq::ad9154):   phase: 122, sync error: 482
[     6.706097s]  INFO(board_artiq::ad9154):   phase min: 58, phase max: 121, phase opt: 89
[     6.724326s]  INFO(board_artiq::ad9154): AD9154-0 found
[     6.728255s]  INFO(board_artiq::ad9154): AD9154-0 configuration...
[     6.821008s]  INFO(board_artiq::ad9154): AD9154-1 found
[     6.824943s]  INFO(board_artiq::ad9154): AD9154-1 configuration...
[     6.907527s]  INFO(board_artiq::ad9154): AD9154-1 PRBS test
[     7.912106s]  INFO(board_artiq::ad9154): AD9154-1 SYSREF scan/conf...
[     8.188144s]  INFO(board_artiq::ad9154):   phase: 26, sync error: 481
[     8.835403s]  INFO(board_artiq::ad9154):   phase: 90, sync error: 482
[     8.840558s]  INFO(board_artiq::ad9154):   phase min: 26, phase max: 89, phase opt: 57
[     8.858703s]  INFO(board_artiq::ad9154): AD9154-1 found
[     8.862633s]  INFO(board_artiq::ad9154): AD9154-1 configuration...
[     8.945224s]  INFO(board_artiq::hmc542): card 0 channel 0 set to 4 dB
[     8.952365s]  INFO(board_artiq::hmc542): card 0 channel 1 set to 4 dB
[     8.959490s]  INFO(board_artiq::hmc542): card 1 channel 0 set to 4 dB
[     8.966614s]  INFO(board_artiq::hmc542): card 1 channel 1 set to 4 dB
[     8.973740s]  INFO(board_artiq::hmc542): card 2 channel 0 set to 4 dB
[     8.980864s]  INFO(board_artiq::hmc542): card 2 channel 1 set to 4 dB
[     8.987989s]  INFO(board_artiq::hmc542): card 3 channel 0 set to 4 dB
[     8.995115s]  INFO(board_artiq::hmc542): card 3 channel 1 set to 4 dB
```
I'm leaving them on the server for the investigation of this issue.

sbourdeauducq, 2018-05-10:
Though it is probably a good idea to fix serwb timing before looking at bugs in areas that depend on serwb.

sbourdeauducq, 2018-07-10:
@enjoy-digital Any idea why that happens? If I remember correctly, this bug was also on the KC705, so it's not a Sayma issue.

enjoy-digital, 2018-07-13:
I would need to investigate. I'll try to see if i'm able to reproduce it on the reduced designs with my boards.

sbourdeauducq, 2018-07-13:
Maybe talking to AD (e.g. forums) would go a long way. Could you post there?

closed 2018-08-17 by sbourdeauducq

hartytp, 2018-08-17:
nice!

sbourdeauducq, 2018-08-17:
This bug seems gone for good, all DAC inits worked the first time in 87 reboots.

jbqubit, 2018-08-20:
That's great @sbourdeauducq!

commit 041dc0f6 (2018-08-17): jesd204: update core to v0.10

Closes #727
Closes #1127