m-labs/artiq #728: ARTIQ 2.3 install from source: MISOC and MIGEN versions

jasonamini, 2017-04-27:
What versions of MISOC and MIGEN should I be using for the ARTIQ 2.3 build from source? 

Some while back I was using tag/0.3 for both, but that no longer works. With those versions, I get the error:
_File "/home/qatest/artiq/dev/misoc/misoc/interconnect/csr.py", line 11, in __init__
    raise ValueError("Cannot extract CSR name from code, need to specify.")_
If I upgrade to MISOC tag/0.4 or later, I get instead:
  _File "/home/qatest/artiq/dev/artiq/artiq/gateware/targets/kc705.py", line 11, in <module>
    from migen.fhdl.specials import Keep
ImportError: cannot import name 'Keep'_
Tried various version of MIGEN and was unable to get rid of this error.

I am building on a clean Ubuntu 16.04 VM so no install history. For reference, these are the package versions I am using in my install script:
ARTIQ_BRANCH="release-2"
MISOC_BRANCH="tags/0.3"
PYTHONPARSER_BRANCH="master"
PYGIT2_VERSION="0.24.1" # Pygit2 version for Ubuntu 16.04. Version matches that of libgit2-dev.
LLVMLITE_BRANCH="artiq-3.8"
ASYNCSERIAL_BRANCH="master"
MIGEN_BRANCH="tags/0.3"
LLVM_BRANCH="artiq-3.8"
CLANG_BRANCH="artiq-3.8"

cjbe, 2017-04-27:
If you install the artiq-dev package this will be handled automatically.

Alternatively, you can look at the artiq-dev [conda build script](https://github.com/m-labs/artiq/blob/release-2/conda/artiq-dev/meta.yaml) to see what packages it installs.

sbourdeauducq, 2017-04-27:
> raise ValueError("Cannot extract CSR name from code, need to specify.")

This looks like you're running Python 3.6, which is not supported. See #652.

jasonamini, 2017-04-28:
Swapping to a python 3.5 environment works. Also switched to migen 0.4 to match artiq-dev. Unfortunately, it just gets me back to my original install problem before I inadvertently switched to 3.6 with an error trying to build libunwind. That's a separate issue. If I can't figure it out (will check on versions in artiq-dev), I will open another issue.

@cbje: I had not known that that package exists. I like it. It pretty much eliminates my whole install script (but would still like to get that working). Took a bit to work out how to use it (i.e. needing to separately clone artiq, where to point the path for llvm, and issue #537) but it got through to the Vivado step. I could not find a manual page for this, which would be useful as part of the install from source pages. 

Thx all!

closed 2017-04-28 by jasonamini