m-labs/artiq #729: Missing sphinx_rtd_theme distribution in artiq 2.3

r-srinivas, 2017-05-04:
I installed artiq 2.3 on our virtual box using,

```
rabi@68810MAGTRAPVM:~$ conda create -n artiq-2017-05-04 artiq-kc705-nist_qc2 -c m-labs/channel/main
```
With the following packages in the environment,

```
rabi@68810MAGTRAPVM:~$ source activate artiq-2017-05-04
discarding /home/rabi/anaconda3/bin from PATH
prepending /home/rabi/anaconda3/envs/artiq-2017-05-04/bin to PATH
(artiq-2017-05-04)rabi@68810MAGTRAPVM:~$ conda list
# packages in environment at /home/rabi/anaconda3/envs/artiq-2017-05-04:
#
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
aiohttp                   0.17.2                     py_3    http://conda.anaconda.org/m-labs/channel/main/noarch/aiohttp-0.17.2-py_3.tar.bz2
artiq                     2.3             py_0+gitc78aa86e7272    http://conda.anaconda.org/m-labs/channel/main/noarch/artiq-2.3-py_0+gitc78aa86e7272.tar.bz2
artiq-kc705-nist_qc2      2.3             py_0+gitc78aa86e7272    http://conda.anaconda.org/m-labs/channel/main/noarch/artiq-kc705-nist_qc2-2.3-py_0+gitc78aa86e7272.tar.bz2
asyncserial               0.1             py_11+git758c1e5    http://conda.anaconda.org/m-labs/channel/main/noarch/asyncserial-0.1-py_11+git758c1e5.tar.bz2
binutils-or1k-linux       2.27                          5    http://conda.anaconda.org/m-labs/channel/main/linux-64/binutils-or1k-linux-2.27-5.tar.bz2
cffi                      1.9.1                    py35_0    defaults
chardet                   3.0.2                    py35_0    defaults
colorama                  0.3.7                     <pip>
dateutil                  2.4.2                      py_2    http://conda.anaconda.org/m-labs/channel/dev/noarch/dateutil-2.4.2-py_2.tar.bz2
dbus                      1.10.10                       0    defaults
expat                     2.1.0                         0    defaults
fontconfig                2.12.1                        3    defaults
freetype                  2.5.5                         2    defaults
glib                      2.50.2                        1    defaults
gst-plugins-base          1.8.0                         0    defaults
gstreamer                 1.8.0                         0    defaults
h5py                      2.7.0               np112py35_0    defaults
hdf5                      1.8.17                        1    defaults
icu                       54.1                          0    defaults
jpeg                      9b                            0    defaults
levenshtein               0.12.0                   py35_1    http://conda.anaconda.org/m-labs/channel/main/linux-64/levenshtein-0.12.0-py35_1.tar.bz2
libffi                    3.2.1                         1    defaults
libgcc                    5.2.0                         0    defaults
libgfortran               3.0.0                         1    defaults
libgit2                   0.24.1                        7    http://conda.anaconda.org/m-labs/channel/main/linux-64/libgit2-0.24.1-7.tar.bz2
libiconv                  1.14                          0    defaults
libpng                    1.6.27                        0    defaults
libssh2                   1.7.0.git                     5    http://conda.anaconda.org/m-labs/channel/main/linux-64/libssh2-1.7.0.git-5.tar.bz2
libusb                    1.0.20                        0    http://conda.anaconda.org/m-labs/channel/main/linux-64/libusb-1.0.20-0.tar.bz2
libxcb                    1.12                          1    defaults
libxml2                   2.9.4                         0    defaults
lit                       0.4.1                      py_2    http://conda.anaconda.org/m-labs/channel/main/noarch/lit-0.4.1-py_2.tar.bz2
llvmlite                  0.6.0+176.g6582da0           <pip>
llvmlite-artiq            0.10.0.dev              py35_24    http://conda.anaconda.org/m-labs/channel/main/linux-64/llvmlite-artiq-0.10.0.dev-py35_24.tar.bz2
mkl                       2017.0.1                      0    defaults
numpy                     1.12.1                   py35_0    defaults
openocd                   0.10.0                        1    http://conda.anaconda.org/m-labs/channel/main/linux-64/openocd-0.10.0-1.tar.bz2
openssl                   1.0.2k                        1    defaults
outputcheck               0.4.2                      py_1    http://conda.anaconda.org/m-labs/channel/main/noarch/outputcheck-0.4.2-py_1.tar.bz2
pcre                      8.39                          1    defaults
pip                       9.0.1                    py35_1    defaults
prettytable               0.7.2                      py_3    http://conda.anaconda.org/m-labs/channel/dev/noarch/prettytable-0.7.2-py_3.tar.bz2
pycparser                 2.17                     py35_0    defaults
pygit2                    0.24.0                   py35_4    http://conda.anaconda.org/m-labs/channel/main/linux-64/pygit2-0.24.0-py35_4.tar.bz2
pyqt                      5.6.0                    py35_2    defaults
pyqt5                     5.5.1                    py35_0    http://conda.anaconda.org/m-labs/channel/main/linux-64/pyqt5-5.5.1-py35_0.tar.bz2
pyqtgraph                 0.10.0                   py35_0    defaults
pyserial                  2.7                      py35_0    defaults
python                    3.5.2                         0    defaults
python-dateutil           2.4.2                     <pip>
python-levenshtein        0.12.0                    <pip>
pythonparser              1.0                        py_7    http://conda.anaconda.org/m-labs/channel/main/noarch/pythonparser-1.0-py_7.tar.bz2
qt                        5.6.2                         3    defaults
qt5                       5.5.1                         1    http://conda.anaconda.org/m-labs/channel/main/linux-64/qt5-5.5.1-1.tar.bz2
quamash                   0.5.5                      py_1    http://conda.anaconda.org/m-labs/channel/main/noarch/quamash-0.5.5-py_1.tar.bz2
[... 65 lines cut ...]
    imagesize:              0.7.1-py35_0       
    jinja2:                 2.9.6-py35_0       
    markupsafe:             0.23-py35_2        
    pygments:               2.2.0-py35_0       
    pytz:                   2017.2-py35_0      
    requests:               2.13.0-py35_0      
    snowballstemmer:        1.2.1-py35_0       
    sphinx:                 1.5.4-py35_0       
    sphinx-argparse:        0.1.13-py_4        
    sphinx_rtd_theme:       0.2.4-py35_0       
    sphinxcontrib-wavedrom: 1.1.0-py_1         

The following packages will be UPDATED:

    python:                 3.5.2-0                  --> 3.5.3-1            

The following packages will be DOWNGRADED:

    artiq:                  2.3-py_0+gitc78aa86e7272 --> 2.1-py_0+git453b29a
    artiq-kc705-nist_qc2:   2.3-py_0+gitc78aa86e7272 --> 2.1-py_0+git453b29a

Proceed ([y]/n)? 
```

Did I mess up the installation in some way?

jordens, 2017-05-11:
fbbfdb07

jordens, 2017-05-11:
```diff
--- /tmp/artiq_2.1.info 2017-05-11 14:48:09.249276356 +0200
+++ /tmp/artiq_2.3.info 2017-05-11 14:47:55.612886262 +0200
@@ -1,22 +1,22 @@
 Fetching package metadata .................

-artiq 2.1 py_0+git453b29a
--------------------------
-file name   : artiq-2.1-py_0+git453b29a.tar.bz2
+artiq 2.3 py_0+gitc78aa86e7272
+------------------------------
+file name   : artiq-2.3-py_0+gitc78aa86e7272.tar.bz2
 name        : artiq
-version     : 2.1
-build string: py_0+git453b29a
+version     : 2.3
+build string: py_0+gitc78aa86e7272
 build number: 0
 channel     : m-labs/channel/main
-size        : 1.1 MB
+size        : 1.2 MB
 arch        : None
 license     : GPL
-md5         : a50dbf18d6d698756bc5ea81354397c2
+md5         : 25ad83063e5cc7dba25a00f19e964c81
 noarch      : None
 platform    : None
 requires    : ()
 subdir      : noarch
-url         : https://conda.anaconda.org/m-labs/channel/main/noarch/artiq-2.1-py_0+git453b29a.tar.bz2
+url         : https://conda.anaconda.org/m-labs/channel/main/noarch/artiq-2.3-py_0+gitc78aa86e7272.tar.bz2
 dependencies:
     aiohttp
     asyncserial
@@ -27,16 +27,13 @@
     lit
     llvmlite-artiq 0.10.0.dev py35_24
     numpy
+    openocd >=0.10
     outputcheck
     prettytable
     pygit2
-    pyqt5
-    pyqtgraph
-    python >=3.5.2
-    pythonparser
+    pyqt >=5.5
+    pyqtgraph 0.10.0
+    python 3.5.2
+    pythonparser 1.0
     quamash
     scipy
-    sphinx
-    sphinx-argparse
-    sphinx_rtd_theme
-    sphinxcontrib-wavedrom

```

closed 2017-05-11 by jordens

jordens, 2017-05-12:
@r-srinivas this is fixed and the next 2.4 release will contain that fix. Meanwhile as a workaround with artiq 2.3, you should be able to `conda install sphinx_rtd_theme sphinxcontrib-wavedrom sphinx-argparse python=3.5.2` (after having created the initial environment) or `conda create -n artiq-2017-05-04 artiq-kc705-nist_qc2 sphinx_rtd_theme sphinxcontrib-wavedrom sphinx-argparse python=3.5.2 -c m-labs/channel/main`.