m-labs/artiq #730: Timing not met in NIST_QC2 gateware

cjbe, 2017-05-08:
Using the master branch, I cannot build a NIST_QC2 gateware that meets timing - this seems to be the case since the DMA core was added.

I am using the same build of Vivado as the build-bot (Vivado 2016.2).
The Vivado timing report is [here](https://github.com/m-labs/artiq/files/983610/top_timing.txt)

Any suggestions on how to get this to meet timing?
(I am actually trying to build a custom gateware with a similar number of TTLs as QC2, but without the DDS channels and get similar timing errors).

sbourdeauducq, 2017-05-08:
With Vivado, meeting timing or not is pretty random. I've just started a build on the bot to see if also fails there...

Considering where the timing fails in your case, I would consider adding a register or small FIFO into the data path to relax the timing and move the average performance that Vivado gets further from the timing failure point.

sbourdeauducq, 2017-05-08:
Yup, works on the buildbot. http://buildbot.m-labs.hk/builders/artiq-board/builds/553

cjbe, 2017-05-08:
I have now tried building this 5 times in a row, and all met with similar timing errors.

Do you know of anything that could make the build work on your build-bot and not on my system?

sbourdeauducq, 2017-05-08:
Generated the Vivado input files on the buildserver:
[test_nist_qc2.zip](https://github.com/m-labs/artiq/files/984147/test_nist_qc2.zip)

Double checking if those meet timing there, can you also compile them and check?

sbourdeauducq, 2017-05-08:
Turns out I also get the timing error when run manually, but not when buildbot runs it ...

Anyway, the proper solution is to relax the timing as explained above.

sbourdeauducq, 2017-07-02:
Fixed in 838127d914802eec5dec84ba5fed469ca9920b60.

closed 2017-07-02 by sbourdeauducq