m-labs/artiq #732: "unacceptable SEQ" errors on experiment load

cjbe, 2017-05-09:
Loading the below experiment causes the core device to get stuck in a TCP re-transmission loop during experiment loading. This seems to be another manifestation of #685

```python
class LargeBinary(EnvExperiment):
    def build(self):
        self.setattr_device("core")

        self.len = 4000

    def run(self):
        data = [i for i in range(self.len)]
        self.krun(data)

    @kernel
    def krun(self, data):
        self.core.reset()
        i=0
        for t in data:
            i+=t
        return i
```

Core device log in TRACE mode is [here](https://github.com/m-labs/artiq/files/986469/ttylog_only_router2.txt)
(I patched it to report the ACK sequence number. The problem still occurs with TRACE mode off).

[Packet dump](https://github.com/m-labs/artiq/files/986471/dump_only_router2.zip)

jordens, 2017-05-11:
@whitequark duplicate of #685 ?

whitequark, 2017-05-11:
> Loading the below experiment causes the core device to get stuck in a TCP re-transmission loop during experiment loading.

This is not a loop as smoltcp is making progress (just very poorly).

@jordens yes.

closed 2017-05-12 by jordens

whitequark, 2017-06-26:
@cjbe As a matter of fact, with the current master the entire experiment finishes in under 100ms.