m-labs/artiq #737: Large applet latency (linux->windows)

cjbe, 2017-05-28:
When running an Artiq master on Linux, and plotting a dataset via an applet running on Windows, there is a significant latency, and a noticeable bunching of updates. This makes it tricky to optimise a signal in real time (e.g. tweak a mirror to align a laser beam). 

This problem does not occur with an applet running on the same machine as the master, or when the applet is run on a separate Linux machine.

As an example, running a kernel that updates a dataset every ~55ms:
On Linux, the dataset updates are received by the applet every 55ms +- 5ms
On Windows, the dataset updates are received in a pattern with delta-t {180ms, 9ms, 9ms, 6ms} repeating.
This accounts for the latency (~200ms) and the update bunching.

This appears to be an interaction between Nagles algorithm on the Linux side and the Winsock delayed ack mechanism. Disabling delayed ack on Windows (via the registry) or using TCP_NODELAY on Linux fixes this.

This should be fixed after upgrading to Python 3.6 (#652) via https://bugs.python.org/issue27456

klickverbot, 2017-05-28:
Why don't we just explicitly do `get_extra_info("socket").setsockopt(socket.IPPROTO_TCP, socket.TCP_NODELAY, 1)` to fix this on 3.5 as well?

I don't think we'd ever want to use TCP_NODELAY in ARTIQ anyway – if we somehow end up saturating the link with tiny packets, we can just fix our code instead.

sbourdeauducq, 2017-05-29:
This should be fixed with Python 3.5.3 as well.

sbourdeauducq, 2017-06-01:
@cjbe Do you still have the problem with 3.5.3? Or will it not install for some reason?

cjbe, 2017-06-03:
@sbourdeauducq installing 3.5.3 does indeed fix this.

klickverbot, 2017-06-03:
Let's make sure 3.5.3+ is recommended in the installation instructions and close this, then?

klickverbot, 2017-06-04:
Just requiring it is even less error-prone, of course. Thanks!

closed 2017-06-04 by klickverbot