m-labs/artiq #738: spurious "Mismatch between gateware and software versions" messages

sbourdeauducq, 2017-06-04:
e.g.
Mismatch between gateware (3.0.dev+1074.g8c8203ef) and software (3.0.dev+1074.g8c8203ef7) versions

Probably related to 0d0ddf2e587dc12f15f47beba4858f2448a3b367.

closed 2017-06-05 by sbourdeauducq

jordens, 2017-06-06:
This is just a length issue. Why make it less selective?

sbourdeauducq, 2017-06-06:
Is there any drawback to making it less selective? The advantage is obviously to make it less sensitive to python packaging snafu.

sbourdeauducq, 2017-06-06:
And "just" a length issue, sure. Many bugs that take days to figure out are "just" a one-line fix, right?

jordens, 2017-06-06:
Yes. The drawback is that it doesn't work anymore. If sensitivity is the problem, then getting rid of the version check entirely would be the answer.
From what I can see a length mismatch on the comparison should be resolved with storing the correct strings and not a softening of the comparison, right? Fix and issue appear to be orthogonal.
Since we pragmatically chose to not have a versioned gateware/runtime/software interface, the only thing that reminds one that there can be a mismatch is this warning. And just warning on release number mismatch is not the dominant class of bugs we have been catching so far with this.
About the "just": That's certainly possible if there are other causes for this issue that were not mentioned in these commits and this issue.

whitequark, 2017-06-06:
How about embedding the complete git hash instead, and truncating it only for display?

jordens, 2017-06-06:
Wouldn't that remove the '.dirty' flag as well and prevent eyeballing the mismatch severity using the version numbers?
Other than that: comparing just based on the git hash and then using those "package version strings" (maybe in addition to the full hash) for warning display would be great.

whitequark, 2017-06-06:
No, I mean, complete git hash instead of truncated git hash, so that we don't get these length issues in comparison.

sbourdeauducq, 2017-06-06:
We were already removing ``.dirty`` because it generates a lot of noise (have a few uncommitted modifications laying around, and it triggers).

jordens, 2017-06-06:
I honestly prefer that noise over having to (and forgetting to) walk each user and myself through `git status`/`git diff` just to be sure that there are not "a few uncommitted modifications lying around".
Aren't there more annoying things than that bit of noise?

commit bea7a473 (2017-06-05): coredevice: compare gateware and software versions on release numbers only. Closes #738
commit 478972a0 (2017-06-05): coredevice: compare gateware and software versions on release numbers only. Closes #738