m-labs/artiq #740: Coredevice access to convert SPI busses on Sayma RTM

jordens, 2017-06-06:
f3c7a7de3be9a89a14b2d5e5540affd0393d4591

sbourdeauducq, 2017-06-19:
The files in f3c7a7de3be9a89a14b2d5e5540affd0393d4591, except ``test_ad9154_status.py`` that can be reintroduced as-is, do other manipulations to the JESD core that are a bit annoying to support over DRTIO. What about moving them to firmware? Perhaps those tests could be run at each boot.

jordens, 2017-06-19:
Running them at Sayma boot is good. Depending on the clocking we might have to delay that though.

sbourdeauducq, 2017-06-19:
The satellite manager should be capable of bringing up all the clocking by itself.

sbourdeauducq, 2017-06-20:
Though I remember there was some crash/bug that required a reboot of the device after STAPL or PRBS test had been done. Is that fixed? Cc @enjoy-digital

enjoy-digital, 2017-06-20:
@sbourdeauducq: that's possible you will need to reconfigure the device ADI chip after the STAPL/PRBS. That's something I can test/fix if needed when I'll be integrating and the core on Sayma.

closed 2017-06-21 by sbourdeauducq