m-labs/artiq #741: doc : coredevice method arguments too complex

jbqubit, 2017-06-07:
Core drivers reference artic.coredevice.sawg and .spline have unnecessarily complex argument types presented in documentation. For example sawg.config.set_clr().

In email, @jordens says
> That's a feature of the compiler, not of SAWG. Feel free to raise an issue.

and @sbourdeauducq says
>Getting sphinx to use str() instead of repr() on type annotations should fix this problem.
```
>>> str(TInt32)
'numpy.int32'
>>> repr(TInt32)
"artiq.compiler.types.TMono('int', OrderedDict([('width', artiq.compiler.types.TValue(32))]))"
```

sbourdeauducq, 2017-06-08:
https://github.com/sphinx-doc/sphinx/issues/3857
I have tested that the proposed change in ``format_annotation()`` does result in types being nicely printed.

jordens, 2017-06-08:
We could also just overload repr.

sbourdeauducq, 2017-06-08:
ARTIQ does the right thing in returning an ``eval()``-parseable string in ``repr()``.

whitequark, 2017-06-08:
@jordens It's used when debugging, because the user-readable representation can hide some details that may make inference fail. Of course that is not visible from source code.

dnadlinger, 2019-10-11:
This issue is back after a refactoring in Sphinx; the repr() calls are now in https://github.com/sphinx-doc/sphinx/blob/7faeb793e2e16cde4e5759443fb7f84efddcd9ea/sphinx/util/inspect.py.

dnadlinger, 2019-10-11:
(Testing patch.)

sbourdeauducq, 2019-10-11:
We can also use an actual (not monkey) patch on Hydra, the advantage is it'll likely show clearly when it breaks. But users of unpatched sphinx will have the issue.

sbourdeauducq, 2021-08-05:
It broke again.

commit 14714d3f (2017-11-25): Hack-patch Sphinx so that ARTIQ-Python types are correctly printed.

Modification proposed to Sphinx but my issue is getting ignored.

Closes #741
commit 14e4e1cf (2017-11-25): Hack-patch Sphinx so that ARTIQ-Python types are correctly printed.

Modification proposed to Sphinx but my issue is getting ignored.

Closes #741
commit 97e99470 (2021-08-05): compiler: turn __repr__ into __str__ when sphinx is used. Closes #741
commit dda4121c (2021-08-05): compiler: turn __repr__ into __str__ when sphinx is used. Closes #741
closed 2021-08-05 by sbourdeauducq