m-labs/artiq #742: sawg: shot-to-shot variation

jbqubit, 2017-06-08:
Testing SAWG absolute/relative phase modes (link)[https://github.com/m-labs/artiq/blob/master/artiq/coredevice/sawg.py] I see funny shot-to-shot variation. 

```
from artiq.experiment import *

class SAWGTest(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("led")
        self.setattr_device("ttl_sma")

        self.setattr_device("sawg0")
        self.setattr_device("sawg1")

    @kernel
    def run(self):
        self.core.reset()
        self.ttl_sma.output()
        delay(1*ms)

        while True:
            self.ttl_sma.pulse(3 * us)
            self.ap1()
            delay(1*ms)

    @kernel
    def ap1(self):
        # test: t1
        f0 = 5*MHz
        t = 1*us
        a0 = 0.99

        # prepare
        self.sawg0.frequency0.set(0*MHz)
        self.sawg0.frequency1.set(f0)
        self.sawg0.frequency2.set(0*MHz)
        self.sawg0.amplitude1.set(a0)
        self.sawg0.config.set_clr(1, 1, 1)
        self.sawg1.frequency0.set(0*MHz)
        self.sawg1.frequency1.set(f0)
        self.sawg1.frequency2.set(0*MHz)
        self.sawg1.amplitude1.set(a0)
        self.sawg0.config.set_clr(1, 1, 1)

        with parallel:
            self.sawg0.phase1.set(0.)
            self.sawg1.phase1.set(0.)
        delay(t)

        with parallel:
            self.sawg0.phase1.set(-0.25)
            self.sawg1.phase1.set(0.25)
        delay(t)

        # switch RF off
        self.sawg0.amplitude1.set(0.)
        self.sawg1.amplitude1.set(0.)
```
Here's what I see with scope triggered on ttl_sma. Several single trigger events:
[image]
[image]
[image]
[image]

Unexpected: Glitch varies from one shot to next.

jordens, 2017-06-08:
Let's exclude a couple of things (cheap, hacked scope, interactions of cables with scope, missing termination, balun, DAC output stage) on your end first:

* verify with another, faster scope
* check at small amplitude
* take a 10x (or better FET) probe directly at the resistively terminated differential DAC output

jbqubit, 2017-06-08:
It was a triggering problem with the Rigol scope. Using similar bandwidth Tek scope the shot-to-shot variation goes away. Closing.

closed 2017-06-08 by jbqubit