m-labs/artiq #744: sawg api: absolute vs relative phase modes

jbqubit, 2017-06-09:
Initial [api](https://github.com/m-labs/artiq/blob/master/artiq/coredevice/sawg.py#L52) implementation distinguishes between absolute and relative phase update modes using a switch. 

> set_clr(clr0, clr1, clr2)
Set the phase clear mode for the three phase [accumulators](https://github.com/m-labs/artiq/blob/master/artiq/coredevice/sawg.py#L208). When the ``clr`` bit for a given phase accumulator is set, that phase accumulator will be cleared with every phase RTIOcommand and the output phase will be exactly the phase RTIO value ("absolute phase update mode"). In turn, when the bit is cleared, the phase RTIO channels only provide a phase offset to the current value of the phase accumulator ("relative phase update mode").

Then, to set the absolute (or increment the relative) phase the call is 
````
self.sawg1.phase1.set(ph)
```

This is linguistically annoying. I propose instead doing away with the ```clr``` switch and defining:
```
self.sawg1.phase1.set(ph) # for absolute phase
self.sawg1.phase1.advance(ph) # for relative phase
```

klickverbot, 2017-06-09:
(Fixed code block syntax.)

jordens, 2017-06-09:
@jboulder Can we please not abuse the API just to scratch a linguistic itch?
It would also be wrong. Relative phase mode is just relative to the carrier (and not to "event-time" as in absolute phase mode). Relative phase updates are not cumulative. `advance()` would imply cumulative behavior.
But I am happy to tweak the documentation if it is linguistically challenging.

jordens, 2017-06-09:
d8aee931

closed 2017-06-09 by jordens