m-labs/artiq #745: sawg: RTIO timing error between SAWG and TTLOut

jbqubit, 2017-06-09:
Scope is Keysight MXOS4104A 5 GS/s. Triggering on ttl_sma. 
```
from artiq.experiment import *

class SAWGTest(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("led")
        self.setattr_device("ttl_sma")

        self.setattr_device("sawg0")
        self.setattr_device("sawg1")

    @kernel
    def run(self):
        self.core.reset()
        self.ttl_sma.output()
        delay(3*ms)

        while True:
            self.ttl_sma.pulse(3*us)
            self.freqs()
            delay(1*ms)

    @kernel
    def freqs(self):
        # test: t1
        t = 1*us
        a0 = 0.5

        # prepare
        self.sawg0.config.set_clr(1, 1, 1)
        self.sawg0.frequency0.set(2*MHz)
        self.sawg0.frequency1.set(0*MHz)
        self.sawg0.frequency2.set(0*MHz)
        self.sawg0.amplitude1.set(a0)
        self.sawg0.amplitude2.set(a0)

        self.sawg1.config.set_clr(1, 1, 1)
        self.sawg1.frequency0.set(0*MHz)
        self.sawg1.frequency1.set(0*MHz)
        self.sawg1.frequency2.set(0*MHz)
        self.sawg1.amplitude1.set(a0)
        self.sawg1.amplitude2.set(a0)

        self.sawg1.frequency0.set(2*MHz)
        delay(t)

        self.sawg1.frequency0.set(4*MHz)
        delay(t)

        self.sawg1.frequency0.set(2*MHz)
        delay(t)

        # switch RF off
        self.sawg1.amplitude1.set(0.)
        self.sawg1.amplitude2.set(0.)
```
Movie shows slow slip in relative phase. 
https://drive.google.com/open?id=0B5PIArGKtLa-eWZOQVNZQ05Mamc

jordens, 2017-06-09:
What should I be looking at?
Remember that a high resolution TTL can be at fRTIO/8 resolution while the DAC on that board only does fRTIO/2. Depending on the accumulation and rounding of delays, you are bound to see some effect.

jbqubit, 2017-06-09:
See movie in link which shows slip. ARTIQ needs a mechanism to permit synchronization of all RTIO/DRTIO devices the outset of an experiment.

jordens, 2017-06-09:
I have understood that I should look at the movie. But what in the movie shows "RTIO timing error between SAWG and TTLOut"? If you want decent bug triage, please provide a decent description of the problem.

What do you mean by "a mechanism to permit synchronization of all RTIO/DRTIO devices the outset of an experiment"? You seem to have an implementation in mind. If you just make sure your experiment starts at a LCM of the various time resolutions of your devices, does that resolve your problem? You may remember that I tried to point out to you last year that asking for different resolutions and data rates on different channels/devices will always require significantly more brain work by the user.

jordens, 2017-06-09:
Are you sure you are not just seeing a beating between the timebase of your oscilloscope and the RTIO/DAC frequency? Again, I don't know what I should be looking at in that movie.

jbqubit, 2017-06-09:
Yes I'm sure. Here's another video in which its easier to see. Or you could run the code on your own machine. 

https://drive.google.com/open?id=0B5PIArGKtLa-b3FOS003OTU2aWs

jordens, 2017-06-10:
Please describe what you are trying to do, what you expect as the outcome and how your observation differs from that. Me running code on my machine does not answer any of those questions. Nor do any of your videos. In your video I see a beat of the DDS phase and the TTL pulse. That is not surprising given your experiment. Maybe it is to you but unless you add the required information there is little I can do and we are both wasting time.

jbqubit, 2017-06-12:
I don't want a beat between the DDS phase and the TTL pulse rising edge. 

After playing around a bit it looks like adding ```with parallel``` eliminates the beat. Here's a more minimal example. 
``` 
from artiq.experiment import *

class SAWGTest(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("led")
        self.setattr_device("ttl_sma")

        self.setattr_device("sawg0")

    @kernel
    def run(self):
        self.core.reset()
        self.ttl_sma.output()
    
        # prepare
        t = 1*us
        a0 = 0.5
        self.sawg0.config.set_clr(1, 1, 1)
        self.sawg0.frequency0.set(2*MHz)
        self.sawg0.frequency1.set(0*MHz)
        self.sawg0.frequency2.set(0*MHz)
        self.sawg0.amplitude1.set(0.)
        self.sawg0.amplitude2.set(0.)
        delay(3 * ms)

        while True:
            with parallel:
                self.ttl_sma.pulse(3*us)
                self.sawg0.phase0.set(0.)

            delay(t)
            self.sawg0.amplitude1.set(a0)
            delay(t)
            self.sawg0.amplitude1.set(0.)
            delay(1*ms)
```

sbourdeauducq, 2017-06-12:
So it works as expected? Can we close this?

jbqubit, 2017-06-12:
OK.

closed 2017-06-12 by jbqubit

jordens, 2017-06-12:
If you don't want the beat, then reset the phase or choose a frequency that has zero beat.